espressif/arduino-esp32 #6988: Local Control not enabled

JohnMacrae, 2022-07-14:
### Board

Any

### Device Description

Any

### Hardware Configuration

none

### Version

v2.0.4

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

Local control does not seem to be enabled.  Please can the next version be compiled with this enabled.

### Sketch

```cpp
None required. The setting is in the compiled library
```

### Debug Message

```plain
None required
```

### Other Steps to Reproduce

Of course it works using IDF - its even a default in the current menuconfig

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-16:
@me-no-dev - Any opinion about it - could it be included for 2.0.5 or 2.1.0?

VojtechBartoska, 2022-07-18:
@JohnMacrae Thanks for the idea. We will check if enabling it will cause any problems in Arduino and if not, we will enable this and include it in Arduino Lib-builder.

maxakash, 2024-02-15:
@VojtechBartoska @SuGlider has local control support added or not?

SuGlider, 2024-02-15:
No support in the Arduino Core.

It is possible to add Local Control by using Arduino as IDF Component in the project, calling its IDF API within the Arduino Sketch.

It would be necessary to set it up using IDF Menuconfig and then building the project.

SuGlider, 2025-01-14:
Closing this issue.

Local Control in IDF https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_local_ctrl.html
Example from IDF https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_local_ctrl/README.md

Arduino Core already has necessary elements for its implementation: HTTPS, CA Certificates, WiFi and BLE.
The application side shall implement its own protocol, information exchange and excution based on the IDF Example.

SuGlider, 2025-01-14:
If necessary, `start_esp_local_ctrl_service();` shall be implemented using the IDF Component source code from GitHub.
Another way to add it to Arduino will be by using Arduino as IDF Component project and adding the necessary settings to `sdkconfig` file in order to enable IDF Local Control functionality.

closed 2025-01-14 by SuGlider