espressif/arduino-esp32 #6996: Error Compiling... Arduino IDE version 1.8.19 ... Board Manager ... ESP32 Espressif System version 2.0.4 INSTALLED

rodyeo, 2022-07-16:
### Board

AI Thinker ESP32-CAM

### Device Description

ESP32-S

### Hardware Configuration

Standard programming pins used on 5V FTDI Serial Programmer

### Version

v2.0.4

### IDE Name

Arduino IDE version 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Error Compiling...
Arduino IDE version 1.8.19
Examples\ESP32\Camera\CameraWebServer
Board Manager ... ESP32 Espressif System version 2.0.4 INSTALLED

Error never happen on ESP32 Espressif System version 2.0.3 previously.

Also notice brown out detector library is missing in version 2.0.4 please include it in next release.

#include "soc/soc.h" //disable brownout problems
#include "soc/rtc_cntl_reg.h" //disable brownout problems
WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector

### Sketch

```cpp
#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
//            Ensure ESP32 Wrover Module or other board with PSRAM is selected
//            Partial images will be transmitted if image exceeds buffer size
//
//            You must select partition scheme from the board menu that has at least 3MB APP space.
//            Face Recognition is DISABLED for ESP32 and ESP32-S2, because it takes up from 15 
//            seconds to process single frame. Face Detection is ENABLED if PSRAM is enabled as well

// ===================
// Select camera model
// ===================
//#define CAMERA_MODEL_WROVER_KIT // Has PSRAM
[... 176 lines cut ...]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1255:25: warning: missing initializer for member 'httpd_uri::supported_subprotocol' [-Wmissing-field-initializers]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1261:25: warning: missing initializer for member 'httpd_uri::is_websocket' [-Wmissing-field-initializers]
         .user_ctx = NULL};
                         ^
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1261:25: warning: missing initializer for member 'httpd_uri::handle_ws_control_frames' [-Wmissing-field-initializers]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1261:25: warning: missing initializer for member 'httpd_uri::supported_subprotocol' [-Wmissing-field-initializers]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1267:25: warning: missing initializer for member 'httpd_uri::is_websocket' [-Wmissing-field-initializers]
         .user_ctx = NULL};
                         ^
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1267:25: warning: missing initializer for member 'httpd_uri::handle_ws_control_frames' [-Wmissing-field-initializers]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1267:25: warning: missing initializer for member 'httpd_uri::supported_subprotocol' [-Wmissing-field-initializers]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1273:25: warning: missing initializer for member 'httpd_uri::is_websocket' [-Wmissing-field-initializers]
         .user_ctx = NULL};
                         ^
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1273:25: warning: missing initializer for member 'httpd_uri::handle_ws_control_frames' [-Wmissing-field-initializers]
C:\Users\god\Documents\Arduino\CameraWebServer\app_httpd.cpp:1273:25: warning: missing initializer for member 'httpd_uri::supported_subprotocol' [-Wmissing-field-initializers]
```

### Other Steps to Reproduce

Click on Verify ... Error Compiling ...

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-16:
@rodyeo - Please check #6993 

Let me know if this PR fixes this issue. Thanks.

rodyeo, 2022-07-17:
> @rodyeo - Please check #6993
> 
> Let me know if this PR fixes this issue. Thanks.

I modified the "app_httpd.cpp" include file as per instructions and recompile with ESP32 Board Manager version 2.0.4 latest installed library with no compilation error. The problem have been is resolved! Thank you very much for the solution! You may safely commit this fix to the master branch for global users to install the new fixed CameraWebServer example source code. Thanks a million buddy! Cheers ;) Regards, Rodney Yeo @ http://rodyeo.dyndns.org

rodyeo, 2022-07-17:
Updated fix at my social media page ... https://m.facebook.com/story.php?story_fbid=pfbid0y7DynNxnamRqpENxj4qJLqfP1oDTeTRMsHuqzb79E8FeShw9C3EnzteysC2AaWkYl&id=554017662

rodyeo, 2022-07-17:
I shall close this case as resolved! Thanks everyone ;)

closed 2022-07-17 by rodyeo