espressif/arduino-esp32 #6999: Have Wire1 pins autoset just like Wire

ladyada, 2022-07-16:
### Related area

I2C Peripheral

### Hardware specification

Any hardware based on ESP32

### Is your feature request related to a problem?

Wire is auto-setup for any ESP variant to the SDA/SCL pins
https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_qtpy_esp32s2/pins_arduino.h#L29
but SDA1/SCL1 pins do not do anything... it would be nice if setPins() was not required if these are #defined!
https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_qtpy_esp32s2/pins_arduino.h#L32

### Describe the solution you'd like

we can submit a PR, but want to know whether it would be rejected before spending the time.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-07-16:
>we can submit a PR, but want to know whether it would be rejected before spending the time.

I vote for moving forward with the PR.

Take into consideration that the ESP32C3 has a single I2C peripheral, while ESP32, ESP-S2 and ESP-S3 have two.
Maybe `pins_arduino.h` must reflect it. 

`#ifdef ...` may solve it within 
https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp#L66-L105

=======================
Number of I2C peripherals can be verified using `<soc/soc_caps.h>` at `SOC_I2C_NUM`
https://github.com/espressif/esp-idf/blob/master/components/soc/esp32c3/include/soc/soc_caps.h#L157

https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/soc_caps.h#L168
https://github.com/espressif/esp-idf/blob/master/components/soc/esp32s2/include/soc/soc_caps.h#L150
https://github.com/espressif/esp-idf/blob/master/components/soc/esp32s3/include/soc/soc_caps.h#L156

VojtechBartoska, 2022-07-18:
We discussed it today with whole team and we don't see an issue about this. PR is very welcomed! @ladyada

SuGlider, 2022-07-20:
@ladyada - Are you currently working on it? 
I'm currently working in the Wire API to add a `size_t Wire::set BufferSize(size_t)` function and I can then move to the `Wire1 pin autoset feature` after finishing.
Please let me know.

SuGlider, 2022-07-25:
@ladyada - Please evaluate and test PR #7034

ladyada, 2022-07-31:
hi sorry i was totally halfway thru writing this PR and got completely swamped with a ton of folks visiting NYC in the last two weeks.  will look now!

ladyada, 2022-08-01:
reviewed and works great - it will close this issue when merged :)

closed 2022-08-01 by SuGlider

ladyada, 2022-08-01:
thank you @SuGlider for the great PR! 😅

PR #7034 (2022-08-01): Adds default Wire1 pins auto-setup
## Description of Change
Some boards define SDA1 and SCL1 pins for Wire1 peripheral.
This information isn't used in `Wire1.begin()` as pins default.
The PR adds this functionality and set default pins when they are correctly created in `pins_arduino.h`

These pins shall be set in `variants/<board_name>/pins_arduino.h` like this example:

``` cpp
#define WIRE1_PIN_DEFINED 1             // See Wire.cpp at bool TwoWire::initPins(int sdaPin, int sclPin)
static const uint8_t SDA1 = 2;
static const uint8_t SCL1 = 4;
```

## Tests scenarios
Tested with ESP32/S2/S3.
 [... cut]