espressif/arduino-esp32 #7001: Serial2.onReceive() not work.

neolu, 2022-07-17:
### Board

ESP32

### Device Description

DevKitC

### Hardware Configuration

#define RXD2 16
#define TXD2 17

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

win10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Serial2.onReceive() not work.

  //Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
  Serial2.onReceive(Serial2_callback);

void Serial2_callback() {
    Serial.printf(".");
}

### Sketch

```cpp
Serial2.onReceive() not work.

  //Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
  Serial2.onReceive(Serial2_callback);

void Serial2_callback() {
    Serial.printf(".");
}
```

### Debug Message

```plain
Serial2.onReceive() not work.

  //Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
  Serial2.onReceive(Serial2_callback);

void Serial2_callback() {
    Serial.printf(".");
}
```

### Other Steps to Reproduce

Serial2.onReceive() not work.

  //Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
  Serial2.onReceive(Serial2_callback);

void Serial2_callback() {
    Serial.printf(".");
}

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-18:
It seems that it works correctly, but the issue is actually that it doesn't work using the default pins...
I tried this and it worked fine:

``` cpp
void setup() {
Serial2.begin(115200, SERIAL_8N1, 32, 33);
Serial2.onReceive(Serial2_callback);
}

void Serial2_callback() {
Serial.printf(".");
}
```

I will investigate why it doesn't work with just  a `Serial2.begin(115200);`

SuGlider, 2022-08-12:
I used 2 ESP32 for testing: 

- ESP32 WROOM
- ESP32 WROVER-E

With both I used this Sketch:

``` cpp
void setup() {
  Serial.begin(115200);

  Serial2.begin(115200, SERIAL_8N1, 32, 33);
  Serial2.onReceive(Serial2_callback);
}

void Serial2_callback() {
  Serial.printf(".");
}

void loop() {
}
```

### Connected RX0 pin (GPIO 3) to RX2 pin GPIO 32 in the example sketch
Using Arduino IDE Serial Monitor, send anything to ESP32. RX0 pin will get the signal and it will pass the same signal to RX2 by the wire. This way `Serial2.onReceive()` will be activated.

### ESP32 WROOM can use pins 16 and 17 for Serial2 (default RX2, TX2)
### ESP32 WROVER-E CAN NOT use pins 16 or 17 because those are used with PSRAM. Thus it only work with other pins, such as 32 an 33.

closed 2022-08-12 by SuGlider

asking23, 2022-10-12:
ESP32 DEVKIT V1 when i upgraded from Arduino ESP32 version 1.0.3 to 2.0.5 it stopped working default pins 16/17.

TD-er, 2022-10-12:
Is PSRAM support enabled?

I have this in my setup() function:
```c++
#ifdef BOARD_HAS_PSRAM
  psramInit();
#endif

#ifdef CONFIG_IDF_TARGET_ESP32
  // restore GPIO16/17 if no PSRAM is found
  if (!FoundPSRAM()) {
    // test if the CPU is not pico
    uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_PKG);
    uint32_t pkg_version = chip_ver & 0x7;
    if (pkg_version <= 3) {   // D0WD, S0WD, D2WD
      gpio_reset_pin(GPIO_NUM_16);
      gpio_reset_pin(GPIO_NUM_17);
    }
  }
#endif  // CONFIG_IDF_TARGET_ESP32
```

Along with these functions, borrowed from the Tasmota source code:
```c++
// this function is a replacement for `psramFound()`.
// `psramFound()` can return true even if no PSRAM is actually installed
// This new version also checks `esp_spiram_is_initialized` to know if the PSRAM is initialized
// Original Tasmota:
// https://github.com/arendst/Tasmota/blob/1e6b78a957be538cf494f0e2dc49060d1cb0fe8b/tasmota/support_esp.ino#L470
bool FoundPSRAM() {
# if CONFIG_IDF_TARGET_ESP32C3
  return psramFound();
# else // if CONFIG_IDF_TARGET_ESP32C3
  return psramFound() && esp_spiram_is_initialized();
# endif // if CONFIG_IDF_TARGET_ESP32C3
}

// new function to check whether PSRAM is present and supported (i.e. required pacthes are present)
bool UsePSRAM() {
  static bool can_use_psram = CanUsePSRAM();

  return FoundPSRAM() && can_use_psram;
}

/*
 * ESP32 v1 and v2 needs some special patches to use PSRAM.
 * Original function used from Tasmota:
 * https://github.com/arendst/Tasmota/blob/1e6b78a957be538cf494f0e2dc49060d1cb0fe8b/tasmota/support_esp.ino#L762
 *
 * If using ESP32 v1, please add: `-mfix-esp32-psram-cache-issue -lc-psram-workaround -lm-psram-workaround`
 *
 * This function returns true if the chip supports PSRAM natively (v3) or if the
 * patches are present.
 */
bool CanUsePSRAM() {
  if (!FoundPSRAM()) { return false; }
# ifdef HAS_PSRAM_FIX
  return true;
# endif // ifdef HAS_PSRAM_FIX
# ifdef CONFIG_IDF_TARGET_ESP32
  esp_chip_info_t chip_info;
  esp_chip_info(&chip_info);

  if ((CHIP_ESP32 == chip_info.model) && (chip_info.revision < 3)) {
    return false;
  }
#  if ESP_IDF_VERSION_MAJOR < 4
  uint32_t chip_ver    = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_PKG);
  uint32_t pkg_version = chip_ver & 0x7;

  if ((CHIP_ESP32 == chip_info.model) && (pkg_version >= 6)) {
    return false; // support for embedded PSRAM of ESP32-PICO-V3-02 requires esp-idf 4.4
  }
#  endif // ESP_IDF_VERSION_MAJOR < 4

# endif // CONFIG_IDF_TARGET_ESP32
  return true;
}
```