espressif/arduino-esp32 #7005: ESP32C3 undefined reference to `esp_sleep_enable_ext0_wakeup'

Bei-Ji-Quan, 2022-07-19:
### Board

ESP32C3 Dev Module

### Device Description

ESP32C3 Dev Module

### Hardware Configuration

GPIO

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When build the example "ExternalWakeup" ,  undefined reference to `esp_sleep_enable_ext0_wakeup'

### Sketch

```cpp
/*
Deep Sleep with External Wake Up
=====================================
This code displays how to use deep sleep with
an external trigger as a wake up source and how
to store data in RTC memory to use it over reboots

This code is under Public Domain License.

Hardware Connections
======================
Push Button to GPIO 33 pulled down with a 10K Ohm
resistor

NOTE:
======
Only RTC IO can be used as a source for external wake
source. They are pins: 0,2,4,12-15,25-27,32-39.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define BUTTON_PIN_BITMASK 0x200000000 // 2^33 in hex
extern "C" esp_err_t esp_sleep_enable_ext0_wakeup(gpio_num_t gpio_num, int level);
RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up for an external trigger.
  There are two types for ESP32, ext0 and ext1 .
  ext0 uses RTC_IO to wakeup thus requires RTC peripherals
  to be on while ext1 uses RTC Controller so doesnt need
  peripherals to be powered on.
  Note that using internal pullups/pulldowns also requires
  RTC peripherals to be turned on.
  */
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_9,1); //1 = High, 0 = Low

  //If you were to use ext1, you would use it like
  //esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK,ESP_EXT1_WAKEUP_ANY_HIGH);

  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}
```

### Debug Message

```plain
e:/documents/arduino/hardware/arduino-esp32/arduino-esp32/tools/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\Heltec\AppData\Local\Temp\arduino_build_525227\sketch\ExternalWakeUp.ino.cpp.o: in function `setup()':
C:\Users\Heltec\AppData\Local\Temp\arduino_modified_sketch_258584/ExternalWakeUp.ino:54: undefined reference to `esp_sleep_enable_ext0_wakeup'
collect2.exe: error: ld returned 1 exit status
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-19:
If you redefine the function by calling `extern "C" esp_err_t esp_sleep_enable_ext0_wakeup(gpio_num_t gpio_num, int level);`, it probably isn't going to find the code.  The example as included in the repo compiles.

Bei-Ji-Quan, 2022-07-19:
It is the same without "extern "C" esp_err_t esp_sleep_enable_ext0_wakeup(gpio_num_t gpio_num, int level);"

lbernstone, 2022-07-19:
Ah, esp32-c3 does not support rtc wakeup. esp_deep_sleep_enable_gpio_wakeup should work:
`esp_deep_sleep_enable_gpio_wakeup(1 << 9,ESP_GPIO_WAKEUP_GPIO_HIGH);`

VojtechBartoska, 2022-08-24:
@Bei-Ji-Quan Can we close this as answered?

Bei-Ji-Quan, 2022-08-30:
Yes, it can be closed, thanks.

closed 2022-08-30 by Bei-Ji-Quan

udayalawa, 2023-04-19:
> Ah, esp32-c3 does not support rtc wakeup. esp_deep_sleep_enable_gpio_wakeup should work: `esp_deep_sleep_enable_gpio_wakeup(1 << 9,ESP_GPIO_WAKEUP_GPIO_HIGH);`

thanks!!