espressif/arduino-esp32 #7006: PROBLEMS WITH ACCESS POINT

franciscogimeno2000, 2022-07-19:
### Board

ESP32 DEVKIT V1 (30 PIN)

### Device Description

Just as it comes from the factory, only the USB to program and power

### Hardware Configuration

no

### Version

v2.0.4

### IDE Name

ARDUINO IDE 1.8.19.0

### Operating System

WIN 10

### Flash frequency

240MHZ

### PSRAM enabled

no

### Upload speed

115.200

### Description

Hello friends, I'm going crazy, I can't create an access point with esp32 a few months ago things worked without problems, but I don't know what's happening that the access point doesn't work.

I'm not new to this, and I've tried everything, I don't know what else to do.

I have changed versions of ESP to almost all, and also the Arduino IDE I have tried with 4 or 5 ESP32 DEVKIT boards, NEW. I have programmed them at different speeds and there is no way.

I have also tried compiling from other PCs.

Everything else works, including connecting to a router and making updates to them by downloading a file from a server.

The official example used is unmodified, once transferred the esp starts without problems but the access point is not seen from any pc or from any mobile.

Any ideas? Has anyone experienced this?

Thanks for the help.
Francisco

### Sketch

```cpp
//Unmodified official example

/*
  WiFiAccessPoint.ino creates a WiFi access point and provides a web server on it.

  Steps:
  1. Connect to the access point "yourAp"
  2. Point your web browser to http://192.168.4.1/H to turn the LED on or http://192.168.4.1/L to turn it off
     OR
     Run raw TCP "GET /H" and "GET /L" on PuTTY terminal with 192.168.4.1 as IP address and 80 as port

  Created for arduino-esp32 on 04 July, 2018
  by Elochukwu Ifediora (fedy0)
*/

#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>

#define LED_BUILTIN 2   // Set the GPIO pin where you connected your test LED or comment this line out if your dev board has a built-in LED

// Set these to your desired credentials.
const char *ssid = "yourAP";
const char *password = "yourPassword";

WiFiServer server(80);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();

  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn ON the LED.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn OFF the LED.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Configuring access point...
AP IP address: 192.168.4.1
Server started
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2022-07-19:
Hello @franciscogimeno2000 

I've tried the example with Arduino 2.0.3 and 2.0.4. In both cases it worked as expected. However with 2.0.4 I occasionally see a watchdog error in the log output. I've added a `delay(10` into the `loop()` and the issue went away.

Thanks
Felix

franciscogimeno2000, 2022-07-19:
Hi, thanks, I'll try.
But I have finished projects that have worked without problems and now they don't work... and only the AP part.

On the other hand, I have connected new SONOFF modules and the access points do not appear either on the PC or on the mobile... I don't know what could be happening.

franciscogimeno2000, 2022-07-19:
Witches...ghosts...

franciscogimeno2000, 2022-07-19:
I just tried with the pause and nothing, still the same.
I have tried a module that was finished and working and the access point does not appear either, I do the search with 3 different PCs and 2 mobiles... I don't know what could happen, I am lost, this had never happened to me.

franciscogimeno2000, 2022-07-19:
Nothing... leave it guys I have taken the equipment to another area of the offices and they work. what I have to find out is why they don't work in my office.

Thank you all

closed 2022-07-19 by franciscogimeno2000