espressif/arduino-esp32 #7007: Olimex ESP32 PoE ISO not working since 1.0.6

FStefanni, 2022-07-19:
### Board

Olimex ESP32 PoE ISO

### Device Description

Olimex ESP32 PoE ISO

### Hardware Configuration

None

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi,

by using the Olimex official example, I have discovered that the last working Arduino ESP32 version was 1.0.6.
Since 2.0.0, no version is working (tested with both 2.0.0 and 2.0.4).

The issue seems that no more ethernet events are raised, so the sketch does not work anymore.
Maybe some breaking changes?

Regards

### Sketch

```cpp
// I use the official basic example from Olimex, here reported for your convenience:
// https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino

/*
    This sketch shows the Ethernet event usage
*/
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 12

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

### Debug Message

```plain
No message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2022-07-19:
Hello @FStefanni 

the OLIMEX example you've listed is about 4 years old. Have a look at the current Arduino [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino). It's quite similar but all `SYSTEM_EVENT_` events have been renamed to `ARDUINO_EVENT_`. Maybe that is your issue?

Thanks
Felix

FStefanni, 2022-07-20:
Hi @felmue 

Oh yeah! Thank you very much!
So strange the official Olimex documetation did not updated or pointed to the updated version.

Regards

closed 2022-07-20 by FStefanni