espressif/arduino-esp32 #7011: ESP32S3 chips SHA-256 comparison failed problem

Paryavi, 2022-07-19:
### Board

Custom Designed board with ESP32S3Wroom-1U 

### Device Description

Custom Designed board 

### Hardware Configuration

It is our custom design board and we have added a few functionalities to it.

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

DIO 80 MHz and QIO 80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

An important note is as I have S3 Wroom 1U with 16 MB Flash, I choose ESP32 S3 Dev Module in Board selection in Arduino IDE to program our custom-designed board. 
1. After each upload, I always get SHA-256 comparison failed, how can I solve it?
2. Sometimes I get heap poisoning issues as shown in Debug message. Is it because the ESP32 S3 Arduino code is still in development mode(not stable)? 

### Sketch

```cpp
For most standard sketches, I get the following in S3 chip;

SHA-256 comparison failed:
Calculated: 9477899d59fa5164e630d8218cb3994031873358e9105b31f53911d2dfd36c72
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
```

### Debug Message

```plain
CORRUPT HEAP: Bad head at 0x3fce5a6c. Expected 0xabba1234 got 0x3fce0014

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x403777a6:0x3fce2d000x4037a8f1:0x3fce2d20 0x4038031d:0x3fce2d40 0x4037ff8f:0x3fce2e70 0x40377a75:0x3fce2e90 0x4038034d:0x3fce2eb0 0x42007c1b:0x3fce2ed0 0x42008cb8:0x3fce2f00 0x42001fac:0x3fce2f20 0x42002111:0x3fce2f40 0x42001f22:0x3fce2f60 0x42001daf:0x3fce2f80 0x4200e7bd:0x3fce3030 

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202cbfe
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
SHA-256 comparison failed:
Calculated: 9477899d59fa5164e630d8218cb3994031873358e9105b31f53911d2dfd36c72
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
[   124][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
[   124][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
```

### Other Steps to Reproduce

Thank you!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Paryavi, 2022-07-24:
Hi @VojtechBartoska and @me-no-dev 

We figured out the rebooting issue, we were using GPIO 35 and 36 which were dedicated to PSRAM, we rerouted the board and sent a new PCB order that does not use GPIO 35 and 36 and we will solder the new version soon.
Meanwhile, my research advisor had removed the routes to GPIO 35 and 36 on our current custom designed dev board, and now I still get SHA-256 comparison fail as follows, and program runs and I do not get crash but I can't use buffer;

New Arduino Serial Prints:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
SHA-⸮ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x39c
load:0x403c9700,len:0x9bc
load:0x403cc700,len:0x28dc
SHA-256 comparison failed:
Calculated: 9477899d59fa5164e630d8218cb3994031873358e9105b31f53911d2dfd36c72
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
[   119][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

We switched from ESP32Wroom to this ESP32-S3-WROOM-1U-N16R8 (16MB flash and 8 MB PSRAM) because it has more GPIOs and we are connecting to a u-blox modem.
and we have ordered 250 ESP-32S3 ; https://www.digikey.com/en/products/detail/espressif-systems/ESP32-S3-WROOM-1U-N16R8/16162641

But when running u-blox code I get memory buffer error; https://github.com/u-blox/ubxlib/blob/master/common/sock/api/u_sock_errno.h#L145

ESP32 connect to network using ublox, and it was sending socket message with ESP32 WROOM but on the ESP32 S3 board, it can not allocate buffer? for socket, right?
@me-no-dev How can I solve the (flash or PSRAM?) buffer issue on ESP32 S3(SHA-256 comparison failed)? 

Ublox prints when using ESP32S3:

U_CELL_NET: connected after 5 second(s).
Looking up server address...
U_CELL_SOCK: looking up IP address of "cloudsocket.hologram.io".
AT+CGDCONT?
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","100.69.155.137",0,0,0,0

OK
AT+UDNSRN=0,"cloudsocket.hologram.io"
AT+UDNSRN=0,"cloudsocket.hologram.io"

+UDNSRN: "34.233.146.107", "3.210.32.221"

OK
U_CELL_SOCK: found it at "34.233.146.107".
Address is: IPV4 34.233.146.107
Creating socket...
AT+USOCR=6
AT+USOCR=6

ERROR

ERROR

+USOCR: 0

OK
AT+USOER
AT+USOER

+USOER: 65

OK
U_SOCK: underlying socket layer could not create socket (errno 105).
Unable to connect to server!
Closing socket...
1 Tests 1 Failures 0 Ignored

My Arduino setup for ESP32S3;
[image]

People's solution for SHA fail on ESP-IDF; https://www.esp32.com/viewtopic.php?f=12&t=27454&hilit=SHA+256+comparison+failed
https://www.esp32.com/viewtopic.php?t=10143

VojtechBartoska, 2022-07-25:
Thanks @Paryavi  for nice explanation!

@P-R-O-C-H-Y Can you please take a look?

xinkiknix, 2022-08-01:
I have similar issue with ESP32S3 Dev Module
[image]

lbernstone, 2022-08-02:
This is an "expected error".  It is not possible to do a proper checksum verification with arduino, as the bootloader is not generated for every firmware.

closed 2022-08-02 by Paryavi

manundi, 2022-09-09:
So the solution is to use esressif ide? 

I'm having same error than @xinkiknix with esp 32 c3 13u.

xinkiknix, 2022-09-09:
You just ignore the issue, it has no impact on running code whatsoever

manundi, 2022-09-09:
[image]
But my program code wont start running. Tried simple serial examples etc. I am not using a kit, Im using a usb serial stick to flash.

letis009, 2022-09-26:
@manundi , the rst code is brownout_rst , I doubt the checksum is preventing boot 
had similar issues but turned out those were dodgey caps.

me-no-dev, 2022-09-26:
fixed in 2.0.5

lewisxhe, 2022-10-02:
I encountered the same problem. I found that the root cause of the problem was the reference to the wrong partition table ` huge_ App. csv ', when I use the default partition table, it is normal.. Just as a hint

poopoofoo, 2023-03-22:
> 

I use esp-idf environment ，met the same issue

guidol70, 2023-05-06:
> fixed in 2.0.5

but it seems to be back in 2.0.8 and 2.0.9 (also had it in 2.0.2 and 2.0.3) 
with two ESP32-C3 RiscV CPU-Boards (ESP32-C3-32S and ESP32-C3-01M-KIT)
when I do a SPIFFS.format:

```
E (104) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='1a8e7181'
Image checksum='0ff85c5b'
```
when going back to 2.0.5 and then using SPIFFS.format I do only get the normal BootROM-Messages:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403820b8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
entry 0x403ce000

```

On a ESP32 Lite (normal Xtensa DualCore) this problem doenst exist with 2.0.8 and 2.0.9

guidol70, 2023-05-06:
> fixed in 2.0.5
@me-no-dev 
Did you also see this info in the 2.0.5 Release notes or was that only a result of testing it with 2.0.5?

me-no-dev, 2023-05-06:
@guidol70 this is a totally different "error" and you should disregard it.

ronsims2, 2023-09-10:
I experience the same class of error using the [https://www.espressif.com/en/dev-board/esp32-c3-devkit-rust-1-en](ESP32-C3) board when trying to run [https://github.com/esp-rs/std-training/blob/main/advanced/i2c-sensor-reading/examples/part_2.rs](this example) from the Rust training book.

ronsims2, 2023-09-10:
> I experience the same class of error using the [ESP32-C3](https://www.espressif.com/en/dev-board/esp32-c3-devkit-rust-1-en) board when trying to run [this example](https://github.com/esp-rs/std-training/blob/main/advanced/i2c-sensor-reading/examples/part_2.rs) from the Rust training book.
> 
> It gets stuck in a hex dump before just stopping.

Interestingly, when i run the same code on a different board there are no warnings or errors and the code runs.