espressif/arduino-esp32 #7023: Unable to connect WiFi SSID with Encryption WIFI_AUTH_WPA_PSK : WPA/PSK Encryption since 2.0.2

GitHub1YA, 2022-07-21:
### Board

ESP32 all

### Device Description

Tested and verified on Standard ESP32 Wemos D1, ESP32-CAM ...

PSRAM or not

### Hardware Configuration

no special configuration

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10 pro 

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

9921600

### Description

Can't connect on WiFi SSID witch are encrypted with  WIFI_AUTH_WPA_PSK : WPA/PSK Encryption since version 2.0.2 (include 2.0.2, 2.0.3 and 2.0.4.

Other WiFi can connect such as WIFI_AUTH_WPA2_PSK : WPA2 Encryption

The error code returned is 201

18:19:53.651 -> WiFi scan done : 7 networks found
18:19:53.651 -> 01 : Canal : '04', RSSI (-48), BSSID '02:52:4A:60:84:E3', SSID 'SFR_E301', , Encryption : * WIFI_AUTH_WPA2_PSK : WPA2 Encryption
18:19:53.684 -> 02 : Canal : '04', RSSI (-62), BSSID '24:95:04:CB:E3:04', SSID 'SFR_E300', , Encryption : * WIFI_AUTH_WPA_PSK : WPA/PSK Encryption
....
18:19:53.801 ->
18:19:53.801 -> Starting WiFi on SSID : SFR_E300
18:19:53.801 -> [ 94592][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
18:19:54.251 -> ..................................[111708][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: SFR_E300, BSSID: 00:00:00:00:00:00, Reason: 201
18:20:10.941 -> [111709][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
18:20:10.941 -> [111716][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
18:20:10.941 -> [111723][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
18:20:10.941 -> [111730][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0

### Sketch

```cpp
WiFi.mode(WIFI_STA);                                        // Ensure WiFi mode is Station Mode

  WiFi.begin(SSID, SSIDPasswd);  // Standard WIFI_STA on infrastructure AP cnx to ssid, password
```

### Debug Message

```plain
he error code returned is 201

18:19:53.651 -> WiFi scan done : 7 networks found
18:19:53.651 -> 01 : Canal : '04', RSSI (-48), BSSID '02:52:4A:60:84:E3', SSID 'SFR_E301', , Encryption : * WIFI_AUTH_WPA2_PSK : WPA2 Encryption
18:19:53.684 -> 02 : Canal : '04', RSSI (-62), BSSID '24:95:04:CB:E3:04', SSID 'SFR_E300', , Encryption : * WIFI_AUTH_WPA_PSK : WPA/PSK Encryption
....
18:19:53.801 ->
18:19:53.801 -> Starting WiFi on SSID : SFR_E300
18:19:53.801 -> [ 94592][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
18:19:54.251 -> ..................................[111708][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: SFR_E300, BSSID: 00:00:00:00:00:00, Reason: 201
18:20:10.941 -> [111709][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
18:20:10.941 -> [111716][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
18:20:10.941 -> [111723][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
18:20:10.941 -> [111730][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-21:
WPA_PSK is now considered insecure.  You can enable it by calling:
`WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK);`

GitHub1YA, 2022-07-22:
Thanks for reply !
It work now !

I ask : Why change such a default value between minor revisions ? 

Why return a '201 Not found' error, instead of 'unsafe', or 'unauthorised', error code ?
...
Best regards.

closed 2022-07-22 by GitHub1YA