espressif/arduino-esp32 #7024: sprintf statement only gives compile error if compiled with compiler warning level "All" and "More" (-Wall -Werror=all)

ArminJo, 2022-07-21:
### Board

compiler problem

### Device Description

compiler problem

### Hardware Configuration

compiler problem

### Version

v2.0.4

### IDE Name

Arduino, Sloeber

### Operating System

%

### Flash frequency

compiler problem

### PSRAM enabled

no

### Upload speed

compiler problem

### Description

The sketch below throws the error
```
E:\WORKSPACE_ARDUINO\ArduinoSketches\sketch_jul21a\sketch_jul21a.ino: In function 'void setup()':
sketch_jul21a:7:26: error: '%s' directive writing up to 19 bytes into a region of size 14 [-Werror=format-overflow=]
   sprintf(sStringBuffer, "Acc.X %s", &sStringBuffer[10]);
   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\WORKSPACE_ARDUINO\ArduinoSketches\sketch_jul21a\sketch_jul21a.ino:7:10: note: 'sprintf' output between 7 and 26 bytes into a destination of size 20
   sprintf(sStringBuffer, "Acc.X %s", &sStringBuffer[10]);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors
exit status 1
'%s' directive writing up to 19 bytes into a region of size 14 [-Werror=format-overflow=]
```
 if compiled with -Wall -Werror=all

**For all non ESP32 platforms it compiles without any error!** - see https://github.com/ArminJo/Arduino-BlueDisplay/actions/runs/2714467954

### Sketch

```cpp
char sStringBuffer[20];

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  dtostrf(123.45, 4, 1, &sStringBuffer[10]);
  sprintf(sStringBuffer, "Acc.X %s", &sStringBuffer[10]);
  Serial.println(sStringBuffer);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
E:\WORKSPACE_ARDUINO\ArduinoSketches\sketch_jul21a\sketch_jul21a.ino: In function 'void setup()':
sketch_jul21a:7:26: error: '%s' directive writing up to 19 bytes into a region of size 14 [-Werror=format-overflow=]
   sprintf(sStringBuffer, "Acc.X %s", &sStringBuffer[10]);
   ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:\WORKSPACE_ARDUINO\ArduinoSketches\sketch_jul21a\sketch_jul21a.ino:7:10: note: 'sprintf' output between 7 and 26 bytes into a destination of size 20
   sprintf(sStringBuffer, "Acc.X %s", &sStringBuffer[10]);
   ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors
exit status 1
'%s' directive writing up to 19 bytes into a region of size 14 [-Werror=format-overflow=]
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-24:
@ArminJo - I can't rerproduce the issue.

I was able to successfully compile the proposed sketch and upload the firmware to an ESP32-DevKit board.
The output is:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
Acc.X 123.5
```

SuGlider, 2022-07-24:
My suggestion is to remove the ESP32 board from the IDE and reinstall it again.

ArminJo, 2022-07-24:
Du you set compiler warning level to **error**???
And in the log in the link, the ESP board is installed freshly ad hoc!

SuGlider, 2022-07-24:
There is no way to set Compiler warning to error in the Arduino IDE.
Anyway, there is no warning message from the compiler when compiling this sketch.

ArminJo, 2022-07-24:
Sorry I meant ALL like in the topic
[image]
.

ArminJo, 2022-07-24:
Use File > Preferences

SuGlider, 2022-07-24:
Ok, thanks, indeed, with this setup I see the error message.

ArminJo, 2022-07-24:
Great 👍

SuGlider, 2022-07-24:
I guess you are reading some IMU data for its Axis.
This issue can be solved by an equivalent code using `sprintf()`.
It compiles with the preferences suggested setup and runs fine.

```
  //dtostrf(123.45, 4, 1, &sStringBuffer[10]);
  //sprintf(sStringBuffer, "Acc.X %s", &sStringBuffer[10]);
  float ff = 123.46;    // 123.45 results in "123.4" and 123.46 results in "123.5"
  sprintf(sStringBuffer, "Acc.X %4.1f", ff);
  Serial.println(sStringBuffer);

```

SuGlider, 2022-07-24:
For testing you can directly use:
`Serial.printf("Acc.X %4.1f", 123.45);`

ArminJo, 2022-07-24:
Thanks,
I know how to program and how to avoid this compiler bug.
But all compilers for all other platforms know how to compile this program **without errors** except the ESP32 compiler.

SuGlider, 2022-07-24:
@VojtechBartoska - Could you please address this issue with the Espressif Compiler team - if there is any - not sure.

igrr, 2022-07-25:
Hi folks, as far as I can tell, sufficiently new versions of GCC produce the same error for other architectures.
You can see the result here: https://godbolt.org/z/6e9MP4Mff. The panel on the right shows that the code fails to build for a few other architectures, with GCC version 8 and above. It compiles successfully with GCC 5.

The difference between the Arduino cores might be in:
* compiler version used
* warning flags enabled by each core

Indeed, esp32-arduino core adds `-Werror=all` to `-Wall` flag used by other cores.
Compare:
* arduino-esp32: https://github.com/espressif/arduino-esp32/blob/5f427c998a5eff55abf163a4fbeab17fff06e348/platform.txt#L105-L110
* esp8266: https://github.com/esp8266/Arduino/blob/e149829802f351029888c42ab5ce12c0353c42ad/tools/warnings/more-gcc#L1
* stm32: https://github.com/stm32duino/Arduino_Core_STM32/blob/46297e20214403769f96358368e9fce596d8fcfb/platform.txt#L14-L18

So for other cores, only a warning is reported when a sufficiently new version of the compiler is used, while for arduino-esp32 the warning is also treated as an error.

The reason why this didn't occur previously is that compiler.warning_flags options were not added to the final compiler flags. This was done in https://github.com/espressif/arduino-esp32/pull/6596.

`-Werror` was added to the flags in commit https://github.com/espressif/arduino-esp32/commit/f6d48436806f4182c49ffbc8286f8ae95f5ded2a, however I haven't found the explanation why this was done.

ArminJo, 2022-07-25:
Hi Ivan,
then at least ESP should add -Wno-error=format-overflow  as I have done [here ](https://github.com/ArminJo/Arduino-BlueDisplay/blob/648a11c0700a18e87c0d6ee1ba8365d63ec0c522/.github/workflows/LibraryBuild.yml#L75) to avoid this nasty bug,

igrr, 2022-07-25:
I'm not sure of the exact way to resolve the issue, it's up to the arduino-esp32 team to decide.

Putting my C developer hat on, I'd say that erroring out on format warnings (as most other -Wall warnings) is in general a good idea. Compilers do often report issues which, if unnoticed, cause the program to work incorrectly. That said, there are two things against this:
1) given that there are plenty of other arduino cores out there, arduino-esp32 should try not to deviate much from the norm, if only to make it easier for library developers to write code compatible with multiple cores
2) in this specific case, the warning is a false positive; however the compiler doesn't know about the semantics of the non-standard `dtostrf` function to understand this.

ArminJo, 2022-07-25:
If you replace the unknown
`  dtostrf(123.45, 4, 1, &sStringBuffer[10]);`
by the known
`  sprintf(&sStringBuffer[10], "123.4567");`
you get the same error 😭

SuGlider, 2022-07-29:
I thinks we must remove `-Werror=all` from the `cflags`

https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L105-L110
Should look like this:

```
# Arduino Compile Warning Levels
compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra
```

This way using "Compiler warning: all" will work fine with the issue pointed here.

SuGlider, 2022-07-29:
@ArminJo - please let me know if you are OK with the solution in the PR

ArminJo, 2022-07-29:
@SuGlider - I tested it and now I see it as a false warning, but this is OK for me.
Thank you very much for your responsiveness and for fixing it! 🥇

SuGlider, 2022-08-08:
@ArminJo - you can also use `platform.local.txt` to override some of the Warning compiling settings from `platform.txt`
This file can be created locally to your Arduino Core folder (from github or the one from Arduino15).

Check #7090 for more information.

SuGlider, 2022-08-08:
> Thanks for the hint, but for a running github action this is not a real workaround.

Check #7090 --- it is exactly intended for CI - there are changes done to the scripts that copy `platform.local.txt` from `/tools` into the docker files system after the runner execute `git clone "Arduino-ESP32.git"` and override/replace the `-Wnoxxxxx` when running those Github Actions.

ArminJo, 2022-08-08:
Thanks, but there is no git clone "Arduino-ESP32.git" in the script https://github.com/ArminJo/Arduino-BlueDisplay/actions/runs/2714739545/workflow.
Sorry but ESP32 is not the only attraction in town.

SuGlider, 2022-08-08:
OK, I see that it has been locally solved in your GitHub Actions + PlatformIO:
https://github.com/ArminJo/Arduino-BlueDisplay/blob/master/.github/workflows/LibraryBuild.yml#L75

```Compiling ShowSensorValues with -Wno-error=format-overflow 
arduino-cli compile --verbose --warnings all --fqbn esp32:esp32:featheresp32:FlashFreq=80  --build-property compiler.cpp.extra_flags="-Wno-error=format-overflow" --build-property compiler.c.extra_flags="-Wno-error=format-overflow" --build-property compiler.S.extra_flags="-Wno-error=format-overflow"  ./examples/ShowSensorValues
```

ArminJo, 2022-10-12:
> 2\. in this specific case, the warning is a false positive; however the compiler doesn't know about the semantics of the non-standard `dtostrf` function to understand this.

No I have the next problem with `-Werror=all`.
This gives an error and there is no unknown semantics, just an ESP32 provided macro!!!
```
#include "esp32-hal-log.h"

void setup() {
  // put your setup code here, to run once:
    Serial.begin(9600);                                     // Device to serial monitor feedback
}
void loop() {
        uint8_t eject = Serial.read();
        ESP_LOGW("Bug", "unused-variable???: %d", eject);
}
```

```
ESP32Bug:10:17: error: unused variable 'eject' [-Werror=unused-variable]
         uint8_t eject = Serial.read();
                 ^~~~~
cc1plus.exe: some warnings being treated as errors
```
**so please remove the -Werror=all from platform.txt !!!**

mrengineer7777, 2022-10-12:
It looks like #7060 hasn't been merged yet.  They are targeting it for 2.0.6 release.

VojtechBartoska, 2022-10-12:
> It looks like #7060 hasn't been merged yet. They are targeting it for 2.0.6 release.

Yes, still in progress and will be released in 2.0.6

ArminJo, 2022-10-12:
Thanks for the info 👍

VojtechBartoska, 2024-02-20:
Seems this issue can be closed as solved, if needed you can reopen.

ArminJo, 2024-02-20:
If I compile 2.0.14 without `-Wno-error=format-overflow` I still get the errors 😢 !
See https://github.com/ArminJo/Arduino-BlueDisplay/actions/runs/7975089958/job/21772457525#step:3:420
And I canno reopen the issue, only comment it.

igrr, 2024-02-20:
Reopened as the linked PR hasn't been merged.

VojtechBartoska, 2024-02-20:
I see, keeping it in Backlog.

me-no-dev, 2024-02-20:
@ArminJo https://github.com/ArminJo/Arduino-BlueDisplay/actions/runs/7975089958/job/21772457525#step:3:303 you have enabled all warnings, therefore I see no issue here?

ArminJo, 2024-02-21:
@me-no-dev Strange definition of an issue....
And keep in mind that all other cores do not have this problem.

me-no-dev, 2024-02-21:
@ArminJo as far as I see, the issue comes from lines like `sprintf_P(sStringBuffer, PSTR("Yaw   %s"), &sStringBuffer[10]);` and the error reads `error: '%s' directive writing up to 19 bytes into a region of size 14 [-Werror=format-overflow=]`. Now I see where the compiler gets confused, but how is that in any way fault of the flags or the core?

This issue report is about `-Werror=all`, but your problem is totally different. You can either use another buffer to store the `dtostrf` value or file an issue report in the toolchain repository to see why counts the source size as 19, instead of 9.

Does that make sense?

igrr, 2024-02-21:
> or file an issue report in the toolchain repository to see why counts the source size as 19, instead of 9.

I don't think this is necessary, as the behavior is exactly the same for GCC on other platforms. Here's an example with a really old ARM GCC 8.3.1: https://godbolt.org/z/6bcEvMPnf

ArminJo, 2024-02-21:
@me-no-dev It makes no sense to change a correct code which compiles everywhere without errors, except on ESP32.
ESP32 is not the center of the world.
This is my last comment to it!

igrr, 2024-02-21:
I think we need to compare the compiler arguments used by other Arduino cores with the ones we emit. Based on the godbolt link above, it doesn't look like a compiler-related difference. Probably we have a different subset of warnings enabled?

I do agree that the inconsistent warnings/errors make the work of library maintainers harder, we should try to produce the behavior similar to other Arduino cores if possible.

igrr, 2024-02-21:
@me-no-dev looking at 2.0.14, we do enable -Werror=all for "More" and "All" warning options

https://github.com/espressif/arduino-esp32/blob/44da992b774f76777bb2e931dd76cfcf12b9fe70/platform.txt#L125-L126

Whereas other cores don't do this:

- [esp8266](https://github.com/esp8266/Arduino/blob/master/tools/warnings/more-cflags)
- [stm32](https://github.com/stm32duino/Arduino_Core_STM32/blob/0ca1f4ca3d5936cf10855d2cde37e44735425edf/platform.txt#L17-L18)

The point of https://github.com/espressif/arduino-esp32/pull/7060 which was closed without merging was to only add -Werror=all in CI builds, and keep the user-facing warnings behavior same as for the other cores.

me-no-dev, 2024-02-21:
Yup. I have approached this problem from a totally wrong direction up until now. On it...

closed 2024-02-21 by me-no-dev