espressif/arduino-esp32 #7026: esp_log_level_set not enabling library error output

lonerzzz, 2022-07-22:
### Board

Custom prototype board

### Device Description

ESP-WROOM-32

### Hardware Configuration

Not relevant to issue

### Version

v2.0.4

### IDE Name

Arduino 1.8.19

### Operating System

WIndows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am encountering I2C issues after upgrading and I wish to enable debugging of the hal I2C library to get more detail. I enabled the log and attempted to print via the following, but get no error output from the hal i2c library or via the initial logging attempt. No errors are raised in setting the log output command so I would expect output to be generated.

`  esp_log_level_set("*", ESP_LOG_ERROR);

  log_e("Hello");
`

I can print to the serial port via Serial.print commands. Not sure what else is required. I am not sure whether the issue is insufficient documentation or a bug. I have even tried setting a define prior to the esp_log.h include with no luck.

#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE

### Sketch

```cpp
I will do so if needed.
```

### Debug Message

```plain
No debug output is being generated. That is the issue.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-22:
`esp_log_level_set()` only works in an IDF development environment. It has no effect in Arduino environment.

`log_e()` or any other `log_X()` only works with Arduino development environment. 
The Log Level in Arduino is set before compiling and uploading the application, using:
- Arduino IDE Menu --> tools --> Core Debug Level --> Error

SuGlider, 2022-07-22:
[image]

lonerzzz, 2022-07-22:
Thanks for the response. Is there a spot where this is documented? Multiple web search efforts failed to reveal this. It might also be good to augment/change the name of this option in the menu to "Logging Level" or "Log Level" because that is what people will search for for most software applications.

VojtechBartoska, 2022-07-25:
> Thanks for the response. Is there a spot where this is documented? Multiple web search efforts failed to reveal this. It might also be good to augment/change the name of this option in the menu to [...]

some documentationis availlable [here](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/guides/tools_menu.html?highlight=debug#core-debug-level)

closed 2022-07-28 by SuGlider