espressif/arduino-esp32 #7028: When integrating Wifi and Bluetooth, USB COM3 is not working

DreamPWJ, 2022-07-22:
### Board

esp32-c3-devkitm-1

### Device Description

no

### Hardware Configuration

no

### Version

v2.0.3

### IDE Name

PlatformIO 、   Clion

### Operating System

Windows11

### Flash frequency

160

### PSRAM enabled

no

### Upload speed

921600

### Description

  The program is stuck at this line of code because the USB is not working 。  WiFi.mode(WIFI_STA);
[image]

### Sketch

```cpp
no
```

### Debug Message

```plain
no
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

DreamPWJ, 2022-07-23:
I'm very much wondering what the solution is now, as this problem project is stuck for three days. @SuGlider

Jason2866, 2022-07-23:
Whats wrong? the message can be seen from serial print?
We (Project Tasmota) have no issues with serial console and wifi BLE when the workaround is used.

DreamPWJ, 2022-07-25:
The message is printed to this line and it is not running,  View firmware error 0xffffffff

DreamPWJ, 2022-07-25:
I suspect that the firmware is damaged. When the wifi is integrated, it is normal when it is not inherited, but I don't know how to troubleshoot

VojtechBartoska, 2022-07-25:
@DreamPWJ Can you please retest your issue on v[2.0.4](https://github.com/espressif/arduino-esp32/releases/tag/2.0.4)?

DreamPWJ, 2022-07-25:
PlatformIO How to set the version number?

Jason2866, 2022-07-25:
To use latest core 2.0.4 from platformio you have to use the development github version. The 2.0.4. is not yet available as release
```
platform = https://github.com/platformio/platform-espressif32.git
```

DreamPWJ, 2022-07-26:
@Jason2866  2.0.4 It still doesn't work. I found that after integrating the wifi module, Serial cannot be connected and fails. Serial is normal when it is not integrated.

Jason2866, 2022-07-26:
It works with Tasmota. You can install with our [WebInstaller](https://tasmota.github.io/install/) select Tasmota32 Bluetooth (english) and ESP32-C3

DreamPWJ, 2022-07-28:
The usb baud rate is set to 115200, the serial port log does not print@Jason2866

Jason2866, 2022-07-28:
Serial Console (with wifi & BLE on):

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x48
load:0x403ce000,len:0x64c
load:0x403d0000,len:0x218c
SHA-256 comparison failed:
Calculated: b402e8900cbac1bc7b413e37e6da52a9e8a32f3532d3be831488ba8913824e71
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000

00:00:00.001-221/45 HDW: ESP32-C3 
00:00:00.018-219/45 UFS: FlashFS mounted with 296 kB free
00:00:00.038 CFG: Loaded from File, Count 10912
00:00:00.098 BRY: Berry initialized, RAM used=4215 bytes
00:00:00.110 Project tasmota - Tasmota Version 12.0.2.4(tasmota)-2_0_4(2022-07-26T08:37:42)
00:00:00.112 iBeacon register for advert callbacks
00:00:00.122 MI32Option0: 1
00:00:00.123 MI32Option1: 0
00:00:00.123 MI32Option2: 0
00:00:00.123 MI32Option4: 1
00:00:00.123 MI32Option5: 0
00:00:00.134 MI32Option6: 0
00:00:00.134 M32: init: request callbacks
00:00:00.572 WIF: Connecting to AP1 Jason_Home_WLAN Channel 1 BSSId 00:09:4F:70:72:BB in mode 11n as sonoff-3B0814-2068...
00:00:03.013 WIF: Connected
00:00:03.270 HTP: Web server active on sonoff-3B0814-2068 with IP address 192.168.2.168
16:56:36.268 MQT: Attempting connection...
16:56:36.305 MQT: Connected
16:56:36.308 MQT: tele/sonoff-3B0814/LWT = Online (retained)
16:56:36.310 MQT: cmnd/sonoff-3B0814/POWER = 
16:56:36.314 MQT: tele/sonoff-3B0814/INFO1 = {"Info1":{"Module":"ESP32C3","Version":"12.0.2.4(tasmota)","FallbackTopic":"cmnd/sonoff-3B0814_fb/","GroupTopic":"cmnd/sonoffs/"}}
16:56:36.328 MQT: tele/sonoff-3B0814/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff-3B0814-2068","IPAddress":"192.168.2.168"}}
16:56:36.341 MQT: tele/sonoff-3B0814/INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":27}}
16:56:40.903 MQT: tele/sonoff-3B0814/STATE = {"Time":"2022-07-28T16:56:40","Uptime":"0T00:00:10","UptimeSec":10,"Heap":171,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":59,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:09:4F:70:72:BB","Channel":1,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:56:40.939 MQT: tele/sonoff-3B0814/SENSOR = {"Time":"2022-07-28T16:56:40","ESP32":{"Temperature":31.0},"TempUnit":"C"}
16:56:40.945 MQT: tele/sonoff-3B0814/BLE = {"Time":"2022-07-28T16:56:40","BLEDevices":{"total":0}}
16:56:40.959 MQT: tele/sonoff-3B0814/BLE = {"Time":"2022-07-28T16:56:40","BLE":{"scans":0,"adverts":0,"devices":0,"resets":0}}
17:01:40.902 MQT: tele/sonoff-3B0814/STATE = {"Time":"2022-07-28T17:01:40","Uptime":"0T00:05:10","UptimeSec":310,"Heap":168,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:09:4F:70:72:BB","Channel":1,"Mode":"11n","RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:01:40.938 MQT: tele/sonoff-3B0814/SENSOR = {"Time":"2022-07-28T17:01:40","ESP32":{"Temperature":31.0},"TempUnit":"C"}
17:01:40.944 MQT: tele/sonoff-3B0814/BLE = {"Time":"2022-07-28T17:01:40","BLEDevices":{"total":0}}
17:01:40.959 MQT: tele/sonoff-3B0814/BLE = {"Time":"2022-07-28T17:01:40","BLE":{"scans":0,"adverts":0,"devices":0,"resets":0}}
```

DreamPWJ, 2022-07-28:
What tool does this use to print the log? But mine doesn't print, is it related to the usb cable?

DreamPWJ, 2022-07-29:
E (30939) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (30939) BT_SMP: smp_derive_link_key_from_long_term_key failed

DreamPWJ, 2022-09-21:
https://github.com/platformio/platform-espressif32.git       has 2.0.5 ?  @VojtechBartoska

VojtechBartoska, 2022-09-23:
Hello @DreamPWJ, I think it's not supported yet in PlatformIO, we just released recently so PIO guys will need some time to add support.

Jason2866, 2022-09-24:
@DreamPWJ you can use with
```
platform = https://github.com/platformio/platform-espressif32.git
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.5/esp32-2.0.5.zip
```

closed 2022-12-27 by DreamPWJ