espressif/arduino-esp32 #7029: ESP can't connect to network: E (477) esp_eth: esp_eth_ioctl(348): ethernet driver handle can't be null

MetriciRO, 2022-07-22:
### Board

ESP32-EVB Olimex

### Device Description

ESP32-EVB Olimex with ESP32-WROOM-32UE

espressif32 @ 3.4.0
PlatformIO Core:  6.1.3

### Hardware Configuration

Hardware configuration per ESP32-EVB Olimex datasheet: 
[product page](https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/open-source-hardware)
[download schematics pdf](https://github.com/OLIMEX/ESP32-EVB/raw/master/HARDWARE/REV-I/ESP32-EVB_Rev_I.pdf)

### Version

v1.0.6

### IDE Name

PlatformioIDE

### Operating System

Windows 11

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I use the same code for all my ESP32, but one of them can't connect to the switch and has the following error:  `E (477) esp_eth: esp_eth_ioctl(348): ethernet driver handle can't be null`

I don't understand why it can't connect since the event: `_eventCallback(): Arduino Event: 22 - ETH_GOT_IP` gets triggered

The boards are all connected to 5V.

I have tried different IPs, different patches, different everything.

I think it's a strange behavior because in terminal the Verbose and Debug messages output `_eventCallback(): Arduino Event: 20 - ETH_CONNECTED` and `_eventCallback(): Arduino Event: 22 - ETH_GOT_IP` , but, as you can see in the code, I have cases for those particular events which don't get triggered:
```
    case SYSTEM_EVENT_ETH_CONNECTED:
        Serial.println(F("ETH Connected"));
        break;
    case SYSTEM_EVENT_ETH_GOT_IP:
        eth_connected = true;
        Serial.print(F("ETH MAC - "));
        Serial.print(ETH.macAddress());
        Serial.print(F(", IPv4: "));
        Serial.print(ETH.localIP());
        if (ETH.fullDuplex())
        {
            Serial.print(F(", FULL_DUPLEX"));
        }
        Serial.print(F(", "));
        Serial.print(ETH.linkSpeed());
        Serial.println(F("Mbps"));
        break;
```

So if `SYSTEM_EVENT_ETH_CONNECTED` gets triggered I should see `Serial.println(F("ETH Connected"));`.

If `SYSTEM_EVENT_ETH_GOT_IP` gets triggered I should see the MAC and IP on serial.

Any thoughts ?

### Sketch

```cpp
#include <network_connection.h>

static bool eth_connected = false;

IPAddress ip_address;
IPAddress gateway;
IPAddress subnet;
IPAddress dns;

void WiFiEvent(WiFiEvent_t event)
{
    switch (event)
    {
    case SYSTEM_EVENT_ETH_START:
        Serial.println(F("ETH Started"));
        // set eth hostname here
        if (!ETH.setHostname("Metrici-Radar-Eth"))
        {
            Serial.println(F("Ethernet hostname failed to configure"));
        }
        break;
    case SYSTEM_EVENT_ETH_CONNECTED:
        Serial.println(F("ETH Connected"));
        break;
    case SYSTEM_EVENT_ETH_GOT_IP:
        eth_connected = true;
        Serial.print(F("ETH MAC - "));
        Serial.print(ETH.macAddress());
        Serial.print(F(", IPv4: "));
        Serial.print(ETH.localIP());
        if (ETH.fullDuplex())
        {
            Serial.print(F(", FULL_DUPLEX"));
        }
        Serial.print(F(", "));
        Serial.print(ETH.linkSpeed());
        Serial.println(F("Mbps"));
        break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
        Serial.println(F("ETH Disconnected"));
        eth_connected = false;
        break;
    case SYSTEM_EVENT_ETH_STOP:
        Serial.println(F("ETH Stopped"));
        eth_connected = false;
        break;
    default:
        break;
    }
}

void ethConnection()
{
    ETH.begin();
    delay(50);

    if (network_settings.ip_type == "Static")
    {
        ip_address.fromString(network_settings.ip_address);
        gateway.fromString(network_settings.gateway);
        subnet.fromString(network_settings.subnet);
        dns.fromString(network_settings.dns);

        if (!ETH.config(ip_address, gateway, subnet, dns))
        {
            logOutput("WARNING: Couldn't configure STATIC IP ! Obtaining DHCP IP !");
        }
        delay(50);
    }

    int k = 0;
    while (!eth_connected && k < 20)
    {
        k++;
        delay(1000);
        Serial.println((String) "[" + k + "] - Establishing ETHERNET Connection ... ");
    }

    if (!eth_connected)
    {
        logOutput("(1) Could not connect to network ! Trying again...");
        logOutput("Radar will restart in 5 seconds !");
        restartSequence(5);
    }

    network_settings.ip_address = ETH.localIP().toString();
    network_settings.gateway = ETH.gatewayIP().toString();
    network_settings.subnet = ETH.subnetMask().toString();
    network_settings.dns = ETH.dnsIP().toString();
    // logOutput((String) "Hostname: " + ETH.getHostname());
    logOutput((String) "IP address: " + network_settings.ip_address);
    logOutput((String) "Gateway: " + network_settings.gateway);
    logOutput((String) "Subnet: " + network_settings.subnet);
    logOutput((String) "DNS: " + network_settings.dns);
}

void startConnection()
{
    if (!WiFi.mode(WIFI_STA))
    {
        logOutput("ERROR: Radar couldn't go in STA_MODE. Restarting in 5 seconds.");
        restartSequence(5);
        return;
    }

    Serial.println((String) "WiFi.getMode() [1 = STA / 2 = AP] : " + WiFi.getMode());

    WiFi.onEvent(WiFiEvent);

    if (network_settings.connection == "Ethernet")
        ethConnection();

    changed_network_config = false;
}
```

### Debug Message

```plain
[     2][V][WiFiServer.h:42] WiFiServer(): WiFiServ�FҺ��M��ٕɡport=10001, ...)
[     6][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (477) esp_eth: esp_eth_ioctl(348): ethernet driver handle can't be null
[   596][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
WiFi.getMode() [1 = STA / 2 = AP] : 1
[   689][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[   690][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[  4718][V][WiFiGeneric.cpp:363] _arduino_event_cb(): Ethernet Started
[  4718][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 18 - ETH_START
ETH MAC - 34:AB:95:40:52:E3, IPv4: 10.10.103.20[1] - Establishing ETHERNET Connection ... 
[2] - Establishing ETHERNET Connection ... 
[3] - Establishing ETHERNET Connection ... 
[4] - Establishing ETHERNET Connection ... 
[5] - Establishing ETHERNET Connection ... 
[6] - Establishing ETHERNET Connection ... 
[7] - Establishing ETHERNET Connection ... 
[ 12719][V][WiFiGeneric.cpp:355] _arduino_event_cb(): Ethernet Link Up
[ 12719][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
[ 12720][V][WiFiGeneric.cpp:370] _arduino_event_cb(): Ethernet got newip:10.10.103.20
[ 12729][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[ 12736][D][WiFiGeneric.cpp:950] _eventCallback(): ETH IP: 10.10.103.20, MASK: 255.255.255.0, GW: 10.10.103.1
[8] - Establishing ETHERNET Connection ... 
[9] - Establishing ETHERNET Connection ... 
[10] - Establishing ETHERNET Connection ... 
[11] - Establishing ETHERNET Connection ... 
[12] - Establishing ETHERNET Connection ... 
[13] - Establishing ETHERNET Connection ... 
[14] - Establishing ETHERNET Connection ... 
[15] - Establishing ETHERNET Connection ... 
[16] - Establishing ETHERNET Connection ... 
[17] - Establishing ETHERNET Connection ... 
[18] - Establishing ETHERNET Connection ... 
[19] - Establishing ETHERNET Connection ... 
[20] - Establishing ETHERNET Connection ... 
(1) Could not connect to network ! Trying again...
Radar will restart in 5 seconds !
Restarting in: 5 seconds.
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-07-22:
Mayne related to https://github.com/espressif/arduino-esp32/pull/6188

VojtechBartoska, 2022-07-25:
Maybe, do you face the same issue on Arduino core version 2.0.4?

MetriciRO, 2022-07-25:
> Maybe, do you face the same issue on Arduino core version 2.0.4?

I haven't checked yet.

MetriciRO, 2022-07-25:
> Maybe related to #6188

I have added a delay just in case:
```cpp
    ETH.begin();
    delay(500);
```

When using `platform = espressif32@3.4.0` everything works.

When using `platform = espressif32` which should be v5.0.0, I get the following output on the terminal:
```
[     2][V][WiFiServer.h:42] WiFiServer(): WiFiServ��Һ��M��ٕɡport=10001, ...)
[     6][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (463) esp_eth: esp_eth_ioctl(348): ethernet driver handle can't be null
[   569][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
WiFi.getMode() [1 = STA / 2 = AP] : 1
[   662][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[   663][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[  2391][V][WiFiGeneric.cpp:363] _arduino_event_cb(): Ethernet Started
[  2391][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 18 - ETH_START
[  2394][V][WiFiGeneric.cpp:355] _arduino_event_cb(): Ethernet Link Up
[  2400][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
[  2946][V][WiFiGeneric.cpp:370] _arduino_event_cb(): Ethernet got newip:192.168.100.10
[  2946][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[  2950][D][WiFiGeneric.cpp:950] _eventCallback(): ETH IP: 192.168.100.10, MASK: 255.255.255.0, GW: 192.168.100.1
[1] - Establishing ETHERNET Connection ... 
[2] - Establishing ETHERNET Connection ... 
[3] - Establishing ETHERNET Connection ... 
[4] - Establishing ETHERNET Connection ... 
[5] - Establishing ETHERNET Connection ... 
[6] - Establishing ETHERNET Connection ... 
[7] - Establishing ETHERNET Connection ... 
[8] - Establishing ETHERNET Connection ...
[9] - Establishing ETHERNET Connection ...
[10] - Establishing ETHERNET Connection ...
[11] - Establishing ETHERNET Connection ...
[12] - Establishing ETHERNET Connection ...
[13] - Establishing ETHERNET Connection ...
[14] - Establishing ETHERNET Connection ...
[15] - Establishing ETHERNET Connection ...
[16] - Establishing ETHERNET Connection ...
[17] - Establishing ETHERNET Connection ...
[18] - Establishing ETHERNET Connection ...
[19] - Establishing ETHERNET Connection ...
[20] - Establishing ETHERNET Connection ...
(1) Could not connect to network ! Trying again...
Radar will restart in 5 seconds !
Restarting in: 5 seconds.
```

From the Debug and Verbose message I should be getting a valid connection, yet `SYSTEM_EVENT_ETH_GOT_IP` event never gets triggered.

I think something changed between Arduino versions which broke my WiFiEvent() function:
```cpp
void WiFiEvent(WiFiEvent_t event)
{
    switch (event)
    {
    case SYSTEM_EVENT_ETH_START:
        Serial.println(F("ETH Started"));
        // set eth hostname here
        if (!ETH.setHostname("Metrici-Radar-Eth"))
        {
            Serial.println(F("Ethernet hostname failed to configure"));
        }
        break;
    case SYSTEM_EVENT_ETH_CONNECTED:
        Serial.println(F("ETH Connected"));
        break;
    case SYSTEM_EVENT_ETH_GOT_IP:
        eth_connected = true;
        Serial.print(F("ETH MAC - "));
        Serial.print(ETH.macAddress());
        Serial.print(F(", IPv4: "));
        Serial.print(ETH.localIP());
        if (ETH.fullDuplex())
        {
            Serial.print(F(", FULL_DUPLEX"));
        }
        Serial.print(F(", "));
        Serial.print(ETH.linkSpeed());
        Serial.println(F("Mbps"));
        break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
        Serial.println(F("ETH Disconnected"));
        eth_connected = false;
        break;
    case SYSTEM_EVENT_ETH_STOP:
        Serial.println(F("ETH Stopped"));
        eth_connected = false;
        break;
    default:
        break;
    }
}
```

GeorgeFlorian, 2022-07-26:
> > Maybe, do you face the same issue on Arduino core version 2.0.4?
> 
> I am checking.

Actually I am getting this bug when using v.2 or above of Arduino Core.

When using v1.0.6 the `void WiFiEvent(WiFiEvent_t event)` function works, when using v2.0.3 the same function does do its job and the Ethernet doesn't work.

MetriciRO, 2022-07-26:
> Maybe, do you face the same issue on Arduino core version 2.0.4?

Actually I am getting this bug when using v.2 or above of Arduino Core.

When using v1.0.6 the `void WiFiEvent(WiFiEvent_t event)` function works, when using v2.0.3 the same function doesn't do its job and the Ethernet doesn't work.

felmue, 2022-07-26:
Hello @MetriciRO 

`SYSTEM_EVENT_` events have been renamed to `ARDUINO_EVENT_`. Maybe that is your issue?

Thanks
Felix

VojtechBartoska, 2022-07-26:
Is this still valid under v2.0.4?

felmue, 2022-07-26:
Hello @VojtechBartoska 

I'd say so unless the Ethernet [example](https://github.com/espressif/arduino-esp32/blob/2.0.4/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino) is incorrect. See here.

Thanks
Felix

VojtechBartoska, 2022-07-27:
@PilnyTomas Please help with this issue. Thanks

PilnyTomas, 2022-07-29:
I have come to the same conclusion as felmue...
@MetriciRO please change `SYSTEM_EVENT...` to `ARDUINO_EVENT...` and let us know if that helped.

closed 2022-08-11 by PilnyTomas

MetriciRO, 2022-08-18:
> I have come to the same conclusion as felmue... @MetriciRO please change `SYSTEM_EVENT...` to `ARDUINO_EVENT...` and let us know if that helped.

I am sorry for the late reply.
Changing  `SYSTEM_EVENT...` to `ARDUINO_EVENT...` indeed works.

PilnyTomas, 2022-08-19:
> Changing `SYSTEM_EVENT...` to `ARDUINO_EVENT...` indeed works.

Good to hear that!