espressif/arduino-esp32 #7031: Guru Meditation Error when attempting to use WiFi

badmotorfinger, 2022-07-24:
### Board

ESP32 Wrover 4MB 

### Device Description

ESP32 Chip model = ESP32-D0WDQ6 Rev 1
This chip has 2 cores
Chip ID: 14129192

### Hardware Configuration

Nothing is connected, just the board is connected to my PC via USB 2

### Version

v2.0.3

### IDE Name

Arduino IDE 1.8.19

### Operating System

Linux Mint 18

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm having an issue when trying to do anything whatsoever with WiFi. The device just keeps rebooting, so I decided to just print out some information using `WiFi.printDiag(Serial);` which is when I discovered what might be the problem.

Any idea on what my WiFi is not working correctly? I have tried different ESP32 Wrover boards which are all the same revision and unopened (Brand new from a batch of them I purchased). Same result.

### Sketch

```cpp
#include <WiFi.h>

void setup() {
  
  Serial.begin(115200);
  Serial.println("Booting");
  delay(1000);
  WiFi.printDiag(Serial);
}

void loop() { }
```

### Debug Message

```plain
Mode: Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dcb30  PS      : 0x00060a30  A0      : 0x800d5837  A1      : 0x3ffb25f0  
A2      : 0x3ffbdbb4  A3      : 0x400859f0  A4      : 0x400859f0  A5      : 0x3ffb2644  
A6      : 0x00060623  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000003  
A10     : 0x60000000  A11     : 0x3ff40000  A12     : 0x3ffc3f34  A13     : 0x00060a23  
A14     : 0x00060a20  A15     : 0x00000001  SAR     : 0x00000003  EXCCAUSE: 0x00000003  
EXCVADDR: 0x400859f0  LBEG    : 0x4008a9f1  LEND    : 0x4008aa01  LCOUNT  : 0xfffffffe  

Backtrace:0x400dcb2d:0x3ffb25f00x400d5834:0x3ffb2620 0x400d5f1d:0x3ffb2670 0x400d43ec:0x3ffb26a0 0x400d35a1:0x3ffb26c0 0x400d3a3f:0x3ffb26e0 0x400d3a79:0x3ffb2700 0x400d231c:0x3ffb2720 0x400d2231:0x3ffb27f0 0x400d3ece:0x3ffb2820
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-24:
You likely need to set a mode before calling printDiag, in order to initialize the object.  In any case, calling WiFi.printDiag(Serial) is redundant.  It will by default be printing diag out to Serial.

badmotorfinger, 2022-07-24:
@lbernstone - Code issues aside, any idea what that mode might be?

badmotorfinger, 2022-07-24:
I've tried all supported modes, still the same issue.

lbernstone, 2022-07-24:
No, it is possible this is a bug in printDiag.  I don't have a rig that I can test on here. @SuGlider Take a quick look if this is reproducible?

SuGlider, 2022-07-24:
It's necessary to initialize WiFi with `WiFi.mode()` as below:

```cpp
#include <WiFi.h>

void setup() {
  
  Serial.begin(115200);
  WiFi.mode(WIFI_AP);   // tested with WIFI_STA, WIFI_MODE_APSTA, WIFI_AP
  Serial.println("Booting");
  delay(1000);
  WiFi.printDiag(Serial); // FAILS HERE
}

void loop() { }
```

Output:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
Booting
Mode: AP
Channel: 1
SSID (18): ESP32-Access-Point
Passphrase (0): 
BSSID set: 0

```

badmotorfinger, 2022-07-24:
Code updated to

```
void setup() {
  Serial.begin(115200);
  Serial.println("Set baud to 115200");
  WiFi.mode(WIFI_AP); // Reboots here
  Serial.println("Set mode to AP");
  delay(1000);
  WiFi.printDiag(Serial); // Never gets here
}
```

Output

```
Set baud to 115200
Set baud to 115200
Set baud to 115200
Set baud to 115200
```

Something is very wrong with WiFi. Sorry I haven't been more help in diagnosing this, I'm just starting out with the ESP32 as a beginner.

SuGlider, 2022-07-25:
Nothing wrong with WiFi ... Tested with ESP32-devkit board and Arduino Core 2.0.4. All Fine.

Complete Sketch:
``` cpp
#include "WiFi.h"

void setup() {
  Serial.begin(115200);
  Serial.println("Set baud to 115200");
  WiFi.mode(WIFI_AP); // Reboots here
  Serial.println("Set mode to AP");
  delay(1000);
  WiFi.printDiag(Serial); // Never gets here
}
void loop(){}
```

Output:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
Set baud to 115200
Set mode to AP
Mode: AP
Channel: 1
SSID (6): myWiFi
Passphrase (8): password
BSSID set: 0
```

badmotorfinger, 2022-07-25:
@SuGlider - I appreciate you trying to help me out here, but how is using my code to show me how it works for you meant to help me? It works for you, fantastic!

Does anyone here have anything I can try?

SuGlider, 2022-07-25:
I understand that the Guru Mediation issue is gone, right?

What is the issue now?

lbernstone, 2022-07-25:
Please describe your device better.  Is this a complete dev board you purchased somewhere (include a link), or something you built?

badmotorfinger, 2022-07-25:
@SuGlider - No, the issue is not gone. I posted this already but I will post again.

This code

```
void setup() {
  Serial.begin(115200);
  Serial.println("Set baud to 115200");
  WiFi.mode(WIFI_AP); // Reboots here
  Serial.println("Set mode to AP");
  delay(1000);
  WiFi.printDiag(Serial); // Never gets here
}
```

**On my device**, prints this.

```
Set baud to 115200
Set baud to 115200
Set baud to 115200
Set baud to 115200
```

As you can see, it never makes it to the other `Serial.println` statements. Something is wrong but I don't know what. _I suspect_ it's the same reason  `WiFi.printDiag(Serial);` generates the `Guru Meditation Error`, but again, I'm not sure.

@liebman - I've already posted the hardware specs already as part of this issue. This is where I purchased it https://lonelybinary.com/collections/esp32/products/esp32-wrover-8mb-psram-the-best

SuGlider, 2022-07-25:
Let's look to the Board configuration from the Arduino IDE now.

Could you please send a screenshot of the IDE->Menu->Tools open with your board selection and options?

felmue, 2022-07-25:
Hello guys

could it be the USB power supply / PC USB port is too weak?

Thanks
Felix

lbernstone, 2022-07-25:
We are asking you these questions for a reason- your board should be working.
The most likely issue is a physical problem, especially power.
Getting irritated by people trying to help you is a good way to ensure they stop.
Have you disabled boot logging?  There should be some bootloader logging after each loop, similar to what you see at the start of @SuGlider 's output.  Please set the core debug level to verbose and see if you get any additional information in the output.

SuGlider, 2022-07-25:
> Hello guys
> 
> could it be the USB power supply / PC USB port is too weak?
> 
> Thanks Felix

Yes, it is possible. Today, talking to some colleagues, this possibility was pointed as a potential source for this issue.
The suggestion was to change USB cable used to connect the board to the PC.

@lbernstone @badmotorfinger
Regarding not seen any LOG output, it was pointed that GPIO15 is a strapping pin used to disable (when grounded within boot) the LOG output. Not sure if this is the case.

badmotorfinger, 2022-07-26:
Thank you everyone who took the time to write a message and trying to diagnose this for me. I feel like a complete idiot, it was the cable. I changed cable and WiFi is working now.

While this did fix the problem, does anyone know why I can upload firmware without issue and monitor the serial, but using the WiFi causes the device to reboot?

SuGlider, 2022-07-27:
> While this did fix the problem, does anyone know why I can upload firmware without issue and monitor the serial, but using the WiFi causes the device to reboot?

The cable has 2 functions: data communication (low power, digital signaling) and power supply.
When the WiFi radio is turned on, it pulls a peak of current, thus, I think that the cable had some resistance and could not supply that peak. At this moment, the CPU gets under-powered and resets.

closed 2022-07-27 by badmotorfinger