espressif/arduino-esp32 #7033: V2.0.3 & V2.0.4 SD_MMC 1-bit mode failed

JohnTaipei, 2022-07-24:
### Board

ESP32

### Device Description

esp32-cam

### Hardware Configuration

esp32-cam
micro sd slot

### Version

v2.0.4

### IDE Name

arduino

### Operating System

win10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

hello:
V2.0.3 & V2.0.4  SD_MMC 1-bit mode failed
V1.0.4  SD_MMC 1-bit mode OK

Thanks

### Sketch

```cpp
use example sd_mmc
```

### Debug Message

```plain
Card Mount Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-08-01:
Hi @JohnTaipei,
I have tested SD_MMC example on latest 2.0.4 Arduino-esp32 version and both default 4-bit mode and 1-bit mode works.
To use default 4-bit just run the SD_MMC example.
To use 1-bit mode, change `SD_MMC.begin()` to `SD_MMC.begin("/sdcard", true)`.

It should work for you. Let me know :)

JohnTaipei, 2022-08-03:
@P-R-O-C-H-Y 
Thanks

closed 2022-08-04 by P-R-O-C-H-Y