espressif/arduino-esp32 #7035: upgrade the firmware,found a problem

hzhh110, 2022-07-25:
### Related area

esp32-wroom

### Hardware specification

esp32-wroom

### Is your feature request related to a problem?

Found a problem:
When the SDK version: 3.5, the writer version: 3.6.8, when the storage is 32Mbit, if you choose 64Mbit or 128Mbit, the firmware will be used normally during programming, and the firmware will be upgraded with the Bin file compiled with this SDK version. It is also normal, and then when I use the SDK version: 4.2, the compiled bin file will crash when I upgrade the firmware.

### Describe the solution you'd like

Is this phenomenon unavoidable? Because it has been shipped out in batches, and the old version cannot be tested because of the test, so there are some people who accidentally click 64Mbit and 128Mbit when programming, so the bin file compiled in version 4.2 Cannot be used for firmware upgrade

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-07-25:
Can you post a couple small example sketch .bin's and an exact process that causes the problem.  It will be easier to understand the issue if we can reproduce it.

hzhh110, 2022-07-25:
[image]

hzhh110, 2022-07-25:
esp32-wroom-32d 32Mbit, choose 64Mbit and 128Mbit when programming, the two results are different, correspondingly, the results caused by firmware upgrade are also different, when SDK3.5 is selected incorrectly, the operation is normal, but the firmware upgrade (files compiled by 3.5SDK) are normal, firmware upgrade (files compiled by 4.2SDK) are abnormal

arduino SDK: 3.5 select when programming,
arduino SDK: When 4.2 is programmed, it cannot be used normally

Jason2866, 2022-07-25:
A wrong flash size was choosen not matching the real flash size when the initial flash was done? If this is the case you are stuck on this SDK Version.
With the changes how the bootloader are working between the SDKs that false setting will result in a stuck device.
With SDK4.x the selected flash size has to be the real flash size.
SDK 3.x did sometimes tolerate such a false setting.

JimDrewGH, 2022-07-29:
I have run into this problem for several years when trying to reflash code using PlatformIO over top what was flashed with the Arduino IDE (same project, different setups).  PlatformIO does not include the 3 other files during the flash process (maybe it can, I have just not figured out a way).  This drives me nuts, and the only solution is to make sure that both Arduino IDE and PlatformIO are using the exact same core (SDK) versions.

Jason2866, 2022-07-29:
@JimDrewGH PlatformIO does flash all needed files. Without the ESP32 couldnt boot.
You are right mixing builds based on different SDKs is not a good idea.
Since the versionin scheme from Platformio is completly different (imho no logic in the versioning) it is not always easy.

VojtechBartoska, 2022-08-24:
@hzhh110 Can we close this?

closed 2022-10-06 by hzhh110