espressif/arduino-esp32 #7039: LILYGO® TTGO T7 Mini32 V1.5 ESP32-WROVER-B missing

AlexFundorin, 2022-07-25:
### Related area

Board support

### Hardware specification

LILYGO® TTGO T7 Mini32 V1.5 ESP32-WROVER-B

### Is your feature request related to a problem?

T7 Mini32 V1.5 Introduce

Note:The difference between T7 V1.4 changed to V1.5:

GPIO has replaced the position, we replaced GPIO16 and GPIO17 two pins that can no longer be used for external expansion with GPIO25 and GPIO27 that can be used with defined functions, so that they can be compatible with those of miniD1 The expansion board is not affected by the original GPIO16 and GPIO17 pins.

http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=978
https://aliexpress.ru/item/32977375539.html

### Describe the solution you'd like

Add this board please

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-07-26:
Hello @AlexFundorin,

3rd party boards updates and additions needs to be done by vendors itself. It's not in our scope to update every HW based on Espressif products.

I will gladly ask you to send PR directly instead of an issue.

atkulp, 2023-04-07:
It looks like v1.3 and v1.4 of this board are already included in the package. Is there a reason that v1.5 would be handled differently?

Shaquum, 2023-04-17:
> Hello @AlexFundorin,
> 
> 3rd party boards updates and additions needs to be done by vendors itself. It's not in our scope to update every HW based on Espressif products.
> 
> I will gladly ask you to send PR directly instead of an issue.

closed 2025-01-16 by Parsaabasi