espressif/arduino-esp32 #7041: Need a delay otherwise socket read loop causes wifi to break

torntrousers, 2022-07-26:
### Board

ESP32

### Device Description

M5Stack EduKit

### Hardware Configuration

n/a

### Version

v2.0.4

### IDE Name

Platformio

### Operating System

Windows

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Trying to get the ArduinoBearSSL library to work on an ESP32 and I find it needs a delay in the clientRead function otherwise during the TLS handshake it spins around failing to get any data from the network and after 10 seconds it fails because the Wifi disconnects:
```
. . . 
BearSSLClient::clientRead - 0 - 
BearSSLClient::clientRead - 0 - 
BearSSLClient::clientRead - 0 -
[ 15321][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: BT-P2CKZW, BSSID: ac:84:c9:b7:8f:82, Reason: 200    
BearSSLClient::clientRead xxx- [ 15326][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 15336][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[ 15343][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
0 - [ 15352][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0

[ 15372][D][WiFiClient.cpp:545] connected(): Disconnected: RES: -1, ERR: 113
```
If i add [this delay here](https://github.com/HarringayMakerSpace/ArduinoBearSSL/blob/ec608-edukit/src/BearSSLClient.cpp#L529) then it all works fine.

I thought needing to worry about delays like this was only on the old esp8266 arduino frameowrk and not needed on the esp32? Can you help me understand whats going on?  

### Sketch

```cpp
.
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-29:
@torntrousers - Not sure this is an Arduino issue. 
It would be very helpful if you could create the smallest possible code that demonstrates the issue. Thanks!

VojtechBartoska, 2022-08-23:
Hi @torntrousers, can you provide a bit more info as requested by @SuGlider? Thanks a lot!

VojtechBartoska, 2022-10-10:
I'm closing this issue as expired due to no answer. If needed, you can reopen it. Thanks for understanding.

closed 2022-10-10 by VojtechBartoska

torntrousers, 2022-11-13:
sorry, i completely missed the email on this asking for more