espressif/arduino-esp32 #7061: Python3 calls Xcode

bluewyvern2, 2022-07-29:
### Board

ESP32

### Device Description

38 pin narrow ESP32-WROOM-32 KEEYEE board

### Hardware Configuration

this is a compiler problem haven't gotten to the board yet.

### Version

v2.0.4

### IDE Name

Arduino

### Operating System

MacOS 11.6.8

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

While compiling a test sketch in Arduino IDE for ESP32 I received an error "The "python3" command requires the command line developer tools." This is a part of Apple's Xcode which is a development environment that I do not have and which takes a significant amount of space on a computer. Is there a way around this or do I absolutely have to install this thing? 

My computer is a 2013 MacBook Air and the current versions of Xcode available on the App Store and Apples Developer site will not run on it and I can't seem to find older versions on the Apple developer site. I just want to work with the esp32 not the entire MACOS.

Thank you for any help you can provide. 

### Sketch

```cpp
/*
  ledcWrite_RGB.ino
  Runs through the full 255 color spectrum for an rgb led 
  Demonstrate ledcWrite functionality for driving leds with PWM on ESP32
 
  This example code is in the public domain.
  
  Some basic modifications were made by vseven, mostly commenting.
 */
 
// Set up the rgb led names
uint8_t ledR = 2;
uint8_t ledG = 4;
uint8_t ledB = 5; 

uint8_t ledArray[3] = {1, 2, 3}; // three led channels

const boolean invert = true; // set true if common anode, false if common cathode

uint8_t color = 0;          // a value from 0 to 255 representing the hue
uint32_t R, G, B;           // the Red Green and Blue color components
uint8_t brightness = 255;  // 255 is maximum brightness, but can be changed.  Might need 256 for common anode to fully turn off.

// the setup routine runs once when you press reset:
void setup() 
{            
  Serial.begin(115200);
  delay(10); 
  
  ledcAttachPin(ledR, 1); // assign RGB led pins to channels
  ledcAttachPin(ledG, 2);
  ledcAttachPin(ledB, 3);
  
  // Initialize channels 
  // channels 0-15, resolution 1-16 bits, freq limits depend on resolution
  // ledcSetup(uint8_t channel, uint32_t freq, uint8_t resolution_bits);
  ledcSetup(1, 12000, 8); // 12 kHz PWM, 8-bit resolution
  ledcSetup(2, 12000, 8);
  ledcSetup(3, 12000, 8);
}

// void loop runs over and over again
void loop() 
{
  Serial.println("Send all LEDs a 255 and wait 2 seconds.");
  // If your RGB LED turns off instead of on here you should check if the LED is common anode or cathode.
  // If it doesn't fully turn off and is common anode try using 256.
  ledcWrite(1, 255);
  ledcWrite(2, 255);
  ledcWrite(3, 255);
  delay(2000);
  Serial.println("Send all LEDs a 0 and wait 2 seconds.");
  ledcWrite(1, 0);
  ledcWrite(2, 0);
  ledcWrite(3, 0);
  delay(2000);
 
  Serial.println("Starting color fade loop.");
  
 for (color = 0; color < 255; color++) { // Slew through the color spectrum

  hueToRGB(color, brightness);  // call function to convert hue to RGB

  // write the RGB values to the pins
  ledcWrite(1, R); // write red component to channel 1, etc.
  ledcWrite(2, G);   
  ledcWrite(3, B); 
 
  delay(100); // full cycle of rgb over 256 colors takes 26 seconds
 }
 
}

// Courtesy http://www.instructables.com/id/How-to-Use-an-RGB-LED/?ALLSTEPS
// function to convert a color to its Red, Green, and Blue components.

void hueToRGB(uint8_t hue, uint8_t brightness)
{
    uint16_t scaledHue = (hue * 6);
    uint8_t segment = scaledHue / 256; // segment 0 to 5 around the
                                            // color wheel
    uint16_t segmentOffset =
      scaledHue - (segment * 256); // position within the segment

    uint8_t complement = 0;
    uint16_t prev = (brightness * ( 255 -  segmentOffset)) / 256;
    uint16_t next = (brightness *  segmentOffset) / 256;

    if(invert)
    {
      brightness = 255 - brightness;
      complement = 255;
      prev = 255 - prev;
      next = 255 - next;
    }

    switch(segment ) {
    case 0:      // red
        R = brightness;
        G = next;
        B = complement;
    break;
    case 1:     // yellow
        R = prev;
        G = brightness;
        B = complement;
    break;
    case 2:     // green
        R = complement;
        G = brightness;
        B = next;
    break;
    case 3:    // cyan
        R = complement;
        G = prev;
        B = brightness;
    break;
    case 4:    // blue
        R = next;
        G = complement;
        B = brightness;
    break;
   case 5:      // magenta
    default:
        R = brightness;
        G = complement;
        B = prev;
    break;
    }
}
```

### Debug Message

```plain
xcode-select: note: no developer tools were found at '/Applications/Xcode.app', requesting install. Choose an option in the dialog to download the command line developer tools.
exit status 1
Error compiling for board ESP32-WROOM-DA Module.
```

### Other Steps to Reproduce

To reproduce this error you will need to have a MACOS without Xcode installed. When I clicked the install request no software was found by the Apple search.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-29:
@bluewyvern2 - What is your MacOS version?

SuGlider, 2022-07-29:
Have you tried to use `Homebrew` or `Macports` to install Python 3?

`$ brew install python`

SuGlider, 2022-07-29:
https://docs.python-guide.org/starting/install3/osx/

bluewyvern2, 2022-07-29:
I have installed python3 but not Xcode. MacOS is version 11.6.8 cannot update to 12 and above due to age of computer. 

Bluewyvern2:  Lee Griffis

SuGlider, 2022-07-29:
What happens when you type `python --version` in an terminal?

bluewyvern2, 2022-07-29:
Ah, still says version 2.7. I must have installed it wrong. I'll try again if I can figure out homebrew.

bluewyvern2, 2022-07-29:
I found command line tools for Xcode 13.2 and am downloading them. Homebrew won't install without them. This should also let me run the compiler. I'll let you know when I get that far.

bluewyvern2, 2022-07-29:
Downloading the CLT for Xcode 13.2 seems to have fixed the compiling issue. Thank you for the ideas.

SuGlider, 2022-07-31:
Can we close this issue?

closed 2022-08-23 by VojtechBartoska