espressif/arduino-esp32 #7063: Quad SPI mode

0x0fe, 2022-07-30:
### Related area

adding QSPI mode to the SPI driver

### Hardware specification

ESP32, ESP32C3

### Is your feature request related to a problem?

I could not find any implementation of QSPI mode in the ESP32 core. QSPI is useful in variosu scenario, flash, ADCs/DACs, custom interfaces.

### Describe the solution you'd like

Being able to select the QSPI mode on a SPI interface, as it has been done here, https://github.com/CutClassH/sea_esp32_qspi but this does not work anymore.

### Describe alternatives you've considered

in my case i want to send 3 parallele 8 bit streams, that is, sending 3 separate bistreams on 3 GPIOs synced with the same clock.QSPI seems like the best way to do it at decent speed. An alternative is to bitbang, the last recourse, but bitbanging on ESP32 is not really practical (by experience).

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2022-08-17:
The connected flash (where the sketch is flashed) to the ESP32, C3, S2 can use mode qio. Or do you mean something different?

0x0fe, 2022-08-17:
no, i mean for the SPI driver of arduino core, the "user side" driver. Currently it does not support QSPI (at least on C3).

0x0fe, 2022-08-17:
if you check the initial message i think it was qquite clear though, with example linked

VojtechBartoska, 2023-05-09:
Related request from the discussion: https://github.com/espressif/arduino-esp32/discussions/8174

0x0fe, 2024-02-21:
@VojtechBartoska  OK, well noted

kylongmu, 2024-02-28:
> Postponing this to 3.1.0 Milestone, we need to evaluate this again.

Waiting for the qspi support in arduino too.

xChumimin, 2024-07-16:
I need more documentation, examples and support on both qspi and dual io spi

closed 2025-01-16 by Parsaabasi