espressif/arduino-esp32 #7066: Can libraries wont even reach setup code.

bvernham, 2022-07-30:
### Board

Sparkfun Thingplus and hilego

### Device Description

Sparkfun thing plus

### Hardware Configuration

Board by itself

### Version

v2.0.4

### IDE Name

Arduino

### Operating System

Windows 10 pro

### Flash frequency

80 mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

SparkFun Thing Plus has the same chip as the EVTV board the ESP32-D0WDQ6. I tried the examples (altpins) and but the the board will not even make it to the Setup print statement. I commented out any of the CAN related code and I had no issue.

Any ideas? The core just updated to 2.04.

Thanks

### Sketch

```cpp
Using collin80 and Esp32-can demo
```

### Debug Message

```plain
No output
```

### Other Steps to Reproduce

Tried two board and two can libraries.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-31:
This forum is for issues with the code in this repo.  You will need to ask your 3rd party library author for help. 
> but the the board will not even make it to the Setup print statement
This is not the way to report an error.  If you don't actually explain what is happening, nobody is going to help you.  We are not mind readers, nor are we obligated to extract a decent report out of you.

bvernham, 2022-07-31:
I guess then people like you are just above peons like myself.

I bow to your greatness.

BTW this issue is not the CAN library but how Serial acts with and without this library being include.

Never mind, I am so sorry to waste the time of your greatness!

SuGlider, 2022-08-01:
collin80 and Esp32-can demo are made for ESP32.
[AltPins](https://github.com/collin80/esp32_can/blob/master/examples/CANTestESP32_BuiltIn_AltPins/CANTestESP32_BuiltIn_AltPins.ino) should work fine.

I can tell that this example uses GPIO 16, but in the [SparkFun Thing Plus Board](https://www.sparkfun.com/products/15663), GPIO 16 is also used for UART RX1. 
Not sure if this is the kind of issue you are talking about...

VojtechBartoska, 2022-08-02:
@bvernham Take a look on the last comment of @SuGlider please. It looks like it can be the reason of your problems. :)

closed 2025-01-16 by Parsaabasi