espressif/arduino-esp32 #7067: ESP32 ledc 1bit PWM Configuration is impossible

mhuttner, 2022-07-31:
### Board

ESP32

### Device Description

ESP32-WROOM

### Hardware Configuration

GPIO19 is LED PWM Pin

### Version

v2.0.4

### IDE Name

arduino-cli

### Operating System

macOS 12.4

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

ledcWrite makes it impossible to set the duty cycle to 50% for 1bit PWM because of this:

https://github.com/espressif/arduino-esp32/blob/943216308d0425091498a33ba73e375cbd76287a/cores/esp32/esp32-hal-ledc.c#L89-L96

when `channels_resolution[chan]` is 1 the possible duty cycles should be 0 (0%), 1 (50%) and 2(100%). 
But for a duty of 1 `max_duty` will be 1  (`1<<1 - 1`) and since `duty == max_duty` the duty will be set to `2` 

### Sketch

```cpp
void setup() {
    uint8_t channel = 0;
    
    uint8_t bits = 1;
    uint8_t pin = 19;
    uint32_t freq = 40 * 1000 * 1000 // 40 MHz
    uint8_t duty = 1; // 50% duty

    ledcSetup(channel, freq, bits);
    ledcAttachPin(pin, channel);
    ledcWrite(channel, duty);
}

void loop(){}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-31:
1-bit PWM means the duty cycle is fixed at 50%. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolutions

mhuttner, 2022-07-31:
Yes, but i can not manage to configure it to get output on the pin.
 
Removing the `ledcWrite(channel, duty);`results in no output. 
If i comment out 
```
if(duty == max_duty){ 
     duty = max_duty + 1; 
 } 
```
the example above works as expected and i get a PWM Signal.

P-R-O-C-H-Y, 2022-08-01:
Hi @mhuttner, this part of code was added, because of a simple reason. If all bits from given resolution are set, then LEDC should be full on.

Quick example is on 8-bit resolution. The value is between 0-255, thats 8 bits. But without this piece of code the 255 won't be 100% on, you need to write 256 to be fully on, but it makes no sense.
According to @lbernstone mentioned link, I will make a PR to change 1-bit resolution to be 0 - fully off and 1 be 50%. It makes sense. With 1 bit you can never set duty to 0% , 50% and 100%. Thats 3 values, you have only 2 available (0/1).

If you need to be able to set these, select 2 bits resolution.
Example using only 2 bits: 
0 - always low
1 -> actually sets to 2 - 50% duty
2 -> actually sets to 3 - 75% duty
3 -> actually sets to 4 - always high

P-R-O-C-H-Y, 2022-08-01:
@mhuttner Can you please test your code with this change:

```
    if((duty == max_duty) && (max_duty != 1)) {
        duty = max_duty + 1;
    }
```
Thanks!

mhuttner, 2022-08-04:
@P-R-O-C-H-Y  Thanks for the response. The change works with my code.

PR #7079 (2022-08-08): LEDC 1-bit mode duty fix.
## Description of Change

```
if(duty == max_duty){ 
     duty = max_duty + 1; 
 } 

```
With this previous change we fixed if all bits from given resolution are set, then the LEDC PWM should be full on. But 1-bit mode duty option is 0 - fully off or 1 -fully on which is wrong. As it is in [ESP-IDF docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolutions) the 1-bit PWM means the duty cycle is fixed at 50%. 

This PR fixes that to align with ESP-IDF docs.

## Tests scenarios
 [... cut]
closed 2022-08-08 by SuGlider