espressif/arduino-esp32 #7068: DHCP client failed on 2.0.4

tsctrl, 2022-07-31:
### Board

ESP32 Dev Module

### Device Description

DevKitC

### Hardware Configuration

None

### Version

v2.0.4

### IDE Name

IDF Component, Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

9600

### Description

**dhcp client failed on 2.0.4**
after update to 2.0.4 my dhcp client is not working anymore and i have to do below configuration to call to the internet:

IPAddress STAip(192, 168, 0, 100);
IPAddress STAgateway(192, 168, 0, 1);
IPAddress STAsubnet(255, 255, 255, 0);
IPAddress STAdns(192, 168, 0, 1);

IPAddress primaryDNS(8, 8, 8, 8); // optional
IPAddress secondaryDNS(8, 8, 4, 4); // optional

WiFi.config(STAip, STAgateway, STAsubnet, primaryDNS, secondaryDNS);

else my DNS was not able to automatically assign ip and call to the internet was failed. no error was displayed but i notice that the dns was always set to 255, 255, 255, 255 which i think the cause since it was different that the one in my local network. previously i did not have to do WiFi.config and STA mode automatically configure the connection to my router with proper ip.

Thanks

### Sketch

```cpp
#include <WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-31:
If you run the included code without the WiFi.config() do you also get 255.255.255.255?

tsctrl, 2022-07-31:
without config is getting 255.255.255.255, with config i am getting fixed ip as expected.
i am reverting back @SuGlider changes here and seems to work

https://github.com/espressif/arduino-esp32/pull/6731/commits/3bd9d243bd615ff96c521dc4b3ee0608d05ef79e

SuGlider, 2022-07-31:
@tsctrl - Did you try the fix from #7059 and it worked after that or it still doesn't work even with that fix?

SuGlider, 2022-07-31:
### Output with Static IP when accessing "google.com/index.html"

const char* ssid     = "WiFiTestSSID";
const char* password = "password";
const char* host     = "google.com";
const char* url      = "/index.html";

IPAddress local_IP(192, 168, 15, 3);
IPAddress gateway(192, 168, 15, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8); //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[   450][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   504][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   594][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 192.168.15.3, MASK: 255.255.255.0, GW: 192.168.15.1
[   594][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   601][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
Connecting to WiFiTestSSID
[   653][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: Rogabe1, BSSID: a4:33:d7:cd:c0:28, Channel: 3, Auth: WPA_WPA2_PSK
[   655][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   656][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.15.3
[   670][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   677][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.15.3, MASK: 255.255.255.0, GW: 192.168.15.1
.
WiFi connected!
IP address: 192.168.15.3
ESP Mac Address: 78:E3:6D:CD:06:A8
Subnet Mask: 255.255.255.0
Gateway IP: 192.168.15.1
DNS: 8.8.8.8
connecting to google.com
[  6134][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  6134][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
Requesting URL: /index.html
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/index.html
Content-Type: text/html; charset=UTF-8
Date: Sun, 31 Jul 2022 14:45:42 GMT
Expires: Tue, 30 Aug 2022 14:45:42 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 229
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN
Connection: close

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/index.html">here</A>.
</BODY></HTML>

closing connection
```

### Output using Dynamic DHCP IP when accessing "google.com/index.html"

``` cpp
void setup()
{
  Serial.begin(115200);

//  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
//    Serial.println("STA Failed to configure");
//  }

  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  // ... 
```

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[   450][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
Connecting to WiFiTestSSID
[   504][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   595][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   597][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   597][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[   657][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: Rogabe1, BSSID: a4:33:d7:cd:c0:28, Channel: 3, Auth: WPA_WPA2_PSK
[   659][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
..[  1659][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.15.16
[  1659][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1662][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.15.16, MASK: 255.255.255.0, GW: 192.168.15.1
.
WiFi connected!
IP address: 192.168.15.16
ESP Mac Address: 78:E3:6D:CD:06:A8
Subnet Mask: 255.255.255.0
Gateway IP: 192.168.15.1
DNS: 192.168.15.1
connecting to google.com
[  7135][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  7135][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
Requesting URL: /index.html
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/index.html
Content-Type: text/html; charset=UTF-8
Date: Sun, 31 Jul 2022 14:49:41 GMT
Expires: Tue, 30 Aug 2022 14:49:41 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 229
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN
Connection: close

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/index.html">here</A>.
</BODY></HTML>

closing connection
```

SuGlider, 2022-07-31:
@tsctrl - 

> else my DNS was not able to automatically assign ip

DNS doesn't assign IP Address. DHCP server that runs on the AP that ESP32 will connect is usually the DHCP Server of the network.

The Log Output of this issue doesn't seem to be from Arduino. 
Are you using Arduino as Component or something else?

SuGlider, 2022-07-31:
@tsctrl - I think that you are using Arduino as Component and your IDF isn't  updated
I have submited a change to IDF to deal with the issue of INADDR_NONE from LWIP and Arduino IPAddress Class.
Make sure that your IDF is updated to the latest 4.4.1 version from IDF github.

h1romas4, 2022-08-05:
I had the same issue on ESP32-C3 (M5Stamp).　I also use arduino-esp32 as a component.

- esp-idf v4.4.1 (1329b19f)
- arduino-esp32 2.0.4 (c93bf11f)

An incorrect IP address is returned from DHCP.

```
esp_netif_handlers: sta ip: 255.255.255.255, mask: 255.255.255.255, gw: 255.255.255.255
```

I believe I am using esp-idf v4.4.1 correctly. 

Incidentally, I upgraded esp-idf to v4.4.2(1b16ef6c) as a trial and the problem was resolved.

SuGlider, 2022-08-05:
#### For Arduino as Component:
It seems that IDF v4.4.2-1-gce1fde4b3b has the update I mentitoned before.
It is also possible to use Arduino Core from the the next branch: https://github.com/espressif/arduino-esp32/tree/idf-release/v4.4

h1romas4, 2022-08-06:
@SuGlider 
Thank you for your response!

I can confirm that the combination of esp-idf 4.4.2 and `idf-release/v4.4`(c25a6403) works correctly and the problem is solved.

```
I (30) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (30) boot: compile time 13:36:24
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
...snip...
I (3968) pp: pp rom version: 9387209
I (3968) net80211: net80211 rom version: 9387209
I (3978) wifi:wifi driver task: 3fc9bb24, prio:23, stack:6656, core=0
...snip...
I (4278) wifi:pm start, type: 1
I (4278) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (4368) wifi:BcnInt:102400, DTIM:1
I (5468) esp_netif_handlers: sta ip: 192.168.0.10, mask: 255.255.255.0, gw: 192.168.0.1
```

asier70, 2022-08-08:
I also can confirm that esp-idf ver 4.4.2 solves problem when compiling project with arduino as component.

SuGlider, 2022-08-08:
We just merged IDF 4.4.2 into ESP32 Arduino Github - Master branch.
Using latest master will also solve this issue.

Closing this issue at this time. 
Feel free to reopen it or open a new one is necessary.

closed 2022-08-08 by SuGlider

badrun-bandi, 2022-08-14:
hi @SuGlider as the time i open this issue i did an update to my idf using the idf installer. the installer automatically pick up the 4.4 release tags which was pointing to the revision in the list. the installer it self does not pick the correct idf version that contain the fixes which i guess was causing this issue. understand, to keep both in sync is very difficult. some suggestion, esp arduino should have matching idf branch which should be pair to avoid this unnessesary  issue or at least have idf version in commit comment as idf revision for reference.

i wouldnt know which was matching branch to use and my suggestion, to be safe, esp32 commit that does not compatible or will break on idf master or release tags should not be deployed until the idf have the revision bumps on the release tags that will be compatible with the new esp32 arduino changes. if as such way you did mention the fix is in idf master. master are not release specific where i would not be getting my idf to the master branch as i did not know which esp32 arduino revision are compatible to latest master and i consider that master are not a stable release.

thank you!

badrun-bandi, 2022-08-14:
i saw menodev commits on esp32 arduino release based on which idf. that should be good. some how i miss the match release during the update or i did not able to find 4.4.2 branch during the last update. 

thank you