espressif/arduino-esp32 #7069: SSL - A buffer is too small to receive or write a message

SheetLightning, 2022-08-01:
### Board

ESP32 WROOM

### Device Description

ESP32 devkit V1

### Hardware Configuration

No other devices attached.

### Version

v2.0.4

### IDE Name

Arduino IDE version 1.8.19

### Operating System

Linux Mint 20.3

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When the GET request is issued, the following error is returned:

`[ 43299][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():375]: (-27136) SSL - A buffer is too small to receive
`
Debug mode is set to 'Verbose', otherwise no events are returned from the Calendar API even when calendar events are available for that day. The sketch works fine on the ESP8266.

### Sketch

```cpp
ZIP attached.
```

### Debug Message

```plain
Fetching events from: 
https://script.google.com:/macros/s/AKfycbx7OxaWUoywhIxFxrDDcxPzzEB1Lw43SZ_WKfCN8oFVeWvXZPjwRaMrfdm21O63bH1H/exec
[ 43293][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 150 bytes...
[ 43293][V][ssl_client.cpp:374] send_ssl_data(): Handling error -27136
[ 43299][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():375]: (-27136) SSL - A buffer is too small to receive or write a message
[ 43313][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
Performed GET
```

### Other Steps to Reproduce

Requires a Google Calendar API script to be installed.

https://script.google.com/home/start

I have included the script in the minimal example ZIP.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SheetLightning, 2022-08-16:
I have cut down the minimal example a bit further by removing the RTC and NTP code.

[Google-Minimal.zip](https://github.com/espressif/arduino-esp32/files/9349349/Google-Minimal.zip)

BasDeBass, 2022-08-23:
@SheetLightning I seem to be having the same issue with (coincidentally) almost the same setup. Have you found a solution yet?

SheetLightning, 2022-08-24:
The attached PlatformIO example is a re-written version of the code that removes the need for the httpsRedirect library and uses only the WebClientSecure.print() method to make the HTTP requests. The same problem (and other errors) occurs almost always on the first attempt to make the re-directed GET request and randomly on subsequent attempts.
 
[main.cpp.txt](https://github.com/espressif/arduino-esp32/files/9419360/main.cpp.txt)

probonopd, 2023-04-01:
I am also getting `[  3217][E][ssl_client.cpp:37] _handle_error(): [send_ssl_data():387]: (-27136) SSL - A buffer is too small to receive or write a message` trying to get data from `script.google.com`. I was trying to use https://github.com/jbuszkie/HTTPSRedirect.git#31601fd.

PaulKC, 2023-04-16:
I had the same issue, in the end I managed to get data from `script.google.com` using the method described here https://iotdesignpro.com/articles/esp32-data-logging-to-google-sheets-with-google-scripts

closed 2025-01-16 by Parsaabasi