espressif/arduino-esp32 #7070: Uart data mismatch and appending junk data

asvith, 2022-08-01:
### Related area

uart

### Hardware specification

esp32s2mini1u

### Is your feature request related to a problem?

We are using 2 different WiFI modules for multiple projects.
ESP8266 for older version and ESP32S2-MINI-1U for new projects.
ESP8266 is working fine without any issue for older products. We are facing one unknown issue in ESP32S2-MNI Dev board, and request you to support on higher priority.
Normal UART data out and also data out through serial monitor works fine for any amount of data.
But whenever we connect WiFi or Call WIFI.Begin() function, and start sending data through UART, some Junk data will append at inperiodic intervals.

I am using Arduino latest version IDE, with ESP32 Ver 2.0.2 Library.

Awaiting for your technical team support ASAP.

### Describe the solution you'd like

ESP8266 is working fine without any issue for older products. We are facing one unknown issue in ESP32S2-MNI Dev board, and request you to support on higher priority.

01 03 00 64 00 7E 84 35 
7/21/2022 10:04:03.303 [RX] - 01 03 00 E4 06 10 22 55 FC --> wrong data(junk added)
7/21/2022 10:04:04.345 [RX] - 01 03 00 64 00 7E 84 35   ----> correct data
7/21/2022 10:04:05.385 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:06.410 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:07.451 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:08.491 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:09.516 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:10.552 [RX] - 01 03 00 64 00 7E 0C 35 00---> wrong data(junk added)
7/21/2022 10:04:11.595 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:12.619 [RX] - 01 03 00 64 00 7E 84 35  ----> correct data
7/21/2022 10:04:13.452 [RX] - 01 03 00 64 00 7E 84 35 F8 --->wrong data(junk added)

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-08-01:
@asvith Please retest your code with latest version 2.0.4. We did bunch of updates to UART from version 2.0.2. Thanks

SuGlider, 2022-11-05:
Please update to Arduino Core 2.0.5 and let me know.

VojtechBartoska, 2022-11-28:
Due to no answer, I'm closing this as expired. If needed, you can reopen it. Thanks for understanding.

closed 2022-11-28 by VojtechBartoska