espressif/arduino-esp32 #7071: ledc: frequency affects the brightness of the led

ramdaron, 2022-08-01:
### Board

ESP-WROOM-32

### Device Description

NodeMCU ESP-32S (38pin)

### Hardware Configuration

GPIO 16 and 27 connected to LED
GPIO 26 connected to button

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows  10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

I tested the frequency change with the same duty and noticed that at some values ​​the brightness of the LED is very different.
For example, with a resolution of 6 and a duty of 1 at a frequency of 15415, the LED shines brightly, but if you change the frequency to 15414, then the LED shines much weaker. At a resolution of 5 at 15399, the brightness is lower than at 15400. There may be other combinations.
Unfortunately I don't have an oscilloscope and can't figure out what's happening on the board's pins, but I did measure the voltage. With a power supply of 3.3V and a resolution of 5 at the output without load, I got 0.16V and 0.65V, with a resolution of 6 - 0.12V and 0.27V.

### Sketch

```cpp
#define PIN_LED1 16
#define PIN_LED2 27
#define CHANNEL_LED1 0
#define CHANNEL_LED2 7
#define PIN_BUTTON 26

const int duty = 1;
/*
const int resolution = 6;
const int freqDimly = 15414;
const int freqBrightly = 15415;
*/
const int resolution = 5;
const int freqDimly = 15399;
const int freqBrightly = 15400;

int buttonState = 0;

void setup() {
  ledcSetup(CHANNEL_LED1, freqDimly, resolution);
  ledcAttachPin(PIN_LED1, CHANNEL_LED1);
  ledcWrite(CHANNEL_LED1, duty);

  ledcSetup(CHANNEL_LED2, freqBrightly, resolution);
  ledcAttachPin(PIN_LED2, CHANNEL_LED2);
  ledcWrite(CHANNEL_LED2, duty);

  pinMode(PIN_BUTTON, INPUT_PULLUP);
}

void loop() {
  buttonState = digitalRead(PIN_BUTTON);
  if (buttonState == HIGH ) {
    // button pressed
      ledcChangeFrequency(CHANNEL_LED1, freqBrightly, resolution);
  } else {
    if (ledcReadFreq(CHANNEL_LED1) == freqBrightly) {
      ledcChangeFrequency(CHANNEL_LED1, freqDimly, resolution);
    }
  }
}
```

### Debug Message

```plain
esptool.py v3.3
Serial port COM3
Connecting..........
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:c7:b9:f8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00047fff...
Flash params set to 0x022f
Compressed 17440 bytes to 12090...
Writing at 0x00001000... (100 %)
Wrote 17440 bytes (12090 compressed) at 0x00001000 in 0.5 seconds (effective 279.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 291.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 388.6 kbit/s)...
Hash of data verified.
Compressed 228608 bytes to 125938...
Writing at 0x00010000... (12 %)
Writing at 0x0001dbbc... (25 %)
Writing at 0x000238c7... (37 %)
Writing at 0x00028ca6... (50 %)
Writing at 0x0002df99... (62 %)
Writing at 0x000364ac... (75 %)
Writing at 0x0003e6f7... (87 %)
Writing at 0x00043dc9... (100 %)
Wrote 228608 bytes (125938 compressed) at 0x00010000 in 2.4 seconds (effective 772.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### Other Steps to Reproduce

In the sketch, I installed two identical LEDs on different channels, with the same duty and different frequencies. LED1 will glow weaker than LED2. But if you press the button, their brightness will become the same.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-08-01:
@P-R-O-C-H-Y please help with the triage.

P-R-O-C-H-Y, 2022-08-03:
Hi @ramdaron, I can confirm that its problem with your sketch. You are always calling `ledcChangeFrequency()` function. Adding a `delay(10)` in the loop will solve the issue. You are restarting the timer each `ledcChangeFrequency()` call and thats why the one led is so much brighter, because on Osciloscope, I can see such a high freq around 121 kHz. Can you try that?

VojtechBartoska, 2022-08-23:
@ramdaron DId you solve your issue by the comment above?

ramdaron, 2022-08-29:
@P-R-O-C-H-Y, @VojtechBartoska, sorry for delay.
My code is now:
```
void loop() {
  buttonState = digitalRead(PIN_BUTTON);
  if (buttonState == HIGH ) {
    ledcChangeFrequency(CHANNEL_LED1, freqBrightly, resolution);
    delay(50);
  }
}
```

Pressing the button still changes the brightness of the LED. I'm not sure if the frequency is the problem, can you check the duty for both modes on your oscilloscope?

@P-R-O-C-H-Y, about your tip: thanks, but I have some questions.
My code:
```
#define PIN_LED1 32
#define PIN_LED2 33
#define CHANNEL_LED1 0
#define CHANNEL_LED2 7
#define PIN_BUTTON 26

const int duty = 1;
const int resolution = 5;
const int freqDimly = 15399;
const int freqBrightly = 15400;

int buttonState = 0;
int real Freq;
int freq;

void setup() {
  freq = freqDimly;
  Serial.begin(115200);

  delay(2000);
  realFreq = ledcSetup(CHANNEL_LED1, freqDimly, resolution);
  Serial.printf("Setup LED1 frequency: %i, returned: %i\n", freqDimly, realFreq);
  ledcAttachPin(PIN_LED1, CHANNEL_LED1);
  ledcWrite(CHANNEL_LED1, duty);
  delay(1000);
  Serial.print("Read frequency LED1: ");
  Serial.println(ledcReadFreq(CHANNEL_LED1));

  delay(1000);
  realFreq = ledcSetup(CHANNEL_LED2, freqBrightly, resolution);
  Serial.printf("Setup LED2 frequency: %i, returned: %i\n", freqBrightly, realFreq);
  ledcAttachPin(PIN_LED2, CHANNEL_LED2);
  ledcWrite(CHANNEL_LED2, duty);
  delay(1000);
  Serial.print("Read frequency LED2: ");
  Serial.println(ledcReadFreq(CHANNEL_LED2));

  pinMode(PIN_BUTTON, INPUT_PULLUP);
}

void loop() {
  buttonState = digitalRead(PIN_BUTTON);
  if (buttonState == HIGH) {
    freq++;
    realFreq = ledcChangeFrequency(CHANNEL_LED1, freq, resolution);
    Serial.printf("Set %i, real ", freq);
    Serial.println(realFreq);
    delay(200);
  }
}
```

With this code, I get these values ​​in the serial monitor:
```
05:13:04.065 -> Setup LED1 frequency: 15399, returned: 15384
05:13:05.068 -> Read frequency LED1: 15384
05:13:06.071 -> Setup LED2 frequency: 15400, returned: 15414
05:13:07.073 -> Read frequency LED2: 15414
05:13:21.404 -> Set 15400, real 15414
05:13:24.723 -> Set 15401, real 15414
<skip 15402-15428>
05:13:30.345 -> Set 15429, real 15414
05:13:30.538 -> Set 15430, real 15444
<skip 15431-15457>
05:13:36.607 -> Set 15458, real 15444
05:13:36.800 -> Set 15459, real 15473
<skip 15458-15487>
05:13:42.591 -> Set 15488, real 15473
05:13:42.784 -> Set 15489, real 15503
<skip 15489-15517>
05:20:37.984 -> Set 15518, real 15503
05:20:38.216 -> Set 15519, real 15533
```
And I don't understand why ledcChangeFrequency() returns different values ​​than what I'm trying to set.
Can you help me understand?

P-R-O-C-H-Y, 2022-09-01:
Hi @ramdaron The LEDC peripheral is trying to setup the timer with given resolution to closest possible frequency. Thats why the frequency differ a bit, but its negligible. Can you try to run your code with resolution 8 to see if the actual set frequency gets closer? :) 

I will check your sketch on oscilloscope again to see what's happening.

P-R-O-C-H-Y, 2022-09-05:
@ramdaron I have tested both codes on oscilloscope you posted in last comment. And I can say that both both channels have same duty,
in test 1 - the channel 1 you set frequency in loop while button is pressed hold its duty, completely fine.
in test 2 - the channel 1 according to the code is just setting higher frequency, but there is no hazard. Everything runs smooth.

ramdaron, 2022-09-20:
> The LEDC peripheral is trying to set up the timer with a given resolution to the closest possible frequency.

This is useful information, thanks!

> Can you try to run your code with resolution 8 to see if the actual set frequency gets closer? :)

I specified a fixed value for duty and resolution, and in a loop calling ledcChangeFrequency() tried to specify different frequencies in the range from 1 kHz to 300 kHz. In the end, I got the frequencies that my ESP32 can really tune. I entered these values ​​in the first column in my spreadsheet: https://docs.google.com/spreadsheets/d/1hhFOQk4xuiKGc1dZDvnyGrN9-9z40Lk9LK9QjIZTzbg/edit?usp=sharing
In the second column, I entered the difference between the frequency values ​​that the LEDC peripheral is capable of setting.
I've added a graph over these two columns, and it seems interesting to me: the higher the frequency, the less accurately it can be set by calling ledcChangeFrequency(). After a certain value (which is visible as a peak on the graph), the frequency values ​​can be set quite accurately.

> I have tested both codes on oscilloscope you posted in last comment

Thanks for your oscilloscope test. I was sure duty would be different...

I did another test by building a circuit with an IR2101 gate driver and an IRF3205 mosfet. I connected an LED strip to the mosfet, as it shows changes in brightness better.
My sketch:
```
#define PIN_LED1 32
#define CHANNEL_LED1 0
#define LED_ORANGE 33
#define LED_GREEN 27

const int duty = 1;
const int resolution = 5;
int freq_counter;

// resolution 5
const int freqs[] = {20000, 20050, 20100, 20151, 20202, 20253, 20304, 20356, 20408, 20460, 20512, 20565, 20618, 20671, 20725, 20779, 20833, 20887, 20942, 20997, 21052, 21108, 21164, 21220, 21276, 21333, 21390, 21447, 21505, 21563, 21621, 21680, 21739, 21798, 21857, 21917, 21978, 22038, 22099, 22160, 22222, 22284, 22346, 22408, 22471, 22535, 22598, 22662, 22727, 22792, 22857, 22922, 22988, 23054, 23121, 23188, 23255, 23323, 23391, 23460, 23529, 23598, 23668, 23738, 23809, 23880, 23952, 24024, 24096, 24169, 24242, 24316, 24390, 24464, 24539, 24615, 24691, 24767, 24844, 24922, 25000, 25078, 25157, 25236, 25316, 25396, 25477, 25559, 25641, 25723, 25806, 25889, 25974, 26058, 26143, 26229, 26315, 26402, 26490, 26578, 26666, 26755, 26845, 26936, 27027, 27118, 27210, 27303, 27397, 27491, 27586, 27681, 27777, 27874, 27972, 28070, 28169, 28268, 28368, 28469, 28571, 28673, 28776, 28880, 28985, 29090, 29197, 29304, 29411, 29520, 29629, 29739, 29850, 29962, 30075, 30188, 30303, 30418, 30534, 30651, 30769, 30888, 31007, 31128, 31250, 31312, 31314, 31315, 31317, 31318, 31320, 31321, 31323, 31324, 31326, 31328, 31329, 31331, 31332, 31334, 31335, 31337, 31338, 31340, 31341, 31343, 31344, 31346, 31347, 31349, 31351, 31352, 31354, 31355, 31357, 31358, 31360, 31361, 31363, 31364, 31366, 31367, 31369, 31371, 31372, 31374, 31375, 31377, 31378, 31380, 31381, 31383, 31384, 31386, 31387, 31389, 31391, 31392, 31394, 31395, 31397, 31398, 31400, 31401, 31403, 31404, 31406, 31407, 31409, 31411, 31412, 31414, 31415, 31417, 31418, 31420, 31421, 31423, 31424, 31426, 31428, 31429, 31431, 31432, 31434, 31435, 31437, 31438, 31440, 31441, 31443, 31444, 31446, 31448, 31449, 31451, 31452, 31454, 31455, 31457, 31458, 31460, 31461, 31463, 31465, 31466, 31468, 31469, 31471, 31472, 31474, 31475, 31477, 31479, 31480, 31482, 31483, 31485, 31486, 31488, 31489, 31491, 31492, 31494, 31496, 31497, 31499};

void setup() {
  pinMode(LED_ORANGE, OUTPUT);
  pinMode(LED_GREEN, OUTPUT);
  freq_counter = 0;

  delay(2000);
  ledcSetup(CHANNEL_LED1, freqs[0], resolution);
  ledcAttachPin(PIN_LED1, CHANNEL_LED1);
  ledcWrite(CHANNEL_LED1, duty);
  delay(1000);
}

void loop() {
  for (freq_counter = 0; freq_counter < (sizeof(freqs) / sizeof(freqs[0]) - 1); freq_counter++)
  {
    ledcChangeFrequency(CHANNEL_LED1, freqs[freq_counter], resolution);
    delay(500);
    if (freqs[freq_counter] >= 31250) {
      digitalWrite(LED_ORANGE, LOW);
      digitalWrite(LED_GREEN, HIGH);
    } else {
      digitalWrite(LED_ORANGE, HIGH);
      digitalWrite(LED_GREEN, LOW);
    }
  }
}
```

In the sketch, I have given an array of frequencies that can be set by calling ledcChangeFrequency(). I took the values ​​from the table that cover the peak in the graph for resolution=5: 20000-31312. The orange LED lights up. I also took the values ​​31314-31499. When the frequency reaches 31314, the green LED lights up.
I fixed the exposure and focus of the phone's camera and recorded a video: https://youtu.be/oobhYyxBq3k. As you can see, increasing the frequency to 31312 Hz changes the brightness of the LED strip, but then the brightness does not change.
I thought it might be related to duty, but after your oscilloscope test, I'm not sure anymore.
I also had the idea that my driver and mosfet circuit could not work correctly at such frequencies, which caused the LED strip to flicker, but increasing the frequency after a certain value corrects this. In addition, a regular LED connected to a board pin flickers in the same way.

P-R-O-C-H-Y, 2022-11-15:
@ramdaron Sorry for no reply for a long time. I have a vacation and some other tasks. Do you have any news about this issue? I can do some more tests later this week / next week of needed :)

ramdaron, 2022-11-21:
@P-R-O-C-H-Y Welcome back!)
I updated the Arduino IDE to 2.0.2 (ESP32 Arduino 2.0.5), changed Windows to macOS Ventura, but the problem persists. I hope you can reproduce the problem using the information from my previous post.

P-R-O-C-H-Y, 2022-11-21:
@ramdaron I am testing on Osciloscope right now, and I can see this flickering and brightness changes as you mentioned before with the sketch you provided. I have done few tests and the problem is the resolution and probably the HW with given frequencies. When changing the frequency (<30125) with small resolution the duty slightly changes. I will point out 2 tests.
I have connected LED with resistor directly to the ESP pin 32 (no mosfets).

Test 1 - issue appears
```
resolution = 2 bit (0-4)
duty = 2 (50%)
The duty calculated on oscilloscope when frequency < 31250 varies was from 47,5 to 52,2 % in some cycles.
The duty calculated on oscilloscope when frequency >= 31250 was always 50 %.
```

Test 2 - right behaviour
```
resolution = 10 bit (0-1024)
duty = 512 (50%)
The duty calculated on oscilloscope when frequency < 31250 was always 50 %.
The duty calculated on oscilloscope when frequency >= 31250 was always 50 %.
The is no flickering or brightness changes on the LED, due to fixed duty.
```

So as you can see, to have most accurate **DUTY**, go with the highest possible resolution for the frequency you need.
If you can retest with the parameters of **TEST 2** on your setup, it will be great :) Just change it in your sketch and run it again.

closed 2022-12-15 by P-R-O-C-H-Y