espressif/arduino-esp32 #7072: ESP32 board resetting again and again

Sureshbe, 2022-08-01:
### Board

ESP32 dev module

### Device Description

ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
ets Jun  8 2016 00:22:57

### Hardware Configuration

LoRa moduole connected to ESP32

### Version

v2.0.4

### IDE Name

ARDUINO IDE

[... 254 lines cut ...]
Hash of data verified.
Compressed 263552 bytes to 142322...
Writing at 0x00010000... (11 %)
Writing at 0x0001e01e... (22 %)
Writing at 0x00025119... (33 %)
Writing at 0x0002a737... (44 %)
Writing at 0x0002fbad... (55 %)
Writing at 0x00035ea8... (66 %)
Writing at 0x00040663... (77 %)
Writing at 0x0004718d... (88 %)
Writing at 0x0004c72a... (100 %)
Wrote 263552 bytes (142322 compressed) at 0x00010000 in 2.8 seconds (effective 747.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### Other Steps to Reproduce

I tried on 2 more boards

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-01:
@Sureshbe - It sounds like your sketch is using on of the Flash assigned pins.

Flash SPI QIO pins are: GPIO 6, 7, 8, 9, 10 and 11
If you change to DIO instead of QIO, you may use pins 9 an 10.

#### GPIO 9 is used for the LED and set as OUPUT by the Sketch. In QIO mode, it will reset the board.
``` cpp
#define LED 9

...

void setup() {
  randomSeed(analogRead(0));
  pinMode(LED, OUTPUT);
...
```

Jason2866, 2022-08-03:
I am not sure if using mode DIO solves the reboot. Isnt there still activity on this GPIO (2nd stage bootloader)? 
Gpios 6, 7, 8, 9, 10 and 11 are a no go for ESP32 to use

SuGlider, 2022-08-03:
I tested  setting DIO, and I can use GPIO 9 and 10. 
Those 2 pins are the default pins for Serial1, but it only works with DIO.
Otherwise, using QIO, it is necessary to change Serial1 RX/TX in ```begin()```.

Jason2866, 2022-08-03:
Ahh yes, nice trap. All newer boards have connected the flash to GPIO 9 / 10.
Imho at least it would be good to assign "other" unrestricted Gpios for serial1 when flash mode `qio` is used.
Still have a bad feeling to connect hardware to GPIO 9/10 when the flash chip is connected there too.

dbhaig, 2022-08-04:
### ESP32-S3-DEVKITC-1-N8 
I have the same issue when programming an ESP32-S3-DEVKITC-1-N8 board. The only connection to the board is a USB cable connected to the UART port.

### Compile
I compiled the BareMinimum example using : arduino-cli compile -b esp32:esp32:esp32s3 -v
Resulting in:
. . .
esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32s3-elf-size -A /tmp/arduino-sketch-08CF0556BB4284CF461BE2005A5EA1D0/BareMinimum.ino.elf
Sketch uses 208601 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 12824 bytes (3%) of dynamic memory, leaving 314856 bytes for local variables. Maximum is 327680 bytes.

Used platform Version Path                                                    
esp32:esp32   2.0.4

### Upload
I uploaded sketch using: arduino-cli upload -p /dev/ttyUSB0 --fqbn esp32:esp32:esp32s3 -v
BareMinimum$ esp32-s3_upload.sh 
python3 "/home/don/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py" --chip esp32s3 --port "/dev/ttyUSB0" --baud 921600  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 "/tmp/arduino-sketch-08CF0556BB4284CF461BE2005A5EA1D0/BareMinimum.ino.bootloader.bin" 0x8000 "/tmp/arduino-sketch-08CF0556BB4284CF461BE2005A5EA1D0/BareMinimum.ino.partitions.bin" 0xe000 "/home/don/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/partitions/boot_app0.bin" 0x10000 "/tmp/arduino-sketch-08CF0556BB4284CF461BE2005A5EA1D0/BareMinimum.ino.bin" 
esptool.py v3.3-dev
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:b4:72:70:07:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00043fff...
Compressed 15008 bytes to 10328...
Wrote 15008 bytes (10328 compressed) at 0x00000000 in 0.3 seconds (effective 374.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 511.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 686.1 kbit/s)...
Hash of data verified.
Compressed 208992 bytes to 116565...
Wrote 208992 bytes (116565 compressed) at 0x00010000 in 1.8 seconds (effective 919.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

### Result
Connecting to the UART port  with picocom reveals:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78 
(repeats)

Compiling with the arduino IDE yields the same result. 
Previously, I had success uploading and running sketches on this board. (That was before I upgraded arduino-cli from  version 0.21.1 to 0.25.1)

Jason2866, 2022-08-04:
@dbhaig i think same as https://github.com/espressif/arduino-esp32/issues/6980#issuecomment-1183655704
I do not use Arduino IDE nor CLI. With Platformio it does work

closed 2022-11-05 by SuGlider

Rob58329, 2023-05-01:
Re.ArduinoIDE:  I often get ESP32 "rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)" crashes when doing “EEPROM.commit” or similar EEPROM activities when other processes are using interrupts (both external interrupts and also internal timer-interrupts).  [The solution for me was to specifically disable the offending interrupt before the crashing EEPROM activity.]

Under these circumstances I have seen an EEPROM activity at the top of setup() causing repeated boot-crashes similar to the one shown above.

(I have also seen the use of "SerialM.setDebugOutput(true)" causing the above crash on a subsequent “EEPROM.commit()” when an error message is generated from any other interrupt process during the “EEPROM.commit()” – in instances where removing the "SerialM.setDebugOutput(true)" is sufficient to allow the error-message (eg. “cam_hal: EV-EOF-OVF”) to be printing during the “EEPROM.commit()” without crashing it.)

bmedici, 2023-08-06:
Same crash here with ESP32-S3, 16MB Flash. Not using SPI0 pins it seems.
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378325
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x9b4
load:0x403cc700,len:0x28fc
entry 0x403c98bc

jenschr, 2023-11-08:
> Gpios 6, 7, 8, 9, 10 and 11 are a no go for ESP32 to use

I have a related issue on a new PCB and what @Jason2866 says here does scare me a bit? According to the documentation from Espressif, the pins 6, 7, 8, 9, 10 and 11 are all perfectly good? But for modules with PSRAM, you must absolutely not use pins 35, 36 and 37 since that's connected to the PSRAM (at the end of section 3.2 in ESP32S3-WROOM-1 datasheet). In my case, that is what caused this (and lots of other issues).

jenschr, 2023-11-08:
Thanks for clarifying @Jason2866 !

Shayantika-Dhar, 2024-01-06:
what can we do to resolve the issue of the serial monitor output

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
ets Jun  8 2016 00:22:57

19e2k67, 2024-01-28:
[image]
Some pins of ESP32 cannot be used as GPIO, such as GPIO9, GPIO10, and GPIO11 as shown in the figure

devitalii, 2024-03-07:
> what can we do to resolve the issue of the serial monitor output
>
> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee [...]

I had similar issue, ESP32 C3 -----

PROBLEM was that I tried to use wrong PIN numbers. 
Some pins you can not use as they are RESET or dedicated to FLASH memory. See posts above/

I accidentally tried to use these pins on ESP32 c3 mini (some are not even wired out for usage,  18,23,16,4,5,17 in TFT_eSPI library as DIO pins. (Hate that User_Setup.h that you need to redefine for each ESP32-Screen combo). 
However my sketch was using pin 4 as A4.
I think it was conflict of pins. 

This gave me error: 
rst:0x8 (TG1WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)

Focus06, 2024-10-20:
Hi, continuing this topic I want to ask for advice. I'm confused and I don't understand where is the source of issue.
1) I have XIAO ESP32S3
2) When connected via USB to PC whole code works as expected.
3) When I remove USB and make external power supply (example power bank, simple battery) - code is not working
4) I plugged FTDI with serial port monitoring and supply 5 V. This is what I see:
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
`
5) I was confused so I reduced my code to simple "serial.print"
`void setup() {
Serial.begin(115200);
}
void loop() {
Serial.println("Hi..");
delay(100);
}`

Result the same. I have the same baud rate set in code and serial monitor in Arduino IDE.

How to read this error? How to obey it?

Focus06, 2024-10-20:
I had something similar in my XIAO ESP32S3, I was really confused because I do not use "forbidden" pins. 
Solution was to set "Disable" _USB CDC On Boot_

NikInoxsys, 2024-12-19:
Hi, my 5 cents, for me this problem was if there is mismatch between past and new partition.csv and partition offset in menuconfig