espressif/arduino-esp32 #7074: unused but set variables

lbernstone, 2022-08-02:
### Board

esp32 dev module

### Device Description

any device

### Hardware Configuration

any

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

ubuntu

### Flash frequency

all

### PSRAM enabled

no

### Upload speed

115200

### Description

platform.txt shows -Wno-error=unused-but-set-variable throughout, and yet is overriden somewhere.  Obviously not a major issue, but truly annoying when building out code.

### Sketch

```cpp
char* test;
```

### Debug Message

```plain
compile error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-02:
Hi @lbernstone - There is a `-Werror=all` in the `platform.txt`. Maybe this is related to the issue.
We have opened a PR to remove it. #7060 
Could you please test this PR and let us know if applying it solves the issue?

bertmelis, 2022-08-02:
Please, even if a (void)var is needed, I still think it is better than just silence the warning. Every time I try to build something I have to check if it's the framework's warning or mine.

lbernstone, 2022-08-02:
If don't think it should be silenced @bertmelis, but it should not be fatal.  The framework shouldn't be throwing warnings! I'll try that PR and see if it runs the middle ground (warning that variable is unused), @SuGlider .

SuGlider, 2022-08-08:
For reference: Users can override `-Werro=all` or any `-Wnoxxxx` settings using `platform.local.txt` into user's own Arduino board folder (from github or the one installed into `Arduino15` folder).

https://arduino.github.io/arduino-cli/0.21/platform-specification/#platformlocaltxt

rin67630, 2023-11-12:
I have the same problem.
Cannot compile on Arduino IDE
`/Volumes/Data/Activities/5_Tinker/5_Arduino/libraries/Ai_Esp32_Rotary_Encoder/src/AiEsp32RotaryEncoder.cpp: In member function 'ButtonState AiEsp32RotaryEncoder::readButtonState()':
/Volumes/Data/Activities/5_Tinker/5_Arduino/libraries/Ai_Esp32_Rotary_Encoder/src/AiEsp32RotaryEncoder.cpp:279:14: error: unused variable '_buttonState' [-Werror=unused-variable]
  ButtonState _buttonState = buttonState;
              ^~~~~~~~~~~~
cc1plus: some warnings being treated as errors`

lbernstone, 2023-11-12:
@rin67630 The issue here was this error was being thrown by code from this repo.  Third party libraries are responsible for fixing their own errors.

closed 2023-11-12 by lbernstone