espressif/arduino-esp32 #7075: cannot boot while burnning bin file through "flash_download_tool_3.9.2_0" tool

Rita858, 2022-08-02:
### Board

ESP32-C3-MINI-1 module

### Device Description

wiring between ESP32-C3-MINI-1 and TTL are as below：

ESP32-C3-MINI-1 | TTL 
-------- | -----
GND | GND 
3V3 | 3V3
U0TXD | RXD
U0RXD | TXD
IO9 | DTR
EN | RTS

### Hardware Configuration

None.

### Version

v2.0.3

### IDE Name

Arduino 1.8.15

### Operating System

Windows 10

### Flash frequency

40M

### PSRAM enabled

no

### Upload speed

921600

### Description

Module can run program after burned from arduino IDE while fail when burned through **"flash_download_tool_3.9.2_0 tool"** tool.
All files to be burned are obtained from the file path prompted when downloading through the arduino IDE.
[image]

`
C:\Users\Bat\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0/esptool.exe --chip esp32c3 --port COM32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x0 C:\Users\Bat\AppData\Local\Temp\arduino_build_253608/OTAWebUpdater.ino.bootloader.bin 0x8000 C:\Users\Bat\AppData\Local\Temp\arduino_build_253608/OTAWebUpdater.ino.partitions.bin 0xe000 C:\Users\Bat\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/partitions/boot_app0.bin 0x10000 C:\Users\Bat\AppData\Local\Temp\arduino_build_253608/OTAWebUpdater.ino.bin 
`
How can I fix it? Looking forward to your reply.

### Sketch

```cpp
the code I test is "OTAWebUpdater".the file path is:Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\ArduinoOTA\examples\OTAWebUpdater
```

### Debug Message

```plain
The log of startup failure is as follows：
[14:39:17.463]收←◆ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 6ba1565fa2d2e6a4924306b96201eaf0ca45d5d10ca97809a07150b66f7e6837
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d10fc
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 6ba1565fa2d2e6a4924306b96201eaf0ca45d5d10ca97809a07150b66f7e6837
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d10fc
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 6ba1565fa2d2e6a4924306b96201eaf0ca45d5d10ca97809a07150b66f7e6837
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d10fc
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 6ba1565fa2d2e6a4924306b96201eaf0ca45d5d10ca97809a07150b66f7e6837
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-02:
@Rita858 - not sure exactly what is your question.

When using flash_download_tool, it is necessary to input the information from `esptool.exe` command line into the API.
http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/

Try the tutorial from link above, but change the addresses as in your command line:

1- Select the right chip (C3), Dev Mode and UART
2- After opening the Flash Window, seelect DIO, 40MHz and COM32
3- On the Files:  select one by one using `....` buton, and enter the address in `@` field
- C:\Users\Bat\AppData\Local\Temp\arduino_build_253608/OTAWebUpdater.ino.bootloader.bin  @ 0x0
- C:\Users\Bat\AppData\Local\Temp\arduino_build_253608/OTAWebUpdater.ino.partitions.bin @ 0x8000
- C:\Users\Bat\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/partitions/boot_app0.bin @ 0xe000
- C:\Users\Bat\AppData\Local\Temp\arduino_build_253608/OTAWebUpdater.ino.bin @ 0x10000
4- Press `Start` button and wait.

Something similar (but not exactly - different layout, files and addresses) this screenshot:

[image]

Rita858, 2022-08-09:
@SuGlider I have tried to burn program through flash down "flash_download_tool_3.9.2_0 tool" tool. But the chip did not setup successfully. I suspect that the download file prompted by arduino is wrong, or the address provided is wrong.

> ### Description
> Module can run program after burned from arduino IDE while fail when burned through **"flash_download_tool_3.9.2_0 tool"** tool. All files to be burned are obtained from the file [...]

SuGlider, 2022-08-09:
I just tried it... didn't work.
I'll try to find out what is wrong.

SuGlider, 2022-08-09:
OK. I got it working.... I had to select the 4 files clicking in the [v] check of each one.

SuGlider, 2022-08-09:
[image]

SuGlider, 2022-08-09:
I added the 4 files and their addresses as before. I also clicked in the 4 [checks] of the 4 files. Then I just pressed START.
Select the right COM port, and 921600 as BAUD.

SuGlider, 2022-08-09:
Ok... better to just flash ONLY the main firmware @ 0x10000 and the partition file @0x8000.
It seems that something is wrong with the Bootloader(@0x0) and maybe with the other boot_app0.bin @0xe000

So, follow this sequence:
1) Uopload any (even empty or hello world) sketch usinf Arduino IDE. This will setup the 4 files correctly and make ESP32-C3 work fine.
2) Open the Arduino IDE Serial Monitor and check that the C3 is working as expected. After that CLOSE the Serial Monitor to free the COM port to other applications.
3) Using "Flash Download Tool 3.9.2" select only the main Firmware and Partition file. Do not upload the bootloaders.
4) Select the checkbox of both files, select the COM port, 921600 BAUD and press <START>
5) The progress bar should move. After the green button displays "FINISH", press reset button on the C3 and the new sketch should work.

Rita858, 2022-08-12:
@SuGlider Hi，thanks for your reply.However, the matter is still not solved. Could you please provide the startup file you have mentioned?
[image]
We are in a hurry to produce. Thank you.

SuGlider, 2022-08-12:
This file is at the `Arduino15` folder (arduino installation).
Example:
`C:\Users\<user_name>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\tools\partitions\boot_app0.bin`

or can be found in the ESP32  Arduino Github:
https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/boot_app0.bin

Rita858, 2022-09-09:
Hi,the problem has been solved finally.It is not only related to the boot file but also to the flash tool."Flash Download Tool 3.8.8" can combine file successfully while "Flash Download Tool 3.9.2" fail.

closed 2022-11-05 by SuGlider