espressif/arduino-esp32 #7076: HttpClient.cpp#setCookie()

fvaryu, 2022-08-02:
### Board

ESP32Dev Kit

### Device Description

DevKit

### Hardware Configuration

none

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

 HttpClient.cpp#setCookie() 

`
void HTTPClient::setCookie(String date, String headerValue)
{
    if (!_cookieJar)
    {
        return;
    }
    #define HTTP_TIME_PATTERN "%a, %d %b %Y %H:%M:%S"

    Cookie cookie;
    String value;
    int pos1, pos2;

    headerValue.toLowerCase(); // ? why toLoerCase()
  // other code
}
`
Why turn the value into lowercase, if the server returns uppercase? i don't understand。thx

`
  CookieJar cookies;
  http.setCookieJar(&cookies);
`
The server returns a mixed case token, but I get lowercase by CookieJar

### Sketch

```cpp
void HTTPClient::setCookie(String date, String headerValue)
{
    if (!_cookieJar)
    {
        return;
    }
    #define HTTP_TIME_PATTERN "%a, %d %b %Y %H:%M:%S"

    Cookie cookie;
    String value;
    int pos1, pos2;

   headerValue.toLowerCase();

    struct tm tm;
    strptime(date.c_str(), HTTP_TIME_PATTERN, &tm);
    cookie.date = mktime(&tm);
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-02:
Hummm..... that seems a bug. Cookies are case sensitive.

SuGlider, 2022-08-10:
@fvaryu - If possible, please test the PR #7112 and let me know if it fixes the issue. Thanks.

PR #7112 (2022-08-16): Fixes Cookie Case Sensitivity
## Description of Change
Cookie Names and Value are case sensitive, but attibutes such as "Expires", "Max-Age", "Domain", "Path", "Secure", "HttpOnly" must be case insensitively matched. This PR fixes it in `HTTPClient::setCookie(String date, String headerValue)` function.

## Tests scenarios
ESP32

## Related links
Fixes #7076
closed 2022-08-16 by P-R-O-C-H-Y