espressif/arduino-esp32 #7077: error: std::round has not been declared

jatin-31, 2022-08-02:
### Board

ESP32-CAM

### Device Description

WSP32-S

### Hardware Configuration

HW-417-V1.2

### Version

v2.0.4

### IDE Name

Arduino

### Operating System

Win 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

In file included from C:\Users\*****\AppData\Local\Temp\arduino_build_766528\sketch\CameraWebServer.ino.cpp:1:0:
D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.4\cores\esp32/Arduino.h:183:12: error: 'std::round' has not been declared
 using std::round;
            ^
Multiple libraries were found for "WiFi.h"
 Used: D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi
Using library WiFi at version 2.0.0 in folder: D:\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi 
exit status 1
Error compiling for board ESP32 Wrover Module.

### Sketch

```cpp
CameraWebServer.ino.cpp
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-08-03:
Hi @jatin-31 I was not able to reproduce the issue.

fryefryefrye, 2022-08-04:
This problem can reproduce in arduino-1.8.10.
It can be fixed by using arduino-1.8.19.
I think this issue can be closed.

jatin-31, 2022-08-04:
Yes, It got it fixed in 1.8.19

closed 2022-08-04 by jatin-31

k5map, 2022-08-17:
I'm running IDE v1.8.19 and having this issue so its not resolved for me.... any other suggestions/ideas?

KingofDelusion, 2022-08-22:
I'm also getting the exact same issue in v1.8.19. Would appreciate any help, as it doesn't seem to be solved.

Here's my error:

Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from sketch\Colour_Test.ino.cpp:1:0:

C:\Users\Ily\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/Arduino.h:183:12: error: 'std::round' has not been declared

 using std::round;

            ^

exit status 1

Error compiling for board ESP32 Dev Module.

vdbrink, 2022-08-22:
If you repace it with only `::round` it will work (at least it worked for me)

k5map, 2022-08-22:
I was experiencing this problem and resolved it by changing the URL for the ESP32 board.  The URL I was using, limited me to downloading the v1.x.x driver.  Once I updated my URL to one provided by Adafruit, I can now download v2.0.x of the board driver and my compile error has been resolved.

robomaniac, 2022-08-23:
> I was experiencing this problem and resolved it by changing the URL for the ESP32 board. The URL I was using, limited me to downloading the v1.x.x driver. Once I updated my URL to one provided by Adafruit, I can now download v2.0.x of the board driver and my compile error has been resolved.

could you provide the URL so I can test this? I have issues on 1.8.19

I use these
_https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
https://dl.espressif.com/dl/package_esp32_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json_

k5map, 2022-08-23:
Here are the URL's I'm using (first is for my ESP8266):

http://arduino.esp8266.com/stable/package_esp8266com_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

I was using the URL #2 in your list and that was causing my compile error

robomaniac, 2022-08-23:
SOLVED
in board manager, searching for esp32, I had 1.0.6 and not the latest 2.0.4
I can not compile

thanks to this post
https://forum.arduino.cc/t/solved-esp32-problem-during-compiling/1019611/16

k5map, 2022-08-23:
@robomaniac ... that was my post (k5map) as well :)

VojtechBartoska, 2022-09-23:
please try 2.0.5 @rkoehler2017

SoniaZotz, 2022-10-20:
In file included from C:\Users\zippy\AppData\Local\Temp\arduino-sketch-DE440033806199EEE7E06FC98EE4618E\sketch\CameraWebServer.ino.cpp:1:0:
C:\Users\\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/Arduino.h:183:12: error: 'std::round' has not been declared
 using std::round;
            ^

exit status 1

Compilation error: exit status 1

P-R-O-C-H-Y, 2022-10-20:
@SoniaZotz Can you try installing or reinstalling arduino core inside Arduino IDE boards manager? It should fix the problem.

maximumeffort2002, 2023-04-14:
I also did encounter that error but the only thing that work is deleting the arduino15 and reinstalling the arduino ide