espressif/arduino-esp32 #7078: ESP32 C3 2mb flash version keeps rebooting

kokospalme, 2022-08-02:
### Board

ESP32 C3 2mb

### Device Description

ESP32-C3-13U 2MB

### Hardware Configuration

GPIO 8 to 3.3V
GPIO9 to GND for flashing 

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

OSX

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When I upload a sketch, for example a simple Blink sketch to a board with an ESP32-C3 with 2 MB flash, the board keeps rebooting and detects a lash size, that is smaller than the size in the binary image header. When I change the parameter board_build.f_flash to "20000000L", the compiler doesn't find a bootloader for the board and does not compile. But when I set the flashsize to 4mb, the sketch compiles and I can upload it to the board. So that is my current platform.ini:

`[env:esp32-c3-2mb]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_speed = 921600
upload_port = /dev/cu.wchusbserial14130
monitor_port = /dev/cu.wchusbserial14130
monitor_speed = 115200`

**Is there a way to change the flash size, maybe change the partitions or something?**
I already searched with google and on this issue-page, but I couldn't find anything... 
The Chip is an "ESP32-D0WDQ6"

### Sketch

```cpp
#include <Arduino.h>

#define PIN_LED 0

void setup() {
  pinMode(PIN_LED, OUTPUT);
}

void loop() {
  digitalWrite(PIN_LED, HIGH);
  delay(500);
  digitalWrite(PIN_LED, LOW);
  delay(500);
}
```

### Debug Message

```plain
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381782
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4bd624aff3257f126bd68f521ac53163cd7fe48cae8d3a77a847ef36aa76ad41
Expected: 2dd871cd993fccdf87b5ef3c78d949bd29ad8f4d97ecb130ff4297267fa0ddbe
Attempting to boot anyway...
entry 0x403ce000
E (113) spi_flash: Detected size(2048k) smaller than the size in the binary image header(4096k). Probe failed.

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)
Core  0 register dump:
MEPC    : 0x4038192e  RA      : 0x403848ae  SP      : 0x3fcde190  GP      : 0x3fc8b000  
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fcde2ff  S1      : 0x00000088  A0      : 0x3fcde1f4  A1      : 0x3fc8af15  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc8d000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcde1e8  S3      : 0x00000001  
S4      : 0x3fcde1e8  S5      : 0x42009e88  S6      : 0x42000020  S7      : 0x3c020020  
S8      : 0x00010020  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fcde190: 0x34303936 0x0000080a 0x3c022a2c 0x40389404 0x3fcc0000 0xb33fffff 0x00000000 0x3fc8af14
3fcde1b0: 0x42000020 0x0000006e 0x0000000a 0x00383932 0x400481b0 0x3fc8b744 0x3c022a2c 0x3fc8b728
3fcde1d0: 0x3c022a8f 0x3fc8b754 0x3fcde1bc 0x3fc8b758 0x3c022ae4 0x3fc8af14 0x00000000 0x00000000
3fcde1f0: 0x36000000 0x65737361 0x66207472 0x656c6961 0x64203a64 0x6f635f6f 0x695f6572 0x2074696e
3fcde210: 0x72617473 0x2e707574 0x39323a63 0x66282038 0x6873616c 0x7465725f 0x203d3d20 0x5f505345
3fcde230: 0x00294b4f 0x00000000 0x403814c2 0x3fc8ca70 0x3fcdf5c0 0x3fc8ac64 0x00000104 0x40048722
3fcde250: 0x42000020 0x00000000 0x00000005 0x00000000 0x3fc8ac38 0x00000000 0x3fc8ac64 0x3fcde284
3fcde270: 0x00000000 0x00000000 0x005e6015 0x42005a8c 0x42000020 0x00000071 0x3c021764 0x00000800
3fcde290: 0x00001000 0x000003e8 0x00000000 0x03ffffff 0x42000020 0x00000000 0x00000000 0x00000001
3fcde2b0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x403814c2 0x3fc8ca70
3fcde2d0: 0x3c023000 0x3c023000 0x3fc8ca70 0x42009e88 0x00000001 0x000001e0 0x00000101 0x4200e45a
3fcde2f0: 0x3fc8d000 0x00000000 0x403814c2 0x00020020 0xffff0000 0x0000003e 0x00000000 0x403816c8
3fcde310: 0xed049501 0x3ff1c14c 0x00000000 0x00000007 0x00ffffff 0x3fcd0101 0x00000000 0x7fffffff
3fcde330: 0x00000000 0x0002ffaf 0x0002ffcf 0x403d0cba 0x00002dbc 0x3ff1c14c 0x00000000 0x3fcdf000
3fcde350: 0x00002dbc 0x3ff1c14c 0x0000ffff 0xffffffff 0xffffffff 0x3fcde4b0 0xffffffff 0x403d0fbe
3fcde370: 0xffd5330c 0xd8be9916 0x00010000 0x001f0000 0x00010000 0x200206e9 0x403814c2 0x000000ee
3fcde390: 0x00000005 0x00000000 0x01000000 0x3c020020 0x0000c260 0x3fc8a800 0x0000154c 0x40380000
3fcde3b0: 0x0000283c 0x42000020 0x0001ffb0 0x4038283c 0x00007e84 0x50000010 0x00000010 0x00000000
3fcde3d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde3f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde410: 0x00000000 0x00000000 0x00000000 0x00010020 0x0001c288 0x0001d7dc 0x00020020 0x0003ffd8
3fcde430: 0x00047e64 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde450: 0x00000000 0x00000000 0x00000000 0x00037ea0 0xed049501 0xedf77892 0xf6fc64a1 0x21d7eda5
3fcde470: 0x627fb44f 0xb644e2dd 0x5d657e3c 0x83ab3cb7 0x400481b0 0x00000010 0x0000000a 0x3fce0000
3fcde490: 0x00000000 0x000000a5 0x00002dc0 0x403ce05c 0x02823d04 0x55648b03 0x00000000 0xffffffff
3fcde4b0: 0x00000000 0x00000000 0x00010000 0x001f0000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde4f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde530: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde550: 0x600c5200 0x600c5200 0x80000000 0x4004a296 0x970275ca 0x87403e47 0x3fcde668 0x00000000
3fcde570: 0x31373832 0x00000000 0x00010000 0x200203e9 0x403ce000 0x403d0000 0x00002358 0x000000ee

ELF file SHA256: 0000000000000000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-08-02:
You should be able to create your own partition table following the directions at https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables.
Base your custom table on the [ones in arduino](https://github.com/espressif/arduino-esp32/tree/master/tools/partitions), they must include an otadata partition.
I don't think this is a bootloader problem, since the arduino bootloader use the same files for various flash sizes.

kokospalme, 2022-08-02:
When I change the flash to 2mb, I can compile the code, but I'm not able to upload it. This is what the terminal dumps:
´´´
Processing esp32-c3-devkitm-1 (platform: espressif32; board: esp32-c3-devkitm-1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio/build/esp32-c3-devkitm-1/firmware.elf
Checking size .pio/build/esp32-c3-devkitm-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.0% (used 9684 bytes from 327680 bytes)
Flash: [==        ]  16.6% (used 217370 bytes from 1310720 bytes)
======================================================== [SUCCESS] Took 2.57 seconds ========================================================
 *  Terminal will be reused by tasks, press any key to close it. 
´´´

This is the error When I try to upload it:
´´´
Processing esp32-c3-devkitm-1 (platform: espressif32; board: esp32-c3-devkitm-1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio/build/esp32-c3-devkitm-1/firmware.elf
Checking size .pio/build/esp32-c3-devkitm-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.0% (used 9684 bytes from 327680 bytes)
Flash: [==        ]  16.6% (used 217370 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/cu.wchusbserial14130
Uploading .pio/build/esp32-c3-devkitm-1/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/Users/Chris/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/bin/bootloader_dio_20m.bin'
*** [upload] Error 2
======================================================== [FAILED] Took 3.05 seconds ========================================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
´´´

When I change the partition scheme, I can upload the code (in 4mb mode), but it keeps rebooting with the same error.

lbernstone, 2022-08-02:
Use mode qio unless you know that you need dio. f_flash is the speed, not the size.

SuGlider, 2022-08-02:
There is an option in the `ESP32C3 Dev Module` board selection for 2MB flash and Partion:

[image]

SuGlider, 2022-08-02:
You can try to "export it" to PIO.

@Jason2866 - Any better idea about it?

Jason2866, 2022-08-02:
I use this boards manifest for a C3 with 2MB
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32c3_out.ld"
    },
    "core": "esp32",
    "extra_flags": "-DARDUINO_USB_CDC_ON_BOOT=0 -DESP32_2M -DESP32C3",
    "f_cpu": "160000000L",
    "f_flash": "80000000L",
    "flash_mode": "dout",
    "mcu": "esp32c3",
    "variant": "esp32c3",
    "partitions": "esp32_2M_partition_app1792k_no_ota_spiffs192k.csv"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32c3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "Espressif Generic ESP32-C3 2M Flash, Tasmota 1792k Code No OTA, 192k FS",
  "upload": {
    "flash_size": "2MB",
    "maximum_ram_size": 327680,
    "maximum_size": 2097152,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html",
  "vendor": "Espressif"
}
```
and this custom esp32_2M_partition_app1792k_no_ota_spiffs192k.csv 
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x2F0000,
app1,     app,  ota_1,   0x300000, 0x2F0000,
spiffs,   data, spiffs,  0x5F0000,0x210000,
```
to build Tasmota. It does work.

SuGlider, 2022-08-02:
@Jason2866 I love that you have tried all possible boards and configurations with PIO!

This is really helpful! Thanks for the quick answer!

lbernstone, 2022-08-02:
Well, that partitioning isn't going to fit in 2MB, but the minimal.csv should work with the rest of the definition.

kokospalme, 2022-08-03:
thank you for alle these answers! the following platform.ini worked wfor me:

```
[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
board_build.f_flash = 80000000L  ; flash freq
board_build.flash_mode = qio  ; flash mode
board_upload.flash_size = 2MB ; max. size in MB
board_upload.maximum_size = 2097152 ;  max.  size 
board_uload.maximum_ram_size = 327680  ; max. RAM size
board_build.partitions = minimal.csv   ; partitial scheme
upload_speed = 921600   ; uload speed
upload_port = /dev/cu.wchusbserial14130   ; port
monitor_port = /dev/cu.wchusbserial14130
monitor_speed = 115200  ; port BAUD
```

closed 2022-08-03 by kokospalme

lbernstone, 2022-08-03:
I know it is pedantic, but your maximum_size should actually be 1310720

Jason2866, 2022-08-03:
@lbernstone  You are right copied wrong file! This is the Tasmota used working one partitions.csv
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1C0000,
spiffs,   data, spiffs,  0x1D0000,0x30000,
```