espressif/arduino-esp32 #7084: enable CONFIG_RTC_CLK_SRC in arduino IDE to improve deep sleep time accuracy

Daniel-1276, 2022-08-04:
### Related area

additional compile option

### Hardware specification

ESP32 and all derivates

### Is your feature request related to a problem?

Problem:
not able to set CONFIG_RTC_CLK_SRC in Arduino IDE for battery based applications. Default and hardcoded 150KHz RTC clock in Arduino has very bad time drift, if deep sleep is used for only a couple of hours (in my case + 20min for 15h sleep).

### Describe the solution you'd like

Enable CONFIG_RTC_CLK_SRC in Arduino IDE to use for example far better clock option "RTC_CLK_SRC_INT_8MD256" at the cost of only a few µa. 

### Describe alternatives you've considered

- Using NTP at the cost of constantly waking up connecting to wifi and correct time drift.
- Using external clock source like DS1307 or DS3231 at additional hardware and complexity cost.

### Additional context

This is an important feature request for all battery powered projects using deep sleep with timer wakeup source. 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Daniel-1276, 2022-10-14:
I assume espressif's own esp-idf may change dividers too, to accommodate higher frequencies. No idea on how to compile code with 8Mhz clock properly in Arduino.

WereCatf, 2022-10-14:
> No idea on how to compile code with 8Mhz clock properly in Arduino.

If anyone does, they certainly haven't bothered telling it anywhere.

igotab, 2022-10-18:
This issue has been raised here since early 2018.

Espressif, thank you for the amazing job you have done with ESP32.

Unfortunately, a lot of your customers, including myself, do not have the necessary expertise to use ESP-IDF and menuconfig to enable an external 32.768kHz xtal and the RTC clock is useless in sleep mode without it.

Please add a build flag to enable an external xtal in Arduino framework.

jrl290, 2022-12-22:
I would like to bump this request

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

cleishm, 2025-09-03:
I believe this feature request is not implemented, so would appreciate if you'd consider reopening it. I've also noted that it would be quite valuable to have a way to specify that the esp32 use the higher frequency internal RTC oscillator to reduce clock drift (at the cost of a few µa).