espressif/arduino-esp32 #7086: getApbFrequency() return 80MHz even though CPU < 80 MHz on ESP32-S3

ehong-tl, 2022-08-05:
### Board

ESP32-S3-DevKitC-1-N8R8

### Device Description

None

### Hardware Configuration

None

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

getApbFrequency() return 80 MHz even though CPU frequency is less than 80 MHz, which makes this UART [changes](https://github.com/espressif/arduino-esp32/blob/142fceb8563cd1795d619829e0a103770a344e1a/cores/esp32/esp32-hal-uart.c#L74) ineffective for ESP32-S3. I have to manually multiply the baudrate during Serial.begin(_desired baudrate * multiplier_). I didn't encounter such issue on older ESP32 and ESP32-S2.

### Sketch

```cpp
uint32_t Freq = 0;

void setup() {
  setCpuFrequencyMhz(10);

  Serial.begin(115200 * 8);  // Have to manually multiply the baudrate based on CPU frequency

  Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getApbFrequency();
  Serial.print("APB Freq = ");
  Serial.print(Freq);
  Serial.println(" Hz");

}

void loop() {

}
```

### Debug Message

```plain
11:16:52.749 -> ESP-ROM:esp32s3-20210327
11:16:52.749 -> Build:Mar 27 2021
11:16:52.749 -> rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
11:16:52.770 -> SPIWP:0xee
11:16:52.770 -> mode:DIO, clock div:1
11:16:52.770 -> load:0x3fcd0108,len:0x43c
11:16:52.770 -> load:0x403b6000,len:0xbd0
11:16:52.770 -> load:0x403ba000,len:0x29c8
11:16:52.770 -> SHA-256 comparison failed:
11:16:52.770 -> Calculated: 0bb45b01207a4a9e9fca8880ddfc0f79112bfa4d004bd1e6b2eaa75c5323b19d
11:16:52.770 -> Expected: 6eb14ec480aff4f05e39378c41406102ac163573741e63b6cfd73509ef44e3bd
11:16:52.770 -> Attempting to boot anyway...
11:16:52.816 -> entry 0x403b61d8
11:16:52.863 -> CPU Freq = 10 MHz
11:16:52.863 -> XTAL Freq = 40 MHz
11:16:52.863 -> APB Freq = 80000000 Hz
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-08-05:
Have you verified the wanted changed clock speed has been realy changed (osci)?

ehong-tl, 2022-08-08:
May I know how to check the clock speed using oscilloscope? I turn on the Core Debug level to Verbose, it shows the CPU changed to 10 MHz, but the APB still return 80 MHz.

`
11:00:52.053 -> ESP-ROM:esp32s3-20210327
11:00:52.053 -> Build:Mar 27 2021
11:00:52.053 -> rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
11:00:52.053 -> SPIWP:0xee
11:00:52.053 -> mode:DIO, clock div:1
11:00:52.053 -> load:0x3fcd0108,len:0x43c
11:00:52.053 -> load:0x403b6000,len:0xbd0
11:00:52.053 -> load:0x403ba000,len:0x29c8
11:00:52.053 -> entry 0x403b61d8
11:00:52.148 -> [    81][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): XTAL: 40 / 4 = 10 CPU Freq = 10 MHz
11:00:52.148 -> XTAL Freq = 40 MHz
11:00:52.148 -> APB Freq = 80000000 Hz
`