espressif/arduino-esp32 #7087: (Brown-out Reset) crash

hzhh110, 2022-08-05:
### Related area

ESP32-WROOM-32D

### Hardware specification

ESP32-WROOM-32D

### Is your feature request related to a problem?

Is there any other version of the ESP32-WROOM-32D chip that can solve this problem? Many circuits are stuck in this (Brown-out Reset) crash, sometimes restarting, sometimes directly crashing, and other chips are always running normally, and ESP32 crashes directly, a bit embarrassing

### Describe the solution you'd like

Is there any other version of the ESP32-WROOM-32D chip that can solve this problem? Many circuits are stuck in this (Brown-out Reset) crash, sometimes restarting, sometimes directly crashing, and other chips are always running normally, and ESP32 crashes directly, a bit embarrassing

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2022-08-05:
This is as designed. If powering the esp32 is not good (enough) the brownout detector does a reset.
Solution use a decent power supply.

habazut, 2022-08-09:
Google for the PDF 'ESP32 ECO and workarounds' or so. Search in that PDF for "brownout". Looks like a known behaviour that some HW versions will NOT restart but hang instead.

hzhh110, 2022-08-10:
> Google for the PDF 'ESP32 ECO and workarounds' or so. Search in that PDF for "brownout". Looks like a known behaviour that some HW versions will NOT restart but hang instead.

Thank you very much, I don't know what happened, sometimes the device will crash for no reason, and then it will not print information, and it is not timed, there is no way to track it, it is very difficult to reproduce, is there any way to track this phenomenon? I always thought that the power supply Instability problem, but the problem has been solved in the chip V1 version

podaen, 2022-08-22:
Did you try too select an other board in the Arduino IDE?

VojtechBartoska, 2022-12-15:
Hello, I'm closing this issue as won't do, there is not any action item.

closed 2022-12-15 by VojtechBartoska