espressif/arduino-esp32 #7088: Question: is it possible to write multiple items directly?

hasenradball, 2022-08-05:
### Board

ESP32 dev Module

### Device Description

ESP32 Dev Board

### Hardware Configuration

not necessary

### Version

v2.0.4

### IDE Name

Platform IO

### Operating System

WIn 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Ist it possible to do something like that:

```
bool set_Preferences(Pref_Data *data) {
  Preferences obj;
  bool success {false};
  success = obj.begin("pref", false, "nvs");
  String ssid {data->ssid};
  String password {data->password};
  String hostname {data->hostname};
  obj.putString("ssid", ssid);
  obj.putString("password", password);
  obj.putString("hostname", hostname);

  Serial.printf("the SSID is:       %s\n", obj.getString("ssid", "error").c_str());
  Serial.printf("the Password is:   %s\n", obj.getString("password", "error").c_str());
  Serial.printf("the Hostname is:   %s\n\n", obj.getString("hostname", "error").c_str());
  obj.end();
  return success;
}
```

If I do that only the first item ist placed in the preferences...
After the reboot only the first String is written...

[image]

### Sketch

```cpp
#include <Arduino.h>

[... 338 lines cut ...]
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
>>> ESP32 - Raspi-Geek Config-Server Example <<<

the SSID is:       Testssid
the Password is:
the Hostname is:

Prefeerences : 1
 Connection with Wifi_Start() - failed <<<
 WiFI.status =  NO_SSID_AVAIL

 Connection with Wifi_Start_with_Cred() - failed <<<
 WiFI.status =  NO_SSID_AVAIL

AP Accesspoint starting by Wifi_Start_AP()!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-12-16:
Yes, of course you can put multiple entries in nvs.  That's what it is designed for.  Whatever you are doing with Pref_Data is incorrect.

VojtechBartoska, 2023-02-16:
Hello, closing this issues as expired due to no answer.

closed 2023-02-16 by VojtechBartoska