espressif/arduino-esp32 #7089: SPIFS mount issue SPIFFS: mount failed, -10025

ains98, 2022-08-05:
### Board

esp32-s3-devkitc-1

### Device Description

espressif development board esp32-s3-devkitc-1

### Hardware Configuration

Default hardware config

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

WIndows 11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

11500

### Description

ESP32S3 cant mount SPIFS 
I tested it with 2 different dev kit boards and results are the same.
also i assigned in platformio.ini partition table.csv
[partitions.csv](https://github.com/espressif/arduino-esp32/files/9268748/partitions.csv)

### Sketch

```cpp
#include <Arduino.h>
#include <Wire.h>
#include <SPIFFS.h>
#include <FS.h>

void setup()
{
  SPIFFS.begin(1);
  Serial.begin(115200);
  while (!SPIFFS.format())
  {
  }
  Serial.println("formated spifs");
}
```

### Debug Message

```plain
SHA-256 comparison failed:
Calculated: 5830b36eadda2ccdc73ac0028cd665f9a0ae0dcfc1ccf372c71f9cc6bad2ccc4
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
[   178][I][esp32fota.cpp:45] esp32FOTA(): Current firmware version: 1.0.0
E (7) SPIFFS: mount failed, -10025
```

### Other Steps to Reproduce

i don`t know how to reproduce but 2 days ago worked everything in platformIO IDE

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-11:
The partition file here attached is for 8MB Flash (same as `deafult_8MB` option).
It has reserved 1,600KB for SPIFFS.

I tried the sketch and it failed the first time, but and it succeeded in the second boot.

>i don`t know how to reproduce but 2 days ago worked everything in platformIO IDE

Is it working now? There is no issue?

ains98, 2022-08-11:
Still not working .
NVS only works

Jason2866, 2022-08-17:
Have you set in platformio `memory_type` ?
It is mandantory for the S3 to do! Without in most cases a wrong (not working) Flash/PSRAM type default will be used.
You need to know which type of Flash and PSRAM your board has!
So in platformio.ini add to your board
```
build.arduino.memory_type = opi_qspi
```
for example when you have
Flash = opi (octal spi) flash and PSRAM is qio (quad flash)

The possible combinations can be seen [here](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32s3)

AlbertoEusebio, 2022-09-02:
> Have you set in platformio `memory_type` ? It is mandantory for the S3 to do! Without in most cases a wrong (not working) Flash/PSRAM type default will be used. You need to know which type of Flash [...]

I had the same problems with multiple boards I had just bought and this solved the issue.

fatsheep2020, 2023-08-27:
> I had the same problems with multiple boards I had just bought and this solved the issue.

i tried change my platformio.ini, but it seems not take effect. do my change correct?
Thanks!
[image]

Jason2866, 2023-08-27:
Naming has changed it is now `"memory_type": "opi_qio" But before doing more guess work.
What exactly is printed on the metal shield of the esp32-s3 module?

fatsheep2020, 2023-08-28:
> Naming has changed it is now `"memory_type": "opi_qio" But before doing more guess work. What exactly is printed on the metal shield of the esp32-s3 module?

as the YELLOW show, this configure was not recognized (unknown)
[image]

Jason2866, 2023-08-28:
Ahh,sorry I mixed stuff up. It is `opi_qspi` Still what module exactly do you have?

fatsheep2020, 2023-08-28:
> Ahh,sorry I mixed stuff up. It is `opi_qspi` Still what module exactly do you have? Do you have selected a partition scheme with a SPIFFS partition?

could you help to check this issue:　https://github.com/espressif/arduino-esp32/issues/8571

Jason2866, 2023-08-29:
Posted solution in #8571

closed 2025-01-16 by Parsaabasi