espressif/arduino-esp32 #7091: ledcWrite() ignored... unless I spam it!

matthewlai, 2022-08-06:
### Board

Custom board

### Device Description

Custom PCB with ESP32-PICO-V3

### Hardware Configuration

GPIO8 connected to LED sink (active low).

Full schematics: https://github.com/matthewlai/PlantUndeadifier/raw/main/hardware/doc/garden.pdf

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I was having trouble with ledcWrite()s sometimes getting ignored, and managed to simplify it down to the sketch below.

Output set to 0 turns on the LED, and 1 turns off the LED.

With ledcWrite(0, 100% duty cycle), I expect the LED to be turned off. And it does... but only if I call ledcWrite at least 9 times!

Maybe some kind of race condition going on?

### Sketch

```cpp
void setup() {
  ledcSetup(0, 20000, 11);
  ledcAttachPin(8, 0);

  // Doesn't work with fewer than 9 iterations!
  for (int i = 0; i < 9; ++i) {
    ledcWrite(0, (1 << 11) - 1);
  }
}

void loop() {}
```

### Debug Message

```plain
No serial output (boot messages disabled via strapping).
```

### Other Steps to Reproduce

I tried multiple copies of the board and they behave the same.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-08-07:
Hi @matthewlai, can you please update to 2.0.4. version and test it?

matthewlai, 2022-08-12:
Seems to work fine with 2.0.4. Thanks!

closed 2022-08-12 by matthewlai