espressif/arduino-esp32 #7107: 24 _bitsPerSample not handled in I2SClass::read()

musteresel, 2022-08-10:
### Board

ESP32 NodeMCU

### Device Description

Chip on "standard" dev board, doesn't matter for the issue though.

### Hardware Configuration

Irrelevant for issue, but: 2 INMP441 I2S Microphones, SCK+WS+SD connected, SD with 100kOhm to GND (according to datasheet), L/R set differently (3V3 and GND) on both Microphones.
(https://invensense.tdk.com/wp-content/uploads/2015/02/INMP441.pdf)

### Version

v2.0.4

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

https://github.com/espressif/arduino-esp32/blob/3cb73dccd0541b2222e0341e7261754ef4b52bb2/libraries/I2S/src/I2S.cpp#L584-L587

24 `_bitsPerSample` are silently NOT read. 

### Sketch

```cpp
// in setup
I2S.begin(I2S_PHILIPS_MODE, samplerate, 24);

// in loop
int sample = I2S.read();
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

musteresel, 2022-08-10:
For reference, this is how read 24 bits per sample (stereo) works atm:

~~~c++
int32_t s24le_to_int32(unsigned char const * const buf)
{
  uint32_t r = (buf[2] << 16) | (buf[1] << 8) | buf[0];
  if (buf[2] & 0x80) r |= 0xFF000000; // extend sign
  return static_cast<int32_t>(r);
}

unsigned char snd[8];
int const stat = I2S.read(snd, 8);
// snd[0], snd[4] ==> discard
// snd[3,2,1] => 24 bit signed little endian
// snd[7,6,5] => 24 bit signed little endian
int32_t const left = s24le_to_int32(snd + 1);
int32_t const right = s24le_to_int32(snd + 5);
~~~

So basically you need to read 4 bytes for each sample and channel.

PilnyTomas, 2022-08-12:
Hi, 24bps is not officially supported. Reading data as a buffer should return something.

PilnyTomas, 2022-11-09:
Full support will be available when I finish the PR #7117.

closed 2022-11-09 by PilnyTomas