espressif/arduino-esp32 #7108: WiFi Provisioning esp_prov.py usage

HamzaHajeir, 2022-08-10:
### Related area

Usage of ESP-IDF tools

### Hardware specification

ESP32

### Is your feature request related to a problem?

I'm facing an issue using [esp_prov](https://github.com/espressif/esp-idf/tree/30e8f19f5a/tools/esp_prov) tools that's in esp-idf.

The tool requires the IDF_PATH to be set as environment variable.
Also it expects a specific folder hierarchy that's similar to ESP-IDF framework.

### Describe the solution you'd like

As I imagine, the solution would be to bring up the tool to the current project after tweaking its parameters.

### Additional context

Here's an output of the command:

```
PS C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning> ls

    Directory: C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning

Mode                 LastWriteTime         Length Name
d-----          9/8/2022   2:17 PM                proto
d-----          9/8/2022   2:17 PM                prov
d-----          9/8/2022  11:14 AM                security
d-----          9/8/2022  11:14 AM                transport
d-----          9/8/2022   2:17 PM                utils
-a----          9/8/2022  11:14 AM          20176 esp_prov.py
-a----          9/8/2022  11:14 AM           6124 README.md
-a----          9/8/2022  11:14 AM            178 __init__.py

PS C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning> python esp_prov.py
Traceback (most recent call last):
  File "C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning\esp_prov.py", line 17, 
in <module>
    import prov
  File "C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning\prov\__init__.py", line 6, in <module>
    from .wifi_prov import *  # noqa F403
  File "C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning\prov\wifi_prov.py", line 7, in <module>
    import proto
  File "C:\Users\USER\Documents\PlatformIO\Projects\ESP32 Test\ESP Provisioning\proto\__init__.py", line 24, in <module>
    idf_path = os.environ['IDF_PATH']
  File "C:\Users\USER\AppData\Local\Programs\Python\Python39\lib\os.py", line 679, in __getitem__      
    raise KeyError(key) from None
KeyError: 'IDF_PATH'
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

maxakash, 2024-02-16:
@HamzaHajeir I am facing the same error. is there any workaround?

HamzaHajeir, 2024-02-16:
Hi Akash, unfortunately I've gone with this path, I was trying to figure
out available WiFi provisioning methods backthen.

closed 2025-01-16 by Parsaabasi