espressif/arduino-esp32 #7115: BLE server connection callbacks being made when BLE client connects or disconnects

AlexMompoint, 2022-08-11:
### Board

Tiny Pico

### Device Description

Just Tiny pico using usb power

### Hardware Configuration

I have two other esp32 boards, one acting as a server and one acting as a client just to connect and disconnect from the one I am experiencing the issue I have.

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I was given a task use the esp32 as both a server and a client to allow some device to connect to it, and as well to reach out to a server and connect to it. I attempted performing this using the BLE libraries in the arduino esp32. 

If the server portion receives a connect, or disconnect, it functions as expected, only the server callbacks are made, but when the client portion connects, or disconnects, it performs the call back for both server and client meaning I can't really use the server callback as just a server callback.

### Sketch

```cpp
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

#define SERVICE_UUID1           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID

bool doScan = true;
bool deviceConnected = false;

BLEClient *pClient;
BLEServer *pServer;
BLEScan *pBLEScan;
std::string deviceName = "serverDevice";

class MyServerCallbacks: public BLEServerCallbacks {
  void onConnect(BLEServer* pServer) {
    deviceConnected = true;
    Serial.println("***** Connect");
  }
  void onDisconnect(BLEServer* pServer) {
    Serial.println("***** Disconnect");
    deviceConnected = false;
    BLEDevice::startAdvertising();
  }
[... 219 lines cut ...]
12:05:46.293 -> [ 13794][D][BLEServer.cpp:367] onConnect(): BLEServerCallbacks
12:05:46.293 -> [ 13802][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13810][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13819][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13829][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> Client Connected *****
12:05:46.902 -> [ 14438][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.902 -> [ 14440][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:48.311 -> Not Scanning
12:05:50.291 -> Not Scanning
12:05:52.305 -> Not Scanning
```

### Other Steps to Reproduce

1.  Create both BLE server and BLE Client with respective callbacks
2.  Start Server
3.  Have BLEClient connect to another device
4.  Get both callbacks
5.  Disconnect Client
6.  Get Both callbacks

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

chegewara, 2022-08-11:
> After some more testing the Server doesn't even have to be started the callbacks just need to be set.

How did you test it? Because from your code i see the server is started:
```
pServer = BLEDevice::createServer();
```
Also from your logs i dont see anything wrong, could you isolate the logs you think are important and to describe what you do and what you expect to see or not suppose to see?

AlexMompoint, 2022-08-12:
```
12:05:46.151 -> [ 13681][I][BLEDevice.cpp:622] addPeerDevice(): add conn_id: 1, GATT role: client
12:05:46.151 -> [ 13682][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.151 -> [ 13690][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13783][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
12:05:46.293 -> ***** Connect
12:05:46.293 -> [ 13784][D][BLEServer.cpp:365] onConnect(): BLEServerCallbacks
12:05:46.293 -> [ 13789][D][BLEServer.cpp:366] onConnect(): BLEServerCallbacks
12:05:46.293 -> [ 13794][D][BLEServer.cpp:367] onConnect(): BLEServerCallbacks
12:05:46.293 -> [ 13802][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13810][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13819][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> [ 13829][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.293 -> Client Connected *****
12:05:46.902 -> [ 14438][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:46.902 -> [ 14440][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
12:05:48.311 -> Not Scanning
```

I apologize, when I seem to have misunderstood what it means to start the server, but regardless, In this section here I believe it to be what I am trying to discuss.

My expectation is that when I use the BLEClient to connect to a server while I also have a BLEServer running on the same device, I receive only a client callback which would print "Client Connected *****", but what I am getting is the server callback as well as denoted by the the server call back which would print "***** Connect".

AlexMompoint, 2022-08-30:
bump, I seem to have found somethin from chegewara, but the solution appears to be quite outdated, so it seems to have been an issue in the past. 
https://www.esp32.com/viewtopic.php?t=12191

chegewara, 2022-09-01:
It is not outdated at all. I would use it by myself to debug the problems with BLE.

patrickleboutillier, 2023-01-25:
I'm seeing the same behavior in a sketch that has both a BLEClient and a BLEServer defined. Server callbacks are being called on Client connect/disconnect events.

AlexMompoint, 2023-01-25:
@patrickleboutillier Yes I was experiencing this issue continuously using the arduino esp32 BLE library. Although not an optimal solution, I used a different library nimBLE for arduino, which fixed the issue with minimal required changes to the source code I was working with. I know this is not much of a help, just a band-aid fix since it does not deal with the issue, but it did fix the issue.

Berg0162, 2023-02-01:
I can confirm as well that I am confronted with the same behavior, very consistently and very reproducibly! 
I have one BLE client and one server that both/separately have defined their respective onConnect and onDisconnect callback methods. I describe the behavior that I notice.
When the client connects for the first time it calls: client_Connection_Callbacks::onConnect(BLEClient* pClient).
When the client disconnects for the first time it calls: client_Connection_Callbacks::onDisconnect(BLEClient* pClient),  like is to be expected, and no strange behavior!
However, the next time the client connects again, it calls first of all:
server_Connection_Callbacks::onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t \*param)
and only thereafter: client_Connection_Callbacks::onConnect(BLEClient* pClient).
The very same (mirrored) behavior can be observed when it disconnects again (after its first disconnect) : first of all Server::onDisconnect and only thereafter Client::onDisconnect is called... 
To end up with workable code I had to catch and identify these unexpected pClient calls to avoid a complete mix-up!
I figured out that **esp_ble_gatts_cb_param_t** is filled with data of pClient and that is very helpful to identify the nature/owner of the calls, is it pClient calling? or is it pServer calling?......

see a code snippet:
```c++
void server_Connection_Callbacks::onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {
    // Get some connection parameters of the peer device.
    char fullMacAddress[18] = {}; //
    uint32_t count = pServer->getConnectedCount();
    uint16_t ConnectionHandle = param->connect.conn_id;
    uint8_t RemoteAddress[6];
    memcpy(&RemoteAddress, param->connect.remote_bda, 6);
    ConvertMacAddress(fullMacAddress, RemoteAddress, true); // true -> Native format!
    DEBUG_PRINTF("Server onConnect Callback --> Conn cnt: [%d] Conn handle: [%d] Address: [%s]\n", count, ConnectionHandle, fullMacAddress);
    // For some reason Server_Connection_Callbacks::onConnect is not only called for pServer but 
    // also called when a pClient is Connecting...???? Bug or Feature --> handle it to avoid inconsistencies....!!
    if (memcmp(RemoteAddress, Trainer.PeerAddress, 6) == 0 ) { // Check for Trainer MAC address Notice: pClient_FTMS->getConnId() == 255
      DEBUG_PRINTLN(">>> Client will Connect soon with Server device!");
      return; // Let client Connect happen....
    }   
    BLEDevice::stopAdvertising();
    // Who has been exactly connected?
    DEBUG_PRINTF("ESP32 Server connected to Client device with MAC Address: [%s] Handle: [%d]\n", fullMacAddress, param->connect.conn_id); 
etcetera

void server_Connection_Callbacks::onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {
    uint32_t count = pServer->getConnectedCount();
    // Get some Disconnection parameters of the peer device.
    uint16_t ConnectionHandle = param->connect.conn_id;
    uint8_t Reason = param->disconnect.reason;
    uint8_t RemoteAddress[6] = {};
    memcpy(&RemoteAddress, param->connect.remote_bda, 6);
    char fullMacAddress[18] = {}; //
    ConvertMacAddress(fullMacAddress, RemoteAddress, true); // true -> Native format!
    DEBUG_PRINTF("Server onDisconnect Callback --> Conn cnt: [%d] Conn handle: [%d] Address: [%s] Reason: [%d]\n", count, ConnectionHandle, fullMacAddress, Reason);
    // For some reason Server_Connection_Callbacks::onDisconnect is not only called for pServer but 
    // also called when a pClient is disconnecting...???? Bug or Feature --> handle it to avoid inconsistencies....!!
    if (memcmp(RemoteAddress, Trainer.PeerAddress, 6) == 0 ) { // Check for Trainer MAC address
      DEBUG_PRINTLN(">>> Client will disconnect soon from Server device!");
      return; // Let client disconnect happen....
    }
    if (Laptop.conn_handle == ConnectionHandle ) { // Laptop/Desktop is disconnected
 etcetera
```
see the debug_print output and compare with above code (notice that the pServer connection is meanwhile doing its work, and also reporting!)
> .
.
15:43:56.684 -> Set Indoor Bike Simulation Parameters!
15:43:56.684 -> Wind speed (1000): 0.00 | Grade (100): 0.00 | Crr (10000): 0.00 | Cw (100): 0.51
15:43:56.718 ->  -> Client Rec'd Raw FTM Control Point Response Data: [0] [3] [ 80 11 01 ]
15:44:34.303 -> Server onDisconnect Callback --> Conn cnt: [2] Conn handle: [0] Address: [FD:6C:BD:F3:50:95] Reason: [8]
15:44:34.336 -> >>> Client will disconnect soon from Server device!
15:44:34.336 -> Client Disconnected from Server device with Name: [Sim nRF52] Mac Address: [FD:6C:BD:F3:50:95]!
15:44:34.336 -> Client Starts Scanning again for Server Device with CPS, CSC and FTMS!
.
.
15:44:43.438 -> Set Indoor Bike Simulation Parameters!
15:44:43.438 -> Wind speed (1000): 0.00 | Grade (100): 0.19 | Crr (10000): 0.00 | Cw (100): 0.51
15:44:43.890 -> Central Updated CCCD --> HR_Measurement Notify Enabled!
15:44:45.168 -> Found advertising Peripheral with FTMS enabled! See data:
15:44:45.211 -> Name: Sim nRF52, Address: fd:6c:bd:f3:50:95, serviceUUID: 00001818-0000-1000-8000-00805f9b34fb, serviceUUID: 00001816-0000-1000-8000-00805f9b34fb, serviceUUID: 00001826-0000-1000-8000-00805f9b34fb, txPower: 4, rssi: -39
15:44:45.288 ->  --> Raw FTM Control Point Data [len: 7]  [OpCode: 11] [Values: 00 00 13 00 28 33 00 ]
15:44:45.288 -> Set Indoor Bike Simulation Parameters!
15:44:45.288 -> Wind speed (1000): 0.00 | Grade (100): 0.19 | Crr (10000): 0.00 | Cw (100): 0.51
[... cut ...]  Conn handle: [1] Address: [FD:6C:BD:F3:50:95]
15:44:45.451 -> >>> Client will Connect soon with Server device!
15:44:45.451 -> ESP32 Client connected to Server device with Name: [Sim nRF52] MAC Address: [FD:6C:BD:F3:50:95] Handle: [1] MTU: [20]
15:44:45.451 -> Now checking all Client Services and Characteristics!
15:44:45.451 -> If Mandatory Services Fail --> the Client will disconnect!
15:44:45.490 -> Client Generic Access: Found!
15:44:46.847 ->  -> Client Reads Device Name:   [Sim nRF52]
.
.

Berg0162, 2023-02-01:
> bump, I seem to have found somethin from chegewara, but the solution appears to be quite outdated, so it seems to have been an issue in the past. https://www.esp32.com/viewtopic.php?t=12191

I have tested this suggestion but I could NOT find any change in behavior during connect and/or disconnect!

closed 2023-02-15 by me-no-dev

frank-lin-Sunon, 2023-10-26:
I still have same promble when using sdk v2.0.14