espressif/arduino-esp32 #7116: On certain MQTT received messages: assert failed: xQueueGenericSend queue.c:820 (pxQueue)

konacurrents, 2022-08-12:
### Board

M5Stack-Tiny-CAM

### Device Description

Just the M5 Timer X.

### Hardware Configuration

Nothing attached

(I don't know how to find the Arduino ESP32 version), but I assume it's the latest.

### Version

v2.0.4

### IDE Name

Arduino IDE 1.8.19

### Operating System

maxOSx 12.4

### Flash frequency

240mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Basically I have ESP code that works with ESP32 and the M5 chip nicely. It has BLE Server/Client, MQTT, WIFI, AP mode etc.

So I was successful compiling for the M5Stack-Tiny-CAM, and it receives MQTT messages from my network.

But for a couple of my messages, I get the 

```
[ 50715][E][Wire.cpp:319] beginTransmission(): could not acquire lock
[ 50722][E][Wire.cpp:370] requestFrom(): Unfinished Repeated Start transaction! Expected address do not match! 104 != 0
```
then it processes my MQTT message, then:
```
 50859][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
```

```
assert failed: xQueueGenericSend queue.c:820 (pxQueue)
```

# Previously ran in other threads
In some cases, my MQTT code was trying to use BLE messaging, and the thread /task had to be released, so that the BLEClient code could run (versus just calling that code). 

In this case I don't see that issue.

# Not happening on my other 2 builds. The errors above don't even show up.

Also the code doesn't even have to do anything, but the error still occurs:

```
[ 41605][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[... 472 lines cut ...]
main_saveWIFICredentials{'ssid':'SunnyWhiteriver','ssidPassword':'sunny2021'}
wifi_apModule.updateMQTTInfo(SunnyWhiteriver, sunny2021, scott@konacurrents.com,PASSWD, doggy, In My Pocket on Hike)
calling .. processJSONMessage
DeviceName: M5Camera
[  1882][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  1889][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  1888][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[  1895][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[  1963][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: SunnyWhiteriver, BSSID: 88:de:7c:a3:bc:36, Channel: 11, Auth: WPA2_PSK
[  1966][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1990][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.0.244
[  1991][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1994][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.0.244, MASK: 255.255.255.0, GW: 192.168.0.1
WiFi.status() == WL_CONNECTED()
finishWIFI_Setup()
setupMQTTNetworking_WIFI done..
callPreSetupMQTT(2)
**** setupMQTT *****
Attempting MQTT connection: 'idogwatch.com' 'scott@konacurrents.com' 'PASSWD' count=1
attempt _mqttClient.connect(espM5-d9d5)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-08-12:
Please note in the future that this forum is for issues with the code in this repository, not issues with your code.  [esp32.com](http://esp32.com) is a more open conversation about all manner of esp32 questions.

You say you are having problems with mqtt, but the errors quite directly point at Wire.  My guess is that you have your data collection (from some I2C device) and data delivery intermingled.  Separate them so that you have the data in hand before you make the network connection.  That will reduce the possibility of timeouts and race conditions.  It also makes it easier to add new collection and delivery processes.  The more asynchronous you can make these steps, the more robust your application will be overall.

You can [decode the backtrace](https://github.com/me-no-dev/EspExceptionDecoder) you are getting to pin down exactly at what line in the code the error is occurring.  I think you will see that it is happening while some device is polling the wire.  See if there isn't a better driver for that device out there that is better suited for esp32 (or perhaps more recently updated).

konacurrents, 2022-08-12:
Thanks for the reply and pointer to the backtrace debugging approach. I'll look into that. 

I assumed that since the error I'm getting was not my code (like Wire) that this forum was appropriate. 

Would you suggest that a data collection (like MQTT) - store that data, and release the interrupt, then in the next loop() process the message? That's what I ended up doing with BLE messages.

thanks.

konacurrents, 2022-09-06:
That M5 camera (on right in pic) is a self contained unit so there is no wiring on my part. 

[image]

podaen, 2022-09-06:
Its wireless... so where is the wire.h in use?

konacurrents, 2022-09-06:
I assume the internals of the esp chip to the camera.

I just need a good approach to debug, which I haven't figured out on any of my esp projects.

podaen, 2022-09-07:
Oh. I see now. I don't am exprenced engough with M5 stack. I can only point you in a direction what I think it could be it.

Did you check the freehap of the allocated memorie or maybe somthing is missing in your code that is needed to use the library. How long is the message you are sending to the MQTT? Sending small msg improves the performance.

podaen, 2022-09-08:
Are you using this library?

> /*
>  PubSubClient.h - A simple client for MQTT.
>   Nick O'Leary
>   http://knolleary.net
> */

Then check these parameters in the MQTT library if they are valid for you application.

```
// MQTT_MAX_PACKET_SIZE : Maximum packet size. Override with setBufferSize().
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 256
#endif

// MQTT_KEEPALIVE : keepAlive interval in Seconds. Override with setKeepAlive()
#ifndef MQTT_KEEPALIVE
#define MQTT_KEEPALIVE 15
#endif

// MQTT_SOCKET_TIMEOUT: socket timeout interval in Seconds. Override with setSocketTimeout()
#ifndef MQTT_SOCKET_TIMEOUT
#define MQTT_SOCKET_TIMEOUT 15
#endif
```

VojtechBartoska, 2022-12-15:
is this still valid?

konacurrents, 2022-12-17:
Yes sadly I still get this issue.

podaen, 2022-12-18:
Intresting. I am using now an other mqqt lib with events and I have the same issue.

VojtechBartoska, 2024-06-05:
Hello @konacurrents asi Arduino Team, we are not really able to help here. Is your issue still relevant?

closed 2025-01-16 by Parsaabasi