espressif/arduino-esp32 #7121: C++17 ifdefs are wrong

atlaste, 2022-08-12:
### Board

ESP32 all

### Device Description

All ESP32's

### Hardware Configuration

Irrelevant

### Version

v2.0.4

### IDE Name

Platformio

### Operating System

Windows

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Portmacro.h has a few pieces of code:

```c++
#if defined(__cplusplus) && (__cplusplus >  201703L)
```

Fun fact, when setting C++17 mode, the version of __cplusplus is 201703. So, it should have been:

```c++
#if defined(__cplusplus) && (__cplusplus >= 201703L)
```

After that, everything works according to plan...

### Sketch

```cpp
Again, irrelevant
```

### Debug Message

```plain
Irrelevant
```

### Other Steps to Reproduce

Compile anything with C++17

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-12:
@atlaste - I found this explanation in the code:

``` cpp
/**
 * In the future, we want to switch to C++20. We also want to become compatible with clang.
 * Hence, we provide two versions of the following macros which are using variadic arguments.
 * The first one is using the GNU extension \#\#__VA_ARGS__. The second one is using the C++20 feature __VA_OPT__(,).
 * This allows users to compile their code with standard C++20 enabled instead of the GNU extension.
 * Below C++20, we haven't found any good alternative to using \#\#__VA_ARGS__.
 */
```

SuGlider, 2022-08-12:
Check this also:

https://stackoverflow.com/questions/48045470/portably-detect-va-opt-support

closed 2022-12-15 by VojtechBartoska

dhebbeker, 2023-05-20:
For those who also stumble upon this issue while attempting to compile with C++17: Compile with `-std=gnu++17`.

For PlatformIO I added these lines to the environment in `platformio.ini`:

```ini
build_flags = -std=gnu++17
build_unflags = -std=gnu++11
```

With those I was able to build files including `portmacro.h` with C++17 (GNU C++17).
Also in `.vscode/c_cpp_properties.json` the setting is automatically set correctly:

```JSON
            "cppStandard": "c++17",
```