espressif/arduino-esp32 #7122: 8meg app partition size

manecolooper, 2022-08-12:
### Board

ESP32 wrover IE 16 meg flash

### Device Description

module on custom pcb , working perfectly

### Hardware Configuration

switches and pots to io pins

### Version

v1.0.6

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80MHZ

### PSRAM enabled

no

### Upload speed

921600

### Description

region `drom0_0_seg' overflowed by 2895568 bytes

when trying to compile a 8 megas app on corrected sized partitions 

### Sketch

```cpp
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x800000,
eeprom,   data, 0x99,    0x810000,0x1000,
ffat,     data, fat,     0x811000,0x17000,
```

### Debug Message

```plain
no debug data as it shows at compile time
```

### Other Steps to Reproduce

tried different machines with arduino

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-08-12:
seems like a duplicate of #7113, PR is already open.

manecolooper, 2022-08-13:
#7113  refers to wrong labelling, this is a real error

Jason2866, 2022-08-14:
No support for core 1.0.6. anymore. Update to latest core 2.0.4

manecolooper, 2022-08-14:
updated to 2.04, exactly the same error

closed 2025-01-16 by Parsaabasi