espressif/arduino-esp32 #7123: Arduino as Component compile error

joeholdsworth, 2022-08-13:
### Board

ESP32c3

### Device Description

DevKitC

### Hardware Configuration

No

### Version

v2.0.4

### IDE Name

Visual studio ESP-IDF plugin

### Operating System

Windoes 10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Compiling my project with aduino as a componet fails with this error:

esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-uart.c.obj -c ../components/arduino/cores/esp32/esp32-hal-uart.c
../components/arduino/cores/esp32/esp32-hal-uart.c: In function 'uartAvailableForWrite':
../components/arduino/cores/esp32/esp32-hal-uart.c:242:19: error: implicit declaration of function 'uart_get_tx_buffer_free_size'; did you mean 'uart_get_buffered_data_len'? [-Werror=implicit-function-declaration]
     if (ESP_OK == uart_get_tx_buffer_free_size(uart->num, &txRingBufferAvailable)) {
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   uart_get_buffered_data_len
cc1.exe: some warnings being treated as errors
[33/164] Building CXX object esp-idf/arduino...f_arduino.dir/cores/esp32/MD5Builder.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

### Sketch

```cpp
this occurs on build, it is an error when attempting to compile the ardino as a component which occurs before the main.cpp is compiled

idf.py build flash monitor -p COM3
```

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

I followed the instructions in:

https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

specifically:
mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
idf.py menuconfig

after configuring the arduino component I then attempted a build and got the above error

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-14:
@joeholdsworth - 

>../components/arduino/cores/esp32/esp32-hal-uart.c:242:19: error: implicit declaration of function 'uart_get_tx_buffer_free_size'; did you mean 'uart_get_buffered_data_len'? [-Werror=implicit-function-declaration]
if (ESP_OK == uart_get_tx_buffer_free_size(uart->num, &txRingBufferAvailable)) {
^~~~~~~~~~~~~~~~~~~~~~~~~~~~

The IDF function used in the master branch from Arduino Core Github, `uart_get_tx_buffer_free_size()`, has been included recently to IDF 4.4.2.

Please update your IDF to the latest master for IDF 4.4.2 from Github and  after that it will work fine.

joeholdsworth, 2022-08-15:
Thank you for the information! This means I will need two installations and two environments. I am using Visual Studio Code, I have had a look for how to do this but can only find instructions for a fresh install, can you point me in the direction for the instructions on how to do this and use Visual Studio Code and switch between the two environments?

SuGlider, 2022-11-05:
This issue is solved in Arduino Core 2.0.5 that uses IDF 4.4.2.

closed 2022-11-05 by SuGlider