espressif/arduino-esp32 #7125: Arduino IDE result in bootloop

LaudixGit, 2022-08-13:
### Board

UM TinyS3 & UM FeatherS3

### Device Description

Using a Raspberry Pi 4B
New, 2020.4.04, desktop  image (fails with both 32-bit and 64-bit)
ESP32S3 devices connected via USBC cable (know data cable. see, later, successful serial communications)

### Hardware Configuration

brand new, just unpacked ESP32S3 boards - no modifications no additions

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Linux RPi4Arduino 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

### Flash frequency

QIO

### PSRAM enabled

yes

### Upload speed

921600

### Description

Failure: After successful upload of any sketch, the device continuously reboots.

Expected: Device must reliably and successfully execute code.

cross-referencing:
[6013](https://github.com/espressif/arduino-esp32/issues/6013)
[6519](https://github.com/espressif/arduino-esp32/issues/6519)
[6661](https://github.com/espressif/arduino-esp32/issues/6661)

note: the[ UnexpectedMaker team](https://discord.com/channels/605621786616528915/605622154037428240) assisted in isolating this issue

### Sketch

```cpp
void setup() {
Serial.begin(115200);
}
void loop() {
Serial.println("test");
delay(5000);
}
```

### Debug Message

```plain
this is written to the serial port after any upload.
(redundant in case the timestamps helps understand the frequency)

14:29:18.102 -> ESP-ROM:esp32s3-20210327
14:29:18.102 -> Build:Mar 27 2021
14:29:18.102 -> rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
14:29:18.102 -> Saved PC:0x403bb22e
14:29:18.102 -> SPIWP:0xee
14:29:18.102 -> mode:DIO, clock div:1
14:29:18.102 -> load:0x3fcd0108,len:0x43c
14:29:18.102 -> load:0x403b6000,len:0xbd0
14:29:18.102 -> load:0x403ba000,len:0x29c8
14:29:18.102 -> entry 0x403b61d8
14:29:18.102 -> ESP-ROM:esp32s3-20210327
14:29:18.102 -> Build:Mar 27 2021
14:29:18.102 -> rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
14:29:18.102 -> Saved PC:0x403bb22e
14:29:18.136 -> SPIWP:0xee
14:29:18.136 -> mode:DIO, clock div:1
14:29:18.136 -> load:0x3fcd0108,len:0x43c
14:29:18.136 -> load:0x403b6000,len:0xbd0
14:29:18.136 -> load:0x403ba000,len:0x29c8
14:29:18.136 -> entry 0x403b61d8
14:29:18.136 -> ESP-ROM:esp32s3-20210327
```

### Other Steps to Reproduce

Same results with ESP v2.0.3, v2.0.4, and with RC
Same results with 32-bit and 64-bit version of OS.

Note: ESP32S3 device comes preloaded with neopixel sketch; which worked fine until the upload of a sketch

Note: Using VS Code with PlatfomIO extension, succeeds on the host (no other changes needed).
(e.g., failure is only when using Arduino IDE)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

LaudixGit, 2022-08-13:
Worth noting.  The SD Card for the Raspberry Pi host was created Aug 12, 2022. 
Ran (sudo apt update && sudo apt -y full-upgrade ) before any installation.
rebooted and then installed Arduino and ESP32 boards

wget https://downloads.arduino.cc/arduino-1.8.19-linuxaarch64.tar.xz

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

IOW: everything was freshly installed - no bloat, no residuals

LaudixGit, 2022-08-13:
Also worth noting.  The environment succeeds using all the above, but targeting an older board, wESP32.

e.g., failure seems limited to the ESP32**S3**

Jason2866, 2022-08-14:
Flash type correctly selected and set?
S3 bootloops if flash type for Flash and PSRAM is not 100% correct.

LaudixGit, 2022-08-14:
Hmm. not sure what the correct settings are, but same bootloop occurs with either Flash option, and with with PSRAM enabled or disabled.
But maybe I've got something else wrong? (essentially I kept the defaults the board set when I selected it)

[image]

LaudixGit, 2022-08-14:
My settings are identical to the ones suggested by the device manufacturer on [their getting-started page](https://esp32s3.com/getting-started.html).
[image]

But if different settings will make this work, I am willing to try.  I've attempted a dozen permutations with no better results.

Jason2866, 2022-08-14:
@UnexpectedMaker Can you help?

Jason2866, 2022-08-14:
Regarding https://esp32s3.com/tinys3.html it has 8 MB QIO flash and psram. So choose 8 MB partition (you did not) and enable PSRAM with mode QIO too. Reduce Upload speed to "115200"

UnexpectedMaker, 2022-08-14:
> @UnexpectedMaker Can you help?

I have been helping. I asked the OP to open this issue. Definitely a problem unrelated to HW or tools menu config. Same settings work for Win, Mac. There are build issues on Linux flavours. Other open issues for Ubuntu. Lots of users on my discord unable to use Linux for Arduino dev. 

Something isn’t right.

Jason2866, 2022-08-15:
@UnexpectedMaker  Have you tried with Platformio?

UnexpectedMaker, 2022-08-15:
> @UnexpectedMaker Have you tried with Platformio?

Works for the OP in PIO, not in Arduino IDE. Asking the OP is better than me as he created the issue and has all of the info.

Jason2866, 2022-08-15:
Thx. I am out, since it is working with PlatformIO. Not using ArduinoIDE.

SuGlider, 2022-08-15:
@LaudixGit - It seems from the discussion that the issue is related to ARM64 Linux only and Arduino IDE. Is it right?

VojtechBartoska, 2022-08-15:
Maybe @me-no-dev can help us with that when he is back.

LaudixGit, 2022-08-15:
> related to ARM64 Linux only and Arduino IDE. Is it right?

@SuGlider  Nope:  Same bootloop results with both 32-bit and 64-bit OS when using Arduino IDE.
(and, for the record, PIO works on both OS versions)

SuGlider, 2022-08-16:
@LaudixGit - Could you please try this other Arduino IDE configuration:

There are a few changes to the one you posted, related to OTG/UART0/CDC.

[image]

LaudixGit, 2022-08-16:
@SuGlider Same results - bootlooping

[image]

LaudixGit, 2022-08-16:
@SuGlider same results - bootlooping
(interesting to note that it loops slower though)

[image]

SuGlider, 2022-08-16:
@LaudixGit - but it works fine with PlatformIO, right?

Could you please send me your PIO setup and configuration?

LaudixGit, 2022-08-16:
@SuGlider I am very new to PIO (1st install 2 days ago).  I am not sure where setup and config are, but the files I know of are attached.
And, yes, PIO successfully compiles and uploads; the TinyS3 runs fine. (and so does the FeatherS3)

[platformio.ini.txt](https://github.com/espressif/arduino-esp32/files/9353897/platformio.ini.txt)
[main.cpp.txt](https://github.com/espressif/arduino-esp32/files/9353904/main.cpp.txt)

SuGlider, 2022-08-22:
We need to investigate it further... It may be related to second-stage bootloader used in Arduino.

me-no-dev, 2022-08-22:
Has the board config been checked? Maybe lingering bootloader.bin in the variant folder? Errors seem to point that bootloader was not found or is not valid. Last log even suggests that partition scheme was not found either.

me-no-dev, 2022-08-22:
The problem is in esptool. Versions differ slightly between the ones for win+mac and linux. You can get the correct esptool.py from inside [this archive](https://github.com/espressif/esptool/archive/refs/tags/v3.3.tar.gz) and replace the one in `~/.arduino15/packages/esp32/tools/esptool_py/3.3.0/`. This issue will be resolved in the next version (esptool is updated).

UnexpectedMaker, 2022-08-22:
> The problem is in esptool. Versions differ slightly between the ones for win+mac and linux. You can get the correct esptool.py from inside [this [...]

Oh wicked!!! Thanks heaps for getting this resolved - will be lots of happy S3 Linux users now!

LaudixGit, 2022-09-05:
Before making any changes, I downloaded and extracted [the archive](https://github.com/espressif/esptool/archive/refs/tags/v3.3.tar.gz).

Then listed my original files and the newly  downloaded ones.
I was very surprised to see my original files are all the identical size.  That seems suspicious.

__My orig files__
laudix@RPiArduino:~ $ ls -l ~/.arduino15/packages/esp32/tools/
total 28
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:56 **esptool_py**
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:56 mklittlefs
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:56 mkspiffs
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:54 riscv32-esp-elf-gcc
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:54 xtensa-esp32-elf-gcc
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:55 xtensa-esp32s2-elf-gcc
drwxr-xr-x 3 laudix laudix 4096 Aug 12 22:55 xtensa-esp32s3-elf-gcc

__Newly downloaded files__
laudix@RPiArduino:~ $ ls -l esptool-3.3/
total 408
drwxr-xr-x 2 laudix laudix   4096 Mar 22 05:07 build_tools
-rw-r--r-- 1 laudix laudix   5208 Mar 22 05:07 CONTRIBUTING.rst
drwxr-xr-x 4 laudix laudix   4096 Mar 22 05:07 docs
-rwxr-xr-x 1 laudix laudix   8273 Mar 22 05:07 espefuse.py
drwxr-xr-x 3 laudix laudix   4096 Mar 22 05:07 espressif
-rwxr-xr-x 1 laudix laudix   9902 Mar 22 05:07 esp_rfc2217_server.py
-rwxr-xr-x 1 laudix laudix  51917 Mar 22 05:07 espsecure.py
-rwxr-xr-x 1 laudix laudix 268415 Mar 22 05:07 **esptool.py**
drwxr-xr-x 4 laudix laudix   4096 Mar 22 05:07 flasher_stub
-rw-r--r-- 1 laudix laudix  18092 Mar 22 05:07 LICENSE
-rw-r--r-- 1 laudix laudix    246 Mar 22 05:07 MANIFEST.in
-rw-r--r-- 1 laudix laudix   1371 Mar 22 05:07 README.md
-rw-r--r-- 1 laudix laudix   1070 Mar 22 05:07 setup.cfg
-rw-r--r-- 1 laudix laudix   4609 Mar 22 05:07 setup.py
drwxr-xr-x 7 laudix laudix   4096 Mar 22 05:07 test

LaudixGit, 2022-09-05:
Replacing the esptool.py file, as suggested, did *not* make any difference.

**My environment**
[image]

[image]

**My Code**
`void setup() {
  // put your setup code here, to run once:
Serial.begin(11520);
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println("test");
delay(5000);
}`

**The steps I took**

**Before using the archived file**
- test the TinyS3; programmed via PlatformIO - success
- using ArduinoIDE upload sketch to TinyS3 - FAIL, bootloop

**Making changes**
- wget https://github.com/espressif/esptool/archive/refs/tags/v3.3.tar.gz
- tar -xzf  v3.3.tar.gz
- cp ~/esptool-3.3/esptool.py ~/.arduino15/packages/esp32/tools/

**File listing after change**
laudix@RPiArduino:~ $ ls -l ~/.arduino15/packages/esp32/tools/
total 292
-rwxr-xr-x 1 laudix laudix 268415 Sep  5 10:41 esptool.py
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:56 esptool_py
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:56 mklittlefs
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:56 mkspiffs
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:54 riscv32-esp-elf-gcc
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:54 xtensa-esp32-elf-gcc
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:55 xtensa-esp32s2-elf-gcc
drwxr-xr-x 3 laudix laudix   4096 Aug 12 22:55 xtensa-esp32s3-elf-gcc

**Results**
- using ArduinoIDE upload sketch to TinyS3 - FAIL, bootloop

LaudixGit, 2022-09-05:
It occurred to me I followed the suggestion too literally and the .py file didn't end up in the correct folder
so.

 cp ~/esptool-3.3/esptool.py ~/.arduino15/packages/esp32/tools/esptool_py/3.3.0/

**Results**
- using ArduinoIDE upload sketch to TinyS3 - **SUCCESS**!

**Update the ESP32 platform**
[image]

**Before**
laudix@RPiArduino:~ $ ls -l ~/.arduino15/packages/esp32/tools/esptool_py/3.3.0/
total 264
-rwxr-xr-x 1 laudix laudix 267470 Mar 29 03:15 esptool.py

**Repair QIO**  [see](https://github.com/espressif/arduino-esp32/pull/6962/commits/f5f8f909135d95f80be01e0560c2cfa7db842d56)

nano .arduino15/packages/esp32/hardware/esp32/2.0.4/boards.txt
[image]

**Results**
- using ArduinoIDE upload sketch to TinyS3 - FAIL, bootloop

**Using the archive**
cp ~/esptool-3.3/esptool.py ~/.arduino15/packages/esp32/tools/esptool_py/3.3.0/

laudix@RPiArduino:~ $ ls -l ~/.arduino15/packages/esp32/tools/esptool_py/3.3.0/
total 264
-rwxr-xr-x 1 laudix laudix 268415 Sep  5 11:54 esptool.py

**Results**
- using ArduinoIDE upload sketch to TinyS3 - **SUCCESS**!

LaudixGit, 2022-09-05:
@SuGlider , @UnexpectedMaker 

Using the archive file I was successful at using ArduinoIDE to upload to theTinyS3 with both ESP 2.0.3 and ESP 2.0.4
(see above for details)

SuGlider, 2022-09-05:
Nice! Thanks for the information.

closed 2023-04-13 by mrengineer7777