espressif/arduino-esp32 #7126: Not able to upload code to NODEMCU esp32

Prakul2006, 2022-08-14:
### Board

NodeMCU ESP32

### Device Description

ESP-WROOM-32s NodeMCU by Ai Thinker

### Hardware Configuration

No connection to any pins

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 11.6.5

### Flash frequency

Any Frequency not working 

### PSRAM enabled

no

### Upload speed

921600

### Description

When i try to upload any sketch even blank ones the esptool throws the below error
`Arduino: 1.8.13 (Mac OS X), Board: "NodeMCU-32S, 80MHz, 921600, None"

Sketch uses 243529 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16420 bytes (5%) of dynamic memory, leaving 311260 bytes for local variables. Maximum is 327680 bytes.
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number
esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number
`

### Sketch

```cpp
/* The true ESP32 chip ID is essentially its MAC address.
This sketch provides an alternate chip ID that matches 
the output of the ESP.getChipId() function on ESP8266 
(i.e. a 32-bit integer matching the last 3 bytes of 
the MAC address. This is less unique than the 
MAC address chip ID, but is helpful when you need 
an identifier that can be no more than a 32-bit integer 
(like for switch...case).

created 2020-06-07 by cweinhofer
with help from Cicicok */
	
uint32_t chipId = 0;

void setup() {
	Serial.begin(115200);
}

void loop() {
	for(int i=0; i<17; i=i+8) {
	  chipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
	}

	Serial.printf("ESP32 Chip model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
	Serial.printf("This chip has %d cores\n", ESP.getChipCores());
  Serial.print("Chip ID: "); Serial.println(chipId);
  
	delay(3000);

}
```

### Debug Message

```plain
Arduino: 1.8.13 (Mac OS X), Board: "NodeMCU-32S, 80MHz, 921600, Debug"

Sketch uses 247341 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16484 bytes (5%) of dynamic memory, leaving 311196 bytes for local variables. Maximum is 327680 bytes.
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number
esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-08-14:
@PilnyTomas Do you have a mac to test?
https://github.com/espressif/arduino-esp32/commit/3cb73dccd0541b2222e0341e7261754ef4b52bb2

PilnyTomas, 2022-08-15:
Can you please test PR #7129

danwahl, 2022-08-16:
Had the same issue on 3597881 (Adafruit Feather ESP32-S3 2MB PSRAM, Windows 10, Arduino 1.8.19) and #7129 seems to resolve it

PilnyTomas, 2022-08-25:
@Prakul2006 can you please test with the PR? If it solved your issue, please close the issue.
We will close the issue as solved if there is no response in the next week.

closed 2022-10-04 by PilnyTomas

dallday, 2022-10-23:
Running under Linux and found that the m5stick-c-plus.menu.EraseFlash entries were missing from the boards.txt file

Added them and can now program my M5StickC-plus 
Thanks
David

gitTinker, 2022-10-26:
I had the identical error `argument <address> <filename>: Address "{upload.erase_cmd}" must be a number`
I updated the boards file adding the **bold** text below.
"\Arduino\hardware\espressif\esp32\boards.txt"
With _these_ settings, **successful** upload of blink sketch

**the boards file needs to be corrected**

My environment:
IDE Host: Jetson Nano w/ Ubuntu 18.04.6 LTS (GNU/Linux 4.9.253-tegra aarch64)
Arduino IDE: 1.8.15
ESP Board: 2.05 (from https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json)

[image]

`##############################################################

wt32-eth01.name=WT32-ETH01 Ethernet Module

wt32-eth01.bootloader.tool=esptool_py
wt32-eth01.bootloader.tool.default=esptool_py

wt32-eth01.upload.tool=esptool_py
wt32-eth01.upload.tool.default=esptool_py
wt32-eth01.upload.tool.network=esp_ota

wt32-eth01.upload.maximum_size=8388608
wt32-eth01.upload.maximum_data_size=327680
wt32-eth01.upload.flags=
wt32-eth01.upload.extra_flags=

wt32-eth01.serial.disableDTR=true
wt32-eth01.serial.disableRTS=true

wt32-eth01.build.tarch=xtensa
wt32-eth01.build.bootloader_addr=0x1000
wt32-eth01.build.target=esp32
wt32-eth01.build.mcu=esp32
wt32-eth01.build.core=esp32
wt32-eth01.build.variant=wt32-eth01
wt32-eth01.build.board=WT32_ETH01

wt32-eth01.build.f_cpu=240000000L
wt32-eth01.build.flash_size=4MB
wt32-eth01.build.flash_freq=40m
wt32-eth01.build.flash_mode=dio
wt32-eth01.build.boot=dio
wt32-eth01.build.partitions=default
wt32-eth01.build.defines=
wt32-eth01.build.extra_libs=

wt32-eth01.menu.PartitionScheme.default=Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)
wt32-eth01.menu.PartitionScheme.default.build.partitions=default
wt32-eth01.menu.PartitionScheme.defaultffat=Default 4MB with ffat (1.2MB APP/1.5MB FATFS)
wt32-eth01.menu.PartitionScheme.defaultffat.build.partitions=default_ffat
wt32-eth01.menu.PartitionScheme.minimal=Minimal (1.3MB APP/700KB SPIFFS)
wt32-eth01.menu.PartitionScheme.minimal.build.partitions=minimal
wt32-eth01.menu.PartitionScheme.no_ota=No OTA (2MB APP/2MB SPIFFS)
wt32-eth01.menu.PartitionScheme.no_ota.build.partitions=no_ota
wt32-eth01.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
wt32-eth01.menu.PartitionScheme.noota_3g=No OTA (1MB APP/3MB SPIFFS)
wt32-eth01.menu.PartitionScheme.noota_3g.build.partitions=noota_3g
wt32-eth01.menu.PartitionScheme.noota_3g.upload.maximum_size=1048576
wt32-eth01.menu.PartitionScheme.noota_ffat=No OTA (2MB APP/2MB FATFS)
wt32-eth01.menu.PartitionScheme.noota_ffat.build.partitions=noota_ffat
wt32-eth01.menu.PartitionScheme.noota_ffat.upload.maximum_size=2097152
wt32-eth01.menu.PartitionScheme.noota_3gffat=No OTA (1MB APP/3MB FATFS)
wt32-eth01.menu.PartitionScheme.noota_3gffat.build.partitions=noota_3gffat
wt32-eth01.menu.PartitionScheme.noota_3gffat.upload.maximum_size=1048576
wt32-eth01.menu.PartitionScheme.huge_app=Huge APP (3MB No OTA/1MB SPIFFS)
wt32-eth01.menu.PartitionScheme.huge_app.build.partitions=huge_app
wt32-eth01.menu.PartitionScheme.huge_app.upload.maximum_size=3145728
wt32-eth01.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)
wt32-eth01.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
wt32-eth01.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080

wt32-eth01.menu.FlashMode.qio=QIO
wt32-eth01.menu.FlashMode.qio.build.flash_mode=dio
wt32-eth01.menu.FlashMode.qio.build.boot=qio
wt32-eth01.menu.FlashMode.dio=DIO
wt32-eth01.menu.FlashMode.dio.build.flash_mode=dio
wt32-eth01.menu.FlashMode.dio.build.boot=dio

wt32-eth01.menu.FlashFreq.80=80MHz
wt32-eth01.menu.FlashFreq.80.build.flash_freq=80m
wt32-eth01.menu.FlashFreq.40=40MHz
wt32-eth01.menu.FlashFreq.40.build.flash_freq=40m

wt32-eth01.menu.UploadSpeed.921600=921600
wt32-eth01.menu.UploadSpeed.921600.upload.speed=921600
wt32-eth01.menu.UploadSpeed.115200=115200
wt32-eth01.menu.UploadSpeed.115200.upload.speed=115200
wt32-eth01.menu.UploadSpeed.256000.windows=256000
wt32-eth01.menu.UploadSpeed.256000.upload.speed=256000
wt32-eth01.menu.UploadSpeed.230400.windows.upload.speed=256000
wt32-eth01.menu.UploadSpeed.230400=230400
wt32-eth01.menu.UploadSpeed.230400.upload.speed=230400
wt32-eth01.menu.UploadSpeed.460800.linux=460800
wt32-eth01.menu.UploadSpeed.460800.macosx=460800
wt32-eth01.menu.UploadSpeed.460800.upload.speed=460800
wt32-eth01.menu.UploadSpeed.512000.windows=512000
wt32-eth01.menu.UploadSpeed.512000.upload.speed=512000

wt32-eth01.menu.DebugLevel.none=None
wt32-eth01.menu.DebugLevel.none.build.code_debug=0
wt32-eth01.menu.DebugLevel.error=Error
wt32-eth01.menu.DebugLevel.error.build.code_debug=1
wt32-eth01.menu.DebugLevel.warn=Warn
wt32-eth01.menu.DebugLevel.warn.build.code_debug=2
wt32-eth01.menu.DebugLevel.info=Info
wt32-eth01.menu.DebugLevel.info.build.code_debug=3
wt32-eth01.menu.DebugLevel.debug=Debug
wt32-eth01.menu.DebugLevel.debug.build.code_debug=4
wt32-eth01.menu.DebugLevel.verbose=Verbose
wt32-eth01.menu.DebugLevel.verbose.build.code_debug=5

**wt32-eth01.menu.EraseFlash.none=Disabled
wt32-eth01.menu.EraseFlash.none.upload.erase_cmd=
wt32-eth01.menu.EraseFlash.all=Enabled
wt32-eth01.menu.EraseFlash.all.upload.erase_cmd=-e**

##############################################################`