espressif/arduino-esp32 #7130: Practical usage of Wokwi Online Arduino Simulator

VojtechBartoska, 2022-08-15:


VojtechBartoska, 2022-08-18:
**Wokwi docs links:**
- [ESP32 documentation](https://docs.wokwi.com/guides/esp32)
- [diagram.json file format](https://docs.wokwi.com/diagram-format)

urish, 2022-08-18:
List of the 15 most popular libraries on Wokwi:

| Library             | Project Count |
| ------------------- | ------------- |
| LiquidCrystal_I2C.h | 47647         |
| Servo.h             | 30721         |
| LiquidCrystal.h     | 20950         |
| WiFi.h              | 19537         |
| DHT.h               | 16404         |
| Adafruit_GFX.h      | 14452         |
| Adafruit_SSD1306.h  | 13165         |
| FastLED.h           | 11625         |
| ArduinoTrace.h      | 10857         |
| Keypad.h            | 9483          |
| DHTesp.h            | 8523          |
| RTClib.h            | 7270          |
| PubSubClient.h      | 5302          |
| Adafruit_NeoPixel.h | 5272          |
| EEPROM.h            | 5029          |

VojtechBartoska, 2024-02-20:
We do have now library tests implemented.

Wokwi is used from time to time for demonstration.

I think for now we can close this ticker as there are no action points.

closed 2024-02-20 by VojtechBartoska

urish, 2024-02-20:
Yeah, been open for a long time!