espressif/arduino-esp32 #7133: Cannot call weak function from Arduino C-Library to ESP32 .ino sketch

Erriez, 2022-08-15:
### Board

Any ESP32

### Device Description

Any ESP32

### Hardware Configuration

None.

### Version

v2.0.4

### IDE Name

Arduino 1.8.19

### Operating System

Ubuntu 22.04

### Flash frequency

Not relevant

### PSRAM enabled

no

### Upload speed

Not relevant

### Description

An Arduino C-Library calls a weak function which should be defined in the Arduino `.ino` sketch. Building works for AVR and ESP8266 targets, but generates build errors when building for ESP32.

### Sketch

```cpp
**examples/ErriezTest.ino**
++
#include <Arduino.h>
#include <ErriezTest.h>

#ifdef __cplusplus
extern "C" {
#endif

// Function is called from C-library
void myCallback()
{
    digitalWrite(LED_BUILTIN, HIGH);
}

#ifdef __cplusplus
}
#endif

void setup()
{
    // Initialize built-in LED
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
}

void loop()
{
}

**src/ErriezTest.h**
```c++
#ifndef ERRIEZ_TEST_H_
#define ERRIEZ_TEST_H_

void foo(void);

#endif // ERRIEZ_TEST_H_
```

**src/ErriezTest.c**
```c++
#include <Arduino.h>
#include "ErriezTest.h"
 
/* Function prototypes */
void myCallback(void) __attribute__((weak));
extern void myCallback(void);

void foo(void)
{
    if (myCallback != NULL) {
        myCallback();
    }
}
```
```

### Debug Message

```plain
ErriezTest:44:6: error: conflicting declaration of 'void setup()' with 'C' linkage
 void setup()
      ^~~~~
In file included from /home/user/Arduino/libraries/ErriezTest/examples/ErriezTest/ErriezTest.ino:25:
/home/user/.arduino15/packages/esp32/hardware/esp32/2.0.4/cores/esp32/Arduino.h:136:6: note: previous declaration with 'C++' linkage
 void setup(void);
      ^~~~~
ErriezTest:51:6: error: conflicting declaration of 'void loop()' with 'C' linkage
 void loop()
      ^~~~
In file included from /home/user/Arduino/libraries/ErriezTest/examples/ErriezTest/ErriezTest.ino:25:
/home/user/.arduino15/packages/esp32/hardware/esp32/2.0.4/cores/esp32/Arduino.h:137:6: note: previous declaration with 'C++' linkage
 void loop(void);
      ^~~~
Using library ErriezTest at version 1.0.0 in folder: /home/user/Arduino/libraries/ErriezTest 
exit status 1
conflicting declaration of 'void setup()' with 'C' linkage
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-21:
@Erriez - That's interesting.

I tested your example and it causes the error you have posted.
But if I change the order of declaration in the INO file, it works fine:

``` cpp
#include <Arduino.h>
#include "ErriezTest.h"

void setup()
{
    // Initialize built-in LED
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
}

void loop()
{
}

#ifdef __cplusplus
extern "C" {
#endif

// Function is called from C-library
void myCallback()
{
    digitalWrite(LED_BUILTIN, HIGH);
}

#ifdef __cplusplus
}
#endif
```

Erriez, 2022-08-22:
Thanks for your feedback. I did not expect this behavior as it works for AVR and ESP8266 (and maybe more). In the mean time I've found a workaround by adding a callback function, so I'm not blocked on this.

closed 2022-12-15 by VojtechBartoska