espressif/arduino-esp32 #7135: Arduino as component does not compile with bluetooth libraries

joeholdsworth, 2022-08-16:
### Board

ESP32c3

### Device Description

devkitc

### Hardware Configuration

NA

### Version

v2.0.4

### IDE Name

Visual studio ESP-IDF plugin

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Cant compile using Arduino as a component and with bluetooth enabled with the default options in menuconfig:

> [1433/1435] Linking CXX executable bm.elf
> FAILED: bm.elf
> cmd.exe /C "cd . && C:\Users\User\BM\software\esp\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc -nostartfiles -march=rv32imc --specs=nosys.specs @CMakeFiles\bm.elf.rsp -o bm.elf  && cd ."
> c:/users/user/bm/software/esp/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(BLEAdvertising.cpp.obj): in function `BLEAdvertising::start()':
> C:/Users/User/BM/software/esp/projects/bm/components/arduino/libraries/BLE/src/BLEAdvertising.cpp:247: undefined reference to `esp_ble_gap_start_advertising'
> c:/users/user/bm/software/esp/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/User/BM/software/esp/projects/bm/components/arduino/libraries/BLE/src/BLEAdvertising.cpp:214: undefined reference to `esp_ble_gap_config_adv_data'
> c:/users/user/bm/software/esp/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/User/BM/software/esp/projects/bm/components/arduino/libraries/BLE/src/BLEAdvertising.cpp:225: undefined reference to `esp_ble_gap_config_adv_data'
> collect2.exe: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1

### Sketch

```cpp
These are the includes:

#include "Arduino.h"
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>
```

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

I am using esp-idf release v4.4.2
arduino as component release v2.0.4

to repoduce run 

> idf.py build

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

joeholdsworth, 2022-08-19:
@SuGlider can you help with this one too?

SuGlider, 2022-08-19:
@joeholdsworth - It is very possible that BLE is not enabled in your project `sdkconfig`

Compare it to https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig
Or maybe just copy the Arduino `sdkconfig file` to your project.

These are the Arduino Bluethooth options - this BLE you need Bluedroid or NimBLE.

```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y

#
# Bluetooth controller
#
# CONFIG_BTDM_CTRL_MODE_BLE_ONLY is not set
# CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY is not set
CONFIG_BTDM_CTRL_MODE_BTDM=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=3
CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN=2
CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN=0
# CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_HCI is not set
CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_PCM=y
CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF=1
CONFIG_BTDM_CTRL_PCM_ROLE_EDGE_CONFIG=y
CONFIG_BTDM_CTRL_PCM_ROLE_MASTER=y
# CONFIG_BTDM_CTRL_PCM_ROLE_SLAVE is not set
CONFIG_BTDM_CTRL_PCM_POLAR_FALLING_EDGE=y
# CONFIG_BTDM_CTRL_PCM_POLAR_RISING_EDGE is not set
CONFIG_BTDM_CTRL_PCM_ROLE_EFF=0
CONFIG_BTDM_CTRL_PCM_POLAR_EFF=0
# CONFIG_BTDM_CTRL_AUTO_LATENCY is not set
CONFIG_BTDM_CTRL_LEGACY_AUTH_VENDOR_EVT=y
CONFIG_BTDM_CTRL_LEGACY_AUTH_VENDOR_EVT_EFF=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN_EFF=3
CONFIG_BTDM_CTRL_BR_EDR_MAX_ACL_CONN_EFF=2
CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF=0
CONFIG_BTDM_CTRL_PINNED_TO_CORE_0=y
# CONFIG_BTDM_CTRL_PINNED_TO_CORE_1 is not set
CONFIG_BTDM_CTRL_PINNED_TO_CORE=0
CONFIG_BTDM_CTRL_HCI_MODE_VHCI=y
# CONFIG_BTDM_CTRL_HCI_MODE_UART_H4 is not set

#
# MODEM SLEEP Options
#
CONFIG_BTDM_CTRL_MODEM_SLEEP=y
CONFIG_BTDM_CTRL_MODEM_SLEEP_MODE_ORIG=y
# CONFIG_BTDM_CTRL_MODEM_SLEEP_MODE_EVED is not set
CONFIG_BTDM_CTRL_LPCLK_SEL_MAIN_XTAL=y
# end of MODEM SLEEP Options

CONFIG_BTDM_BLE_DEFAULT_SCA_250PPM=y
CONFIG_BTDM_BLE_SLEEP_CLOCK_ACCURACY_INDEX_EFF=1
CONFIG_BTDM_BLE_SCAN_DUPL=y
CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE=y
# CONFIG_BTDM_SCAN_DUPL_TYPE_DATA is not set
# CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE is not set
CONFIG_BTDM_SCAN_DUPL_TYPE=0
CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE=20
CONFIG_BTDM_BLE_MESH_SCAN_DUPL_EN=y
CONFIG_BTDM_MESH_DUPL_SCAN_CACHE_SIZE=100
CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED=y
CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP=y
CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_NUM=100
CONFIG_BTDM_BLE_ADV_REPORT_DISCARD_THRSHOLD=20
CONFIG_BTDM_RESERVE_DRAM=0xdb5c
CONFIG_BTDM_CTRL_HLI=y
# end of Bluetooth controller
[... 139 lines cut ...]
# CONFIG_BLE_MESH_SCENE_CLI is not set
# CONFIG_BLE_MESH_SCHEDULER_CLI is not set
# CONFIG_BLE_MESH_LIGHT_LIGHTNESS_CLI is not set
# CONFIG_BLE_MESH_LIGHT_CTL_CLI is not set
# CONFIG_BLE_MESH_LIGHT_HSL_CLI is not set
# CONFIG_BLE_MESH_LIGHT_XYL_CLI is not set
# CONFIG_BLE_MESH_LIGHT_LC_CLI is not set
CONFIG_BLE_MESH_GENERIC_SERVER=y
CONFIG_BLE_MESH_SENSOR_SERVER=y
CONFIG_BLE_MESH_TIME_SCENE_SERVER=y
CONFIG_BLE_MESH_LIGHTING_SERVER=y
# end of Support for BLE Mesh Client/Server models

# CONFIG_BLE_MESH_IV_UPDATE_TEST is not set
CONFIG_BLE_MESH_DISCARD_OLD_SEQ_AUTH=y

#
# BLE Mesh specific test option
#
# CONFIG_BLE_MESH_SELF_TEST is not set
# CONFIG_BLE_MESH_SHELL is not set
# CONFIG_BLE_MESH_DEBUG is not set
# end of BLE Mesh specific test option

```

SuGlider, 2022-08-19:
In your project folder run `idf.py menuconfig` to see all IDF options and settings.

SuGlider, 2022-08-24:
@joeholdsworth - Please let us know if this issue is solved. Thanks.

joeholdsworth, 2022-08-24:
will do, thank you very much, will have time in a few days

joeholdsworth, 2022-10-09:
Hi, sorry it took me a while to come back to this.

So I copied the bluetooth settings above into sdkconfig. 

This still failed to compile, going into the idf.py menuconfig and setting 

`CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y `

Fixed the issue and it now compiles.

@SuGlider  Thank you for your help, its very much appreciated

SuGlider, 2022-10-11:
> thanks for testing @joeholdsworth! So we can probably close this @SuGlider?
> 

Yes, I Think that the issue is solved and we can close it.

closed 2022-10-11 by VojtechBartoska